JMRI/JMRI #8196: Switchboard doesn't fully display on iPad

bobjacobsen, 2020-03-01:
A switchboard-style panel can be configured to have graphics indicate the state:

[image]

The graphics don't display on an iPad, although it does properly track changes to sensor state via the +/- indicators:

[image]

Note that the image is correct in the overview panel on the home screen of the JMRI web server (this is with "Symbols" selected instead of "Toggles"):

[image]

rhwood, 2020-04-18:
FWIW, the overview screen is including a PNG of the window as shown in DecoderPro or PanelPro, while the panel URL is rendering the panel using JavaScript.

silverailscolo, 2020-10-13:
Fixed in PR #9075

closed 2020-10-29 by silverailscolo