JMRI/JMRI #8200: Edit Signal Mast Window Grayed Out

JLang1951, 2020-03-02:
When trying to edit a signal head controlled signal mast the edit signal mast window comes up mostly greyed out w/o the ability to edit the signal heads in the blanked out drop down boxes of the signal mast window.
 
[image]

I ran into this for the first time about a month ago and made a post on JMRI users .io group. It looks like it was mentioned again in different post just yesterday.

https://groups.io/g/jmriusers/message/169125?p=,,,20,0,0,0::relevance,,edit+signal+mast,20,2,0,71023472

JMRI 4.19.2
Windows 10/7 or Linux 18.3 mint
Java 1.8.0.241
NCE Power Pro
CMRI

Thanks in advance

bobjacobsen, 2020-03-02:
This is intentional, if undesirable.  All the greyed-out items would change the system name of the mast if they were changed.  JMRI items _can't_ change their system name once created; that's a fundamental characteristic of how the code works.

Consider it this way:  LT2 identifies "The turnout decoder with address 2", which is a specific thing.  LT3 is a different thing, not a different name for the same thing.

Similarly, the IF$shsm:NYC-1956:SL-3-high(IH1)(IH2)(IH3) identifies a signal mast ("IF$") of type signal head signal mast ("sham") implementing the NYC-1956 and of type SL-3-high that uses the IH1, IH2, IH3 signal heads.  That's a very unique item on the layout and can't be confused with any other mast.  If any of that changed, it would be referring to a different mast.

User names are the way to refer to something when you're not sure what device you'll settle on.  A user name of "Turnout East Yard" can be placed on LT2 or LT3, so if you configure JMRI to refer to that user name, you can change the underlying turnout without problems.  (Bugs in that mechanism really are bugs, and will be fixed when raised)  Similarly, if the rest of JMRI is referring to the mast in the figure above by its 283YW user name, putting that name on a different mast can effectively changes the type, system, appearance and outputs.  If that doesn't work for some reason, we should fix that.

JLang1951, 2020-03-02:
I do use user names for my items on my panels. I create my panels in a simulated mode then when connected to the hardware I create the SIMNI outputs as turnouts which in turn I create my heads then masts out of the heads and apply the user name from simulated to the created mast. The problem I was having is I can not delete the mast until I delete all of the logic that is associated with the masts user name. It is just a cumbersome way of doing it but I do now understand about the uniqueness of the system name as it relates to the tables in the panel file after you explained it. Maybe there should not be an edit signal mast window at all except having the ability to change the user name. Thoughts? Thanks for the reply.

John Lang

bobjacobsen, 2020-03-02:
> The problem I was having is I can not delete the mast until I delete all of the logic that is associated with the masts user name.

What happens when you create the new mast first, then move the user name over, then delete the old mast?

JLang1951, 2020-03-02:
I'll try it that way but it won't be till later today. I'll let you know what the outcome is. Thanks

John Lang

JLang1951, 2020-03-02:
> What happens when you create the new mast first, then move the user name over, then delete the old mast?

The work around works fine for me. Should have thought about doing that on my own. Thanks for your time and I'll close the issue.

John Lang

closed 2020-03-02 by JLang1951