JMRI/JMRI #8224: Warrant Table buttons hidden

rhwood, 2020-03-09:
On my MacBook, when the Warrant Table window is the width of the screen, buttons under the table are unreadable (and quantity unknown) (circled in red), and buttons in the table are truncated (circled in orange).

[image]

dsand47, 2020-03-13:
How does this look?

[image]

rhwood, 2020-03-13:
@dsand47 That looks usable.

dsand47, 2020-03-14:
PR #8243 fixes this issue.

rhwood, 2020-04-18:
Fixed by #8243 

@dsand47 Github has [specific syntax](https://help.github.com/en/github/managing-your-work-on-github/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword) for closing an issue when a PR is merged.

closed 2020-04-18 by rhwood