JMRI/JMRI #8238: JMRI Decoder Pro & NCE Power Cab Communications Issue

raymoph, 2020-03-13:
I am a novice JMRI user.  That said, I have used older version of JMRI Decoder Pro & NCE Power Cab to manage my HO train livery for years, with no issues.  It was a very viable tool for programming my trains.

 I recently upgraded to JMRI Production Release 4.18.  I also downloaded JAVA  8.  After transferring my train roster and making/verifying that COM3 was selected and compatible with WINDOWS 10 COM3, I activated Decoder Pro.  The complete roster screen appeared in the initial Decoder Pro window.  On the bottom of the screen window, I noted that the communications prompt was green, suggesting that communications existed be Decoder Pro and the NCE Power Cab.  Within seconds, however, the prompts turned to red and communication between the two.  I have not been able to stop the change from green to red.  Obviously, I cannot program any engines.

I have uninstalled and reinstalled JMRI Decoder Pro two times, with similar results each time.  I need in establishing continuous communications between JMRI Decoder Pro and the NCE Power Cab.  Please advise.

bobjacobsen, 2020-03-13:
@raymoph Did you intend to close this?

raymoph, 2020-03-15:
Bob, no I did not mean to close this issue.  Assuming from your question that I did so, the attempt to close was inadvertent.  I very much need responses to this question/issue.  Thank you.

dheap, 2020-03-16:
@raymoph Please follow the instructions below:

Power Cab Instructions
======================

- The NCE USB uses the SiLabs CP2102 chipset. Download and install SiLabs CP210x drivers from 
<https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers>
- There is a Universal  driver for Windows 10.
- The downloaded file will be a ".zip"
- You open that file and then open the file called:
"CP210xVCPInstaller_x86.exe" or "CP210xVCPInstaller_x64.exe" depending on whether your Windows is 32 bit or 64 bit.
- After doing this you should get the SiLabs device appearing as a new COM port in Windows Device Manager when you plug it in.
- Sometimes you may need to install the driver again after plugging in the NCE USB to get rid of the error symbol in Windows Device Manager.
- If you still have questions marks, try removing driver in Device Manager and try again as above
- When you have it working properly, the COM port should appear/disappear as you plug/unplug the NCE USB.
- Do not try to set the baud rate in Windows Device Manager.
- Ignore all other documentation to the contrary (there is some incorrect information in places) and set up JMRI as below:
- Make sure you always unplug/replug the Cab Bus cable on the NCE USB after changing jumpers or the JMRI baud rate. 

The best beginner-level/diagnostic settings for a Power Cab (any version) and an NCE USB (any version)  are:
- All USB Jumpers off (or at least 2, 3 & 4. Jumper 1 is ignored).
- JMRI preferences are:
  - System manufacturer: NCE
  - System connection:  NCE USB
  - Serial port: <whatever is appropriate for your system>
  - USB version: V7.x.x
  - System: PowerCab
  - Additional Connection Settings checkbox ON
    - Baud rate: 9600.

Open JMRI and wait a short while, then use Help->System Console...->Copy to Clipboard and paste into a text document and upload it for us to see. Alternatively, go to the Help->Locations menu item, click on Open Log Files Location and upload the session.log file to this issue (you'll need to rename it to session.txt for it to be accepted).

Also please advise the behaviour of the two LEDs (marked USB and CAB BUS) on the NCE USB (normal behaviour is both to flicker briefly when processing commands, either or both stuck on denotes a problem). Also the behaviour of the LED on the Power Cab Panel (normal behaviour is on, with a brief flicker when JMRI switches it into or out of programming mode).

raymoph, 2020-03-16:
David,

 

I am truly grateful for the rapid reply.  I will immediately use these instructions to get my JMRI/Pro Cab to operate as they should.

 

Have a great day.

 

Phil Raymond

 

https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

dheap, 2020-03-17:
@raymoph There seems to be some confusion here:

- In your initial post, you said "Power Cab".
- In your last post, you said "Pro Cab".

1. The Power Cab is a standalone DCC system, interfaced to JMRI via an NCE USB board.
2. The Pro Cab is a dumb throttle, which usually connects to a Power Pro system (has a 9 pin serial plug on the front panel, interfaced to JMRI via a third-party USB-Serial adaptor cable

The difference is critical as the hardware, JMRI setup and instructions are totally different.

What is your throttle labelled?

raymoph, 2020-03-17:
Dave,

 

As a reply to your message below, I am not sure what message you received; however, it was not from me.  Once again, I thank you for your original response. To me, as it helped me to solve the issue (#8238) I had experienced.  I am totally grateful.

 

I will go to GitHub and close #8238.  Best wishes.

 

Phil

closed 2020-03-17 by raymoph

dheap, 2020-03-17:
@raymoph I was quoting from this comment of yours: https://github.com/JMRI/JMRI/issues/8238#issuecomment-599660686 

Glad you have it resolved.

Dave