JMRI/JMRI #8250: Need help connecting NCE

petelaguardia, 2020-03-16:
See below system console window.

2020-03-16 16:26:42,130 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-03-16 16:26:42,130 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Pete LaGuardia\JMRI\log\messages.log [main]
2020-03-16 16:26:42,130 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Pete LaGuardia\JMRI\log\session.log [main]
2020-03-16 16:26:42,145 apps.AppsBase                         INFO  - DecoderPro version 4.19.4+R4abda5deb starts under Java 1.8.0_60 on Windows 7 x86 v6.1 at Mon Mar 16 16:26:42 EDT 2020 [main]
2020-03-16 16:26:42,301 gui3.Apps3                            INFO  - Starting with profile Pete_LaGuardia_2020.3f3bf100 [main]
2020-03-16 16:26:42,473 node.NodeIdentity                     INFO  - Using a21944e5-f624-447a-aae5-ef338c31a020 as the JMRI storage identity for profile id 3f3bf100 [AWT-EventQueue-0]
2020-03-16 16:26:42,582 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "NCE" [main]
2020-03-16 16:26:42,847 usbdriver.UsbDriverAdapter            INFO  - NCE USB COM3  port opened at 9600 baud [main]
2020-03-16 16:26:43,035 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-03-16 16:26:43,035 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Pete LaGuardia\JMRI\Pete_LaGuardia_2020.jmri\ [main]
2020-03-16 16:26:43,035 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Pete LaGuardia\JMRI\Pete_LaGuardia_2020.jmri\ [main]
2020-03-16 16:26:43,035 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Pete LaGuardia\JMRI\ [main]
2020-03-16 16:26:43,035 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Pete LaGuardia\ [main]
2020-03-16 16:26:43,035 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2020-03-16 16:26:52,987 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: AA consecutive timeouts = 0 in nce.NceTrafficController [nce.NceTrafficController Transmit thread]
2020-03-16 16:26:53,097 nce.NceConnectionStatus               WARN  - Incorrect or no response from NCE command station [nce.NceTrafficController Transmit thread]
2020-03-16 16:27:03,112 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: AA consecutive timeouts = 1 in nce.NceTrafficController [nce.NceTrafficController Transmit thread]
2020-03-16 16:27:03,221 nce.NceConnectionStatus               WARN  - No response from NCE command station [nce.NceTrafficController Transmit thread]
2020-03-16 16:27:03,221 jmrix.AbstractMRTrafficController     WARN  - timeout flushes receive buffer:  [nce.NceTrafficController Receive thread]
2020-03-16 16:27:13,236 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: AA consecutive timeouts = 2 in nce.NceTrafficController [nce.NceTrafficController Transmit thread]
2020-03-16 16:27:13,345 nce.NceConnectionStatus               WARN  - No response from NCE command station [nce.NceTrafficController Transmit thread]
2020-03-16 16:27:23,361 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: AA consecutive timeouts = 3 in nce.NceTrafficController [nce.NceTrafficController Transmit thread]
2020-03-16 16:27:23,470 nce.NceConnectionStatus               WARN  - No response from NCE command station [nce.NceTrafficController Transmit thread]

dheap, 2020-03-17:
@petelaguardia Thanks for your system log. It appears you have a baud rate matching issue. That is the first thing to address.

Please follow the instructions below:

Power Cab Instructions
======================

- The NCE USB uses the SiLabs CP2102 chipset. Download and install SiLabs CP210x drivers from 
<https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers>
- There is a Universal  driver for Windows 10.
- The downloaded file will be a ".zip"
- You open that file and then open the file called:
"CP210xVCPInstaller_x86.exe" or "CP210xVCPInstaller_x64.exe" depending on whether your Windows is 32 bit or 64 bit.
- After doing this you should get the SiLabs device appearing as a new COM port in Windows Device Manager when you plug it in.
- Sometimes you may need to install the driver again after plugging in the NCE USB to get rid of the error symbol in Windows Device Manager.
- If you still have questions marks, try removing driver in Device Manager and try again as above
- When you have it working properly, the COM port should appear/disappear as you plug/unplug the NCE USB.
- Do not try to set the baud rate in Windows Device Manager.
- Ignore all other documentation to the contrary (there is some incorrect information in places) and set up JMRI as below:
- Make sure you always unplug/replug the Cab Bus cable on the NCE USB after changing jumpers or the JMRI baud rate. 

The best beginner-level/diagnostic settings for a Power Cab (any version) and an NCE USB (any version)  are:
- All USB Jumpers off (or at least 2, 3 & 4. Jumper 1 is ignored).
- JMRI preferences are:
  - System manufacturer: NCE
  - System connection:  NCE USB
  - Serial port: <whatever is appropriate for your system>
  - USB version: V7.x.x
  - System: PowerCab
  - Additional Connection Settings checkbox ON
    - Baud rate: 9600.

Open JMRI and wait a short while, then use Help->System Console...->Copy to Clipboard and paste into a text document and upload it for us to see. Alternatively, go to the Help->Locations menu item, click on Open Log Files Location and upload the session.log file to this issue (you'll need to rename it to session.txt for it to be accepted).

Also please advise the behaviour of the two LEDs (marked USB and CAB BUS) on the NCE USB (normal behaviour is both to flicker briefly when processing commands, either or both stuck on denotes a problem). Also the behaviour of the LED on the Power Cab Panel (normal behaviour is on, with a brief flicker when JMRI switches it into or out of programming mode).

Dave

petelaguardia, 2020-03-17:
Thanks Dave.  

I confirmed the setting is 9600 baud rate.  under the Preferences tab  Can I send a screenshot?

closed 2021-02-23 by silverailscolo