JMRI/JMRI #8252: Missing Icons appear to hang JMRI while loading a CPE panel at 4.19.4.

dsand47, 2020-03-17:
When loading a CPE panel (and maybe PE), a missing icon results in a permanent wait condition.

The attached xml file has one icon with an intentional bad name.  This simulates the frequent support request where custom icons have been placed in the JMRI install location.

The thread dump is also attached.

This issue appears to be related to PR #8112.

[cpe-test.xml.txt](https://github.com/JMRI/JMRI/files/4341521/cpe-test.xml.txt)
[thread dump.txt](https://github.com/JMRI/JMRI/files/4341523/thread.dump.txt)

dsand47, 2020-03-23:
@danielb987 Since you created PR #8112, I am including you in this discussion.

When a PE or CPE panel is being loaded by the xml load process, the panel is created.  PE calls `CatalogPanel.makeDefaultCatalog()` in a runnable thread while CPE uses a SwingWorker.  In either case, the lock created by makeDefaultCatalog persists until the panel loading is complete.

If an error occurs in `jmri.jmrit.display.configurexml.PositionableLabelXml.getNamedIcon`, a dialog is displayed for the user to correct the error, ignore it or skip future errors.  The typical error is a missing custom icon.

The dialog created by `jmri.jmrit.display.Editor$UrlErrorDialog` calls `CatalogPanel.makeDefaultCatalog()` on the `AWT-EventQueue-0` thread.  Since it cannot get the lock which is held by the background thread, JMRI stalls with a deadlock condition.

I have identified two possible solutions.  Since I know very little about threading, etc., there are probably better solutions.  My analysis of the events may also be incomplete.

The first one is to change PE and CPE to call `CatalogPanel.makeDefaultCatalog()` without using a background thread.  The other approach is to change `CatalogPanel.createNewBranch()` so that it does not use `runOnGUI`.

danielb987, 2020-03-24:
I think this error requires some serious thinking about how the code should work and I don't know enough about CatalogPanel and how it is supposed to work to be able to fix this.

@bobjacobsen Can you take a look on this?

dsand47, 2020-03-26:
I decided to implement a third alternative.

PE and CPE were using a separate thread to initialize the program:resources and preference:resources catalog trees.  Depending on the computer hardware, this can take some time, especially for the program tree.  

The approach in PR #8276 is to defer creating the trees until they are actually needed to support icon selection while editing a panel.  This will potentially create a delay for the first icon edit action.

dsand47, 2020-03-28:
PR #8276 has been merged which fixes this issue.

closed 2020-03-28 by dsand47