JMRI/JMRI #8254: Jacoco Sortable Coverage Report Error

icklesteve, 2020-03-17:
See https://www.jmri.org/help/en/html/doc/Technical/jacoco_sort_report.php ( linked from https://www.jmri.org/help/en/html/doc/Technical/CI-status.shtml ).

`Fatal error: Uncaught Error: Class 'DOMDocument' not found in /home/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php:181 Stack trace: #0 /home/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php(314): parse_page_prim('<!DOCTYPE html>...', 'http://jmri.tag...', '', false) #1 /home/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php(59): parse_page('http://jmri.tag...', '', false) #2 {main} thrown in /home/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php on line 181`

danielb987, 2020-03-17:
https://github.com/JMRI/JMRI/pull/5864#issuecomment-423996257

https://github.com/JMRI/JMRI/pull/5864#issuecomment-424180924

The script requires PHP DOM extension to work.

On debian:
sudo apt-get install php7.1-dom

icklesteve, 2021-03-30:
Still unable to view the report, same error.

mattharris, 2021-03-30:
Seems `php7.#-dom` is not (currently) installed on the webserver.

This specific package is not available in the repo used on the webserver, however `php-xml` supposedly does the required work. This is now installed.

mattharris, 2021-03-30:
Hmm - doesn't seem to have done the trick. Note: this is the PHP version on our webserver
`PHP 7.2.24-0ubuntu0.18.04.7 (cli) (built: Oct  7 2020 15:24:25) ( NTS )`

mattharris, 2021-03-30:
Also, the URL from Jenkins is different:

`https://builds.jmri.org/jenkins/job/development/job/jacoco/lastStableBuild/jacoco/`

is what it should be (note casing) https://github.com/JMRI/JMRI/blob/master/help/en/html/doc/Technical/jacoco_sort_report.php#L28 shows an older form

danielb987, 2021-05-14:
I have forgot about this issue but will look into it.

danielb987, 2021-06-28:
This should be fixed.

icklesteve, 2021-06-29:
At https://www.jmri.org/help/en/html/doc/Technical/jacoco_sort_report.php I'm seeing the error message

```
Warning: file_get_contents(https://builds.jmri.org/jenkins/job/Development/job/JaCoCo/lastStableBuild/jacoco/): failed to open stream: HTTP request failed! HTTP/1.1 404 Not Found in /home/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php on line 312

Warning: DOMDocument::loadHTML(): Empty string supplied as input in /home/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php on line 182

Fatal error: Uncaught Error: Call to a member function getElementsByTagName() on null in /home/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php:192 Stack trace: #0 /home/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php(314): parse_page_prim('', 'https://builds....', '', false) #1 /home/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php(59): parse_page('https://builds....', '', false) #2 {main} thrown in /home/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php on line 192
```

danielb987, 2022-02-14:
I'm reopening this since I should fix it but haven't had time to do it yet.

closed 2022-04-15 by stale[bot]