JMRI/JMRI #8258: Cannot install JMRI with OpenJDK8 from AdoptOpenJDK.net.

ckier, 2020-03-19:
Cannot install JMRI with OpenJDK8 from AdoptOpenJDK.net.
Issue may be related to #6809.

System:
Running Windows 10, 32GB Ram, 8 cores.
No JDK installed.
No JMRI installed.

To reproduce:
1. Install OpenjDK 1.8 from AdoptOpenJDK.net (OpenJDK8U-jdk_x64_windows_hotspot_8u242b08.msi)
2. During install accept: Set JAVA_HOME variable and Set JavaSoft registry entries.
3. Install JMRI 4.18, accept defaults.
4. During "installing" phase receive error message "No JAVA installation found".
[image]
5. Click OK installation terminates.

Work around no 1:
1. The OpenJDK installer creates the registry elements:
Computer\HKEY_LOCAL_MACHINE\SOFTWARE\JavaSoft\Java Development Kit\1.8
Computer\HKEY_LOCAL_MACHINE\SOFTWARE\JavaSoft\Java Development Kit\8.0.242.08
2. In looking at the files Windows installer files InstallJMRI.nsi and LaunchJMRI.nsi, I see that they do not include these entries in the JRESearch block.
3. Using the Registry Editor, change the entries to:
Computer\HKEY_LOCAL_MACHINE\SOFTWARE\JDK\1.8
Computer\HKEY_LOCAL_MACHINE\SOFTWARE\JDK\8.0.242.08
4. Reinstall JMRI.

Work around no 2:
1. Use the version 11 OpenJDK installer from AdoptOpenJDK.net. This uses creates registry entries that match the ones in Windows installer files InstallJMRI.nsi and LaunchJMRI.nsi.

Fix:
1. Add additional entries into Windows installer files InstallJMRI.nsi and LaunchJMRI.nsi for:
Computer\HKEY_LOCAL_MACHINE\SOFTWARE\JavaSoft\Java Development Kit\1.8
Computer\HKEY_LOCAL_MACHINE\SOFTWARE\JavaSoft\Java Development Kit\8.0.242.08
2. Rebuild Windows installer and launcher.

icklesteve, 2020-04-15:
Hi @ckier ,
I need to do a tweak to the Windows installer so will have an opportunity to test proposed fix over the next few days,
Steve.

ckier, 2020-04-15:
Thanks Steve, let me know if I was on target with my fix.
-Chris

icklesteve, 2020-04-17:
Hi Chris.

Pretty much on target, there was a minor issue with launcher but the existing fix for Oracle JDK 11 also works with OpenJDK8 ;-)

See #8371 for PR,

Steve.

icklesteve, 2021-03-14:
Since fix applied, no issues reported afaik, closing issue.

closed 2021-03-14 by icklesteve

ckier, 2021-03-14:
Yes, all works well.
Thanks for the update.
Chris