JMRI/JMRI #8285: JMRI loconet throttle broken by #8226

Pugwash1, 2020-03-29:
#8226 
1) The command station is in 28step advanced. The throttles are created in 28step. 128Step advanced and 28Step advanced missing from pull down. 
2) When aquireing a throttle, The first time you aquire, it appears to immediatly release it disabling all controls. Aquiring again sets the controls enabled, but creates a second throttle instance. From here on in you cannot release the throttle from JMRI. Even if you release thru the slot monitor, that only changes the status not destroying the background throttle. This is caused by setting the speed step by status not by setting the speed step, along with other data, when performing the slot write as the final aquistion step of the process.

danielb987, 2020-04-24:
> 2\. When aquireing a throttle, The first time you aquire, it appears to immediatly release it disabling all controls. Aquiring again sets the controls enabled, but creates a second throttle [...]

I confirm this. I have seen this a multiple times with Digitrax Zephyr Extra and LocoBufferUSB.

When this happens, it becomes impossible to dispatch a loco with this address, unless I restart JMRI.

danielb987, 2020-04-24:
Looking at @Pugwash1 second point above, that the throttle gets released immediately.

I'm using LocoBufferUSB and Digitrax Zephyr Extra, and have done a merge upstream/master into my branch.

I start JMRI, opens the LocoNet Monitor and a New Throttle panel. I enter loco address 5502 into the throttle address panel and click on the "Set" button.

```
[BF 2A 7E 14]  Request slot for loco address 5502.
[E7 0E 10 23 7E 00 00 07 00 2A 00 71 02 05]  Report of slot 16 information:
	Loco 5502 is Not Consisted, Idle, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
[BA 10 10 45]  Set status of slot 16 to IN_USE.
[E7 0E 10 33 7E 00 00 07 00 2A 00 71 02 15]  Report of slot 16 information:
	Loco 5502 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
[B5 10 23 79]  Write slot 16 with status value 35 (0x23) - Loco is Not Consisted, Idle and operating in 128 speed step mode.
[EF 0E 10 33 7E 00 00 07 00 2A 00 71 02 1D]  Write slot 16 information:
	Loco 5502 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
[B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
```
The NetBeans output window shows: (I have changed some log.debug() to log.error()
```
2020-04-24 23:25:20,922 PanelPro.PanelPro             INFO  - Main initialization done [main]
2020-04-24 23:27:04,841 loconet.LocoNetThrottle       ERROR - Slot status changed from Idle to In-Use [AWT-EventQueue-0]
2020-04-24 23:27:04,841 throttle.AddressPanel         ERROR - propertyChange: java.beans.PropertyChangeEvent[propertyName=ThrottleConnected; oldValue=false; newValue=true; propagationId=null; source=5502(L)] [AWT-EventQueue-0]
2020-04-24 23:27:04,841 loconet.LocoNetThrottle       ERROR - Slot status changed from In-Use to Idle [AWT-EventQueue-0]
2020-04-24 23:27:04,841 throttle.AddressPanel         ERROR - propertyChange: java.beans.PropertyChangeEvent[propertyName=ThrottleConnected; oldValue=true; newValue=false; propagationId=null; source=5502(L)] [AWT-EventQueue-0]
2020-04-24 23:27:04,841 throttle.AddressPanel         ERROR - notifyThrottleDisposed [AWT-EventQueue-0]
2020-04-24 23:27:04,872 loconet.LocoNetThrottle       ERROR - Slot status changed from Idle to In-Use [AWT-EventQueue-0]
```
After I clicked on "Set", the throttle gets aquired and then released. This happens **every** time if I:
1) Starts JMRI
2) Open New Throttle
3) Waits 5 minutes
4) Enter 5502 as loco address and click "Set".

bobjacobsen, 2020-06-29:
Has somebody confirmed that it really is the #8226 PR?  There's only one commit (cbbb22bc852939425e385f72a0103b7fd73d305d), has anybody who can recreate the issue confirmed that the parent commit 0ca2944d2bb64a6df6f63cbac3a42cb9abf7fdad doesn't have the same issue?

From a quick read of the code, there's nothing LocoNet-specific in the single commit of #8226. This might be a more general problem.

Pugwash1, 2020-06-29:
I think, but do not know, that this is the first use of a faulty routine in the Loconet throttle that changes the speed step mid way thru the aquistion process.

Pugwash1, 2020-06-30:
Line 592 in loconethrottle is the wonky logic.
if (mRefreshTimer != null) // the refresh timer isn't created until
        // after initilization.  We only want to
        // modify the slot after the initilization
        // is complete.
        {
            network.sendLocoNetMessage(slot.writeMode(status));
        }
The timer is started before the slot goes active. So this sends a stepmode with common or idle status in the microseconds between asking and getting the slot.

PR #8758 (2020-07-01): Dont send status update till after Initilization is complete
fixes #8285
closed 2020-07-01 by bobjacobsen