JMRI/JMRI #8298: Layout Editor supporting turnouts do not show Inconsistent state on Webpanels

n3ix, 2020-04-02:
Test condition: a Layout Editor panel crossover with two turnouts.  One turnout is a Loconet turnout (LT) with Exact Feedback, and this turnout address actually controls the switch machines for both turnouts in the crossover on the layout.   Since the same turnout cannot be used more than once on the panel, the second turnout in the crossover is an Internal turnout (IT).  This internal turnout is set as the "Supporting" turnout for the Loconet turnout.

The crossover display works as expected in JMRI.  When the commanded state is changed and the LT starts to move it reports its state as INCONSISTENT.  At that point both the LT and IT on the panel display the correct (Inconsistent) icons.

The behavior on the Webpanel is different.  When the commanded state is changed the IT icon immediately changes to reflect the commanded state when .  Only the LT icon goes through the Inconsistent state.

n3ix, 2020-04-02:
It appears that the Webpanel is using the state of the IT and does not have knowledge of "Supporting" status.  If the supporting turnout state could be made to follow the state of the turnout it supports then the Webpanel would likely work.  I tried all the available modes for the IT without success, so this might require a new "supporting" mode?

rhwood, 2020-04-02:
A preferable fix would be for the supporting turnout's status to match the status of the turnout it is supporting in JMRI. I suspect that *all* server protocols (simple server, SRCP, WiThrottle, etc...) have this same problem.

geowar1, 2020-04-02:
The client web code just does what it's told… if JMRI says a turnout is closed then the client web code displays a closed turnout. IOW: The "fix" isn't in the client web code… it's in JMRI where it sends turnout state to the web server. Without looking at the sources I suspect that it's triggered by changes to the turnouts command state. To get the behavior that you want the web server code would have to be changed to forward actual state instead of command state. (Without looking, this may be trivial.)

rhwood, 2020-04-02:
The web server is providing the known state, not the commanded state:
https://github.com/JMRI/JMRI/blob/0596218a3f7b5d026a25905301caf47d52414227/java/src/jmri/server/json/turnout/JsonTurnoutHttpService.java#L50-L82

So I think it’s more likely that since there is no concept of linking two turnouts in `jmri.Turnout`, which is what the web server deals with, the fix would be to ensure the state of the internal turnout always matches the state of the LocoNet turnout it is mirroring.

pabender, 2020-04-02:
I haven’t looked at how any of the panels react to this, but one difference between Internal turnouts and LocoNet a turnouts is the feedback modes supported.

In particular, LocoNet turnouts support monitoring feedback. If I am not mistaken, this is the default on LocoNet and it changes states when the outgoing message is heard on the LocoNet.

One solution to the problem is to set the two turnouts to both have the same feedback mode.  You can check to make sure this works by setting the feedback mode to Direct mode, which should work on all turnouts.

geowar1, 2020-04-02:
The web client doesn't ask(/poll) for state; initially all turnout states are UNKNOWN. Only after notifications of changed state are sent to the client (via the server) is the client turnout state changed (and redrawn).
IOW: I don' think the "get" code is ever executed because of the client.
The client is notification driven.

rhwood, 2020-04-03:
The method from JsonTurnoutHttpService quoted above is the *only* code in JMRI writes a turnout to JSON. What is relevant is that the known state, not the commanded state, is in the JSON representation of a turnout.

geowar1, 2020-04-03:
So why is the INCONSISTENT state never sent?

rhwood, 2020-04-03:
> So why is the INCONSISTENT state never sent?

Because this method never sets the state to INCONSISTENT for the internal turnout:
https://github.com/JMRI/JMRI/blob/53a38e04141da307f07db7470cab3e74ba8c0a38/java/src/jmri/jmrit/display/layoutEditor/LayoutTurnout.java#L2416-L2435

Unless the internal turnout (an anonymous subclass of AbstractTurnout created by the [InternalTurnoutManager](https://github.com/JMRI/JMRI/blob/53a38e04141da307f07db7470cab3e74ba8c0a38/java/src/jmri/jmrix/internal/InternalTurnoutManager.java#L34-L52)), has a feedback mode of DELAYED, or a TurnoutOperation associated with it, it will never be INCONSISTENT per https://github.com/JMRI/JMRI/blob/53a38e04141da307f07db7470cab3e74ba8c0a38/java/src/jmri/implementation/AbstractTurnout.java#L115-L133

So the problem is one of:
- LayoutEditor is ignoring the state of one of its two turnouts, and drawing both turnouts using only the primary turnout's state
- There is no mechanism in JMRI to allow an internal turnout's state to always match another turnout's known state
- @n3ix is using two turnouts when maybe he only needs one (I can create a single turnout crossover in Layout Editor, but some flexibility in arrangements may require using two turnouts to create the crossover)

i.e. I think that in this case, it is the web panel that is behaving correctly, and the LayoutEditor that is not.

As @pabender pointed out the internal turnout can be set to operate in a DELAYED mode, in which case it will show inconsistent for the length of the delay (4 seconds by default and globally settable only using a script).

pabender, 2020-04-03:
> Unless the internal turnout (an anonymous subclass of AbstractTurnout created by the [...]

AbstractTurnout also handles ONESENSOR and TWOSENSOR feedback modes.  This won’t cause the Turnout to show INCONSISTENT, but it will cause the turnout to show UNKNOWN.

If the goal is to have the turnouts show the same statement then making both turnouts in the crossover internal turnouts using TWOSENSOR feedback ( with internal sensors for the feedback) would allow you to use logix or a script to have both turnouts always show the same state based on the current state of the real turnout.  ( this is actually an old trick.... )

n3ix, 2020-04-03:
@pabender  The Loconet Turnout is in Exact feedback mode.  The JMRI layout editor panel already shows the desired behavior, so my goal was just to have the Webpanel match the JMRI panel.  I could work around this with some of the alternate internal turnout modes but it seemed over complicated.

@rhwood I will try the replacing the 2 turnouts with the Crossover object instead.  Did this panel some time ago and seem to recall an issue with routing/signaling, but perhaps that is ok now.

Thanks.

PR #8308 (2020-04-11): feat: Allow a Turnout to follow another Turnout's state
Some features of JMRI expect that two Turnouts will match the known state, not just the commanded state. However, when two Turnouts have different feedback mechanisms, such as when one Turnout is physical with exact feedback, the second Turnout will not match the first Turnout's state. This feature allows a Turnout to match another Turnout's state (i.e. to follow the Turnout), and is implemented as part of Turnout to avoid the need for other features of JMRI to care if a Turnout is internal or external without making public the ability to [... cut]
closed 2020-04-11 by bobjacobsen