JMRI/JMRI #8307: reSizeToFitOnScreen causing JMRI malfunctions

Pugwash1, 2020-04-05:
reSizeToFitOnScreen, is a routine in JmriFrame that is used to ensure a window is in the visible area of the screen. It is called on every invocation of JmriFrame.pack.
This routine is slow and hogs the swing thread, and also appears to effect background threads.
In a single screen windows 10 environment the delays are miniimal but measurable by effect. A train stopping using a speed profile in a 1000mm block overshoots by a 10 or so millimetres (about 1%) at the other end of the scale a Linux machine with four monitors the same train will over shoot the stopping point by 2000mm. Sensor firings are delayed to the point that when the sensor going active is  processed the train can be into the next block, or onto the floor. Roster speed profiling is totally inaccurate. 
It took me a long time to track this down, and I lost a few engines and graveyard full of little n-scale friends.
As far as I can work out its pupose was to ensure that when loading a window whose co-ordinates had been saved that the window would be visible. So why is it called on every pack?
I think it needs to be only called on loading a screen whose co-ordinates have been saved, and getting the screen sizes and layout should only happen once, at the start of JMRI.

mstevetodd, 2020-04-06:
@Pugwash1 I can verify I've seen those same symptoms with recent test and dev versions, particularly on my RPi. I've also had some user reports which sounded similar, but the user was unable to provide enough specifics to be certain.

Pugwash1, 2020-04-08:
@mstevetodd We are actually doing this to our selves. The reSizeToFitOnScreen routine does take a lot longer than before Nov 2019, but, the AutoTrains window is resizing and repacking on every little change. Most of the time in a loconet scenerio with dual screens under linux you finish up with 2 autoengineers being created on startup because of this. I think we should fix the size of the AutoTrains window and not continually resize and pack.
Also I really dont see why we even have  the reSizeToFitOnScreen routine as there is already code in the start up that ensures a window always opens on a visible screen. Once thats up users can it anywhere, depending on OS even off the visible screen(s).
We also have a similar problem in CreateActive Train window, that does some really weird dodo that causes multiple hidden instances of the windows to occur!!!! What do you, or anyone else think?

Pugwash1, 2020-04-08:
As an extra something, a test that creates 47 windows takes 50% longet to run with the reSize routine than without out it. Think how much faster the tests will run...

bobjacobsen, 2020-04-08:
Would it make more sense to have this in `setVisible(true)` than in `pack()`?

Pugwash1, 2020-04-08:
@bobjacobsen To be perfectly honest I don't know. The routine itself  doubles up the check "is it visible on a display" that is already done when the frame is first created, so really all I can see that it does is stop a developer trying to move the screen around and getting it wrong. I tried Gnome, kde, pixel and windows 10 and couldn't as a user move the window off screen after I removed that code. I didn't look to see how exactly it was doing it, the codes a bit long, the code during frame creation is 5 lines and appears to work fine. I do know we need to fix the two windows, AutoTrains, and ActivateNewTrain as those are real pieces of work.

Pugwash1, 2020-04-08:
Both wrong button.

mstevetodd, 2020-04-08:
> I think we should fix the size of the AutoTrains window and not continually resize and pack.

I'm all for that, or any other improvements to the Dispatcher UI experience. As you mention, there are several issues that crop up periodically.

Pugwash1, 2020-05-04:
Closed by #8378

closed 2020-05-04 by Pugwash1

Pugwash1, 2020-05-05:
@mstevetodd @bobjacobsen 

So, this is as far as I have got with the AutoTrains control..
The action buttons are pilots, not locators.
From left to right, Off/resume , reverser, throttle (depicts 8 steps, and under manual control only has 8 steps + stop), and the Auto/Manual.
Top is train name or roster icon, bottom are dcc address, percent throttle, scale speed (if profiled).
The buttons are a button bar and be draged out.
Each autoengineer is a component. The frame has the stop all. 
I will make an abstract and put the old & this from that.
We also want to make the compoent available in seperate windows, and fully resizable so it works well in W10 tablet mode.
[image]
[image]