JMRI/JMRI #8310: layout editor reverts from user icon back to jmri icon if rotated

n3ix, 2020-04-06:
I replaced a built-in signal mast icon with my own icon, which I put in the equivalent folder under my User Prefs directory.  Layout Editor now displays my icon instead of the built-in one as desired.  However if I set the rotation of the Layout Editor object to anything other than 0, it reverts to displaying the built-in icon.

geowar1, 2020-04-22:
I remember fixing an issue before where rotating an icon would revert to cached values instead of rebuilding the icon cache. Sounds very similar (if not same) as this. I'll give it a look-see.

geowar1, 2020-04-23:
Forward from @n3ix:

> Placed my own version of rule-282.gif signal aspect image in my user prefs directory under the folder resources/icons/smallschematics/aspect/RG-1965/SL-1-high-abs.  With the associated layout editor panel signal mast icon rotation set to 0 and the signal mast set to the aspect in question, the panel shows my new gif. As soon as I set the icon rotation to any non-zero value the image changes back to the original jmri gif.  If I set the icon rotation back to 0 then my new image immediately appears.

n3ix, 2021-04-01:
@geowar1 did you find the cause and resolve this?

closed 2021-05-31 by stale[bot]