JMRI/JMRI #8333: Need Help adding Lenz .xml file to Decoder Pro

bigsteam5344, 2020-04-12:
Hello, 

New to the site. I am having a heck of a time trying to add a .xml file to DecorderPro. Ive done everything I can that is mentioned on the JMRI website and I have had no luck. I bought a Lenz Spur O BR50 steam locomotive. The program does not recognize it, so I went their web site and found the Decoder Template. This is the link (https://www.lenz-elektronik.de/pdf/Lenz_Spur0_BR50.xml) if anyone can help me add this to Decoder Pro that would be amazing,

Thanks,
Bruk.

bobjacobsen, 2020-04-12:
That file isn't in the JMRI format.  Not sure what it's intended for, but all the structural information is in German: `<Group_Allgemein Name="Grundeinstellungen" Key="GR1">`, "CV1_Adresse", "CV3_Anfahrverz" etc. Perhaps some Lenz tool?

bigsteam5344, 2020-04-12:
This is what it is converted to English

[Lenz O BR50 English.txt](https://github.com/JMRI/JMRI/files/4467108/Lenz.O.BR50.English.txt)

Re465, 2020-04-17:
This is a XML-template that is just bey Lenz-CV-Editor.
Them decoder is a OEM decoder for scale 0 locomotive and the CV-list deviates of de gold and silver decoder.
The decoder user manual is only in German, I can write a XML-template for DecoderPro.
But I must have the value of CV7 from you (bey Lenz, I can’t find him)so DecoderPro can find XML-template. Also I can’t test the template because have not them decoder.

bigsteam5344, 2020-04-20:
This is the CV list that I have made from the Decoder Manual converted to English 
[Lenz Spur O BR50 CV Lists.xlsx]

bigsteam5344, 2020-04-20:
Here is a an Excel sheet of the CV's and Values from the Decoder sheet from the website
[Lenz Spur O BR50 CV Lists.xlsx](https://github.com/JMRI/JMRI/files/4506559/Lenz.Spur.O.BR50.CV.Lists.xlsx)

Re465, 2020-04-25:
I have written the xml-file with the scale 0 OEM decoders from Lenz and you can download the XML-file under: https://github.com/JMRI/JMRI/blob/master/xml/decoders/Lenz_Scale_0.xml

bobjacobsen, 2020-05-30:
Closing, as @Re465’s file has been merged.

closed 2020-05-30 by bobjacobsen