JMRI/JMRI #8336: Connection Preferences: Naming limitations for serial-ports

jh95468, 2020-04-12:
The 4.18 distribution file jmri.conf contains examples of how to set serial ports for use by connections to the layout:

`# - Set the serial ports to /dev/loconet and /dev/cmri:
#   default_options="--serial-ports=/dev/loconet,/dev/cmri"
`
These examples however do not work (on Linux/Ubuntu), since they do not conform to the regex filter specified in the .properties file describing valid port names.  Choices such as /dev/loconet or /dev/cmri specified in a --serial-ports argument will not appear in the dropdown box used to select serial ports.   Neither jmri.conf or using --serial-ports as a command line option works.

Suggestions:
(1) change the examples in jmri.conf to choices that will work, and include a description of the regex filter which JMRI uses to allow only certain names so that users can configure their devices appropriately, or
(2) change the code so that whenever --serial-ports is specified, the regex filter is bypassed so that whatever names the user specifies are acceptable as serial ports

rhwood, 2020-04-12:
JMRI does not use any regex filters for controlling TTY device (serial port) names in Linux or macOS, however the library we depend upon requires that the TTY devices be specified by name, not by path.

rhwood, 2020-04-13:
I failed to test earlier, but JMRI will *not* list devices that do not exist, but will happily list devices that are not comm port (TTY) devices (typically serial ports).

jh95468, 2020-04-13:
Thanks for the quick fix!
One other observation - Experiments indicate that there is some regex-like activity going on somewhere with TTY devices.  

Someone pointed me to the file PortNamePatterns.properties in JMRI/java/src/jmri/jmrix/ which has a TTY-related regex.  What I then found is that if I created a serial device acceptable to the regex (using udev .rules), e.g., /dev/ttyUSBloconet, that entry would appear in the Edit/Preferences dropdown as a choice.  A serial device not matching that regex, e.g., /dev/ttyloconet, would not appear as a choice.

What I discovered just a few minutes ago is that such behavior is not dependent on setting serial-ports at all.  Device name matching the regex appear as choices even without setting serial-ports, and names that don't match don't appear.  So my changing the serial-ports parameter was having no effect (because I was using paths rather than names)

But there does seem to be regex involved in the code somewhere.
/Jack in California

rhwood, 2020-04-13:
That's funny, I was testing with device names that don't match those patterns (albeit not to the point of using the devices, where they would have failed because I was naming non-serial port devices for my tests).

You should also be able to add ` -Dpurejavacomm.portnamepattern=foo` to allow (and only allow) `foo` to be a valid port in `jmri.conf` or replace with a regex of your choice.

jh95468, 2020-04-13:
Since I don't know the code itself, I (usually) try to avoid tweaking things I don't understand - all hell can break loose...

Your comment about "JMRI will not list devices that do not exist" may explain some of my experience with things working and then not working and then working again.   Udev devices don't exist unless the USB cable is plugged in, and I was plugging/unplugging lots of stuff to see what worked.  So sometimes my /dev/ttyUSBloconet would be available, and sometimes it wouldn't, despite what I had put in to serial-options (which of course didn't matter anyway but I didn't know that).

It might be worth another comment or two somewhere (jmri.conf?) about which device names are OK (the regex) without need to specify in serial-ports, and the "it must exist" feature.   

It was also curious that, when I had a "bad" spec in serial-ports, the dropdown box would show one of my regular ttyUSBx choices, but in a red font,  followed by "No ports found."   I'm not sure what the red was trying to tell me.

Just thinking about the next time for someone.

rhwood, 2020-04-13:
The red font should indicate that you had already configured that port, but that it could not be opened (i.e. the device was unplugged).

My testing shows that the regexes are ignored when the `--serial-ports=...` is passed on the command line or in jmri.conf, as they are used for the "automatic" discovery of the ports absent that line.

PR #8338 (2020-04-13): fix: Update serial ports docs for macOS and Linux
- [x] Fix examples to not include "/dev/"
- [x] Update documentation to clarify that the name of the port, not the path to it, must be used
- [x] Remove erroneous documentation that macOS ignores `--serial-port` parameter 

Fixes #8336
closed 2020-04-13 by bobjacobsen