JMRI/JMRI #8339: Layout Editor support for multiple turnouts sharing a single address

bobjacobsen, 2020-04-13:
It's common for layouts (the part with the trains) to have multiple turnout motors responding to a single address.

Layout Editor represents turnouts on the layout with a LayoutEditor object. These are labelled sequentially TO1, TO2, etc as they are created.  Each one contains a reference to the jmri.Turnout object used to change the points position and get feedback.

Historically, the LayoutEditor class has enforced a restriction that a jmri.Turnout can be added to only one LayoutTurnout (including the various specializations for slips, crossovers, etc).  Since this doesn't correspond to what's actually on the layout, various workarounds have been necessary.

But why does that restriction exists?  

The `bobjacobsen/LE-allow-dup-turnout` branch has it removed. Running it will allow you to create multiple LayoutTurnouts referencing the same jmri.Turnout:

[image]

(The panel file for this is appended below)

This layout stores and loads OK, with no errors on the console. All the tests run (but maybe nothing is testing multiple turnout use, since that's not been usual).

Can we just remove this constraint?

[MultipleTurnoutLePanel.xml.zip](https://github.com/JMRI/JMRI/files/4467363/MultipleTurnoutLePanel.xml.zip)

geowar1, 2020-04-13:
One gocha you may want to check for and prevent is where one LayoutTurnout can set the state (possibly inverted from the 1st) on a 2nd turnout. If the 1st & 2nd are the same and one is inverted you could get stuck in a loop where TO1 triggers TO1(inverted) which triggers TO1 (non-inverted) which triggers TO1 (inverted)..., etc.
Just something to check.

dsand47, 2020-04-13:
I have updated the sample panel to add blocks, signal masts and signal mast logic.  So far, things are working with one exception.

The masts cannot be added to the second turnout.  The masts are attached to the LayoutTurnout, but the dialog uses the turnout name when looking for block boundaries.  I expect to see the same issue when I do the NX sensors.

[image]

For now, I have edited the xml to add the masts to the second LayoutTurnout.

The block routing and the layout block routing both look ok.  Block routing is stored as Path objects for each block.  Layout block routing is created during panel loading.

[MultipleTurnoutLePanel DS SML.xml.zip](https://github.com/JMRI/JMRI/files/4471464/MultipleTurnoutLePanel.DS.SML.xml.zip)

dsand47, 2020-04-15:
I have found the source of the problem with adding signal masts to the second layout turnout.

The add dialog has two possible states:  The layout turnout can be selected using a dialog box or it has been pre-selected.

The dialog box method used by the LE Tools menu is not currently implemented.
[image]

The **Set Signal Masts** context menu selection uses the pre-selected case.
[image]
It supplies the layout turnout which is then used to select the real name in the hidden turnout combo box.

In either case, `getTurnoutMastInformation` is called to validate the selection.  It uses the combo box selection and calls `LayoutEditorFindItems.findLayoutTurnoutByBean(NamedBean)` to get the layout turnout.  With 2 layout turnouts using the same bean, the odds are 50/50 of returning the correct layout turnout.  For signal masts, `getTurnoutMastInformation` can be changed to return true without checking.

For signal heads, which do support both methods, the user will need to be prompted as to which layout turnout they are expecting when selecting by combo box.  Do they want TO1 or TO2?  Since they have no concept of what a **layout turnout** is, one idea is to present the xy coordinates.

I have not looked at NX Sensors yet.  Since they currently only support the pre-selected approach, the fix will be the same as signal masts.

The other classes calling `findLayoutTurnoutByBean` still need to be investigated.

closed 2021-04-14 by stale[bot]