JMRI/JMRI #8342: test-single target headless

icklesteve, 2020-04-13:
Just wondering if there was a tech reason NOT to move the test-single build target back to headed?

Was changed from headed to headless in https://github.com/JMRI/JMRI/commit/b1da85dd6adecd5c06c1253b8affef27de5cf25b#diff-2cccd7bf48b7a9cc113ff564acd802a8

I keep forgetting to locally edit the build.xml after updating from Github then wasting time wondering why tests that should fail aren't !!  ( twice now I've wasted time on this! )

Ta,
Steve.

rhwood, 2020-04-13:
I would assume that is a cut and paste error from another test. That target must not be headless.

icklesteve, 2020-04-13:
Thanks for confirming @rhwood ,

Was also thinking about

```
<sysproperty key="jmri.skipTestsRequiringSeparateRunning" value="true"/><!-- skip intermittent tests -->
<sysproperty key="cucumber.options" value="--tags 'not @Ignore' --tags 'not @webtest'"/>
```

Would it also make sense to re-enable these at test-single level?
If a developer is running these locally then it would be good to have the flaky ones run more often, so increased chance of being fixed?

rhwood, 2020-04-13:
Yes, that should also be changed for single tests. Make it explicitly false.

closed 2020-04-14 by icklesteve