JMRI/JMRI #8354: "Upload Debug Info ..." capability broken

bobjacobsen, 2020-04-15:
The "Upload Debug Info..." item in the Help menu isn't currently working, at least for me.  Setting the default options:
[image]

and clicking the Send button puts the following in the log after a short pause:

```
     [java] 2020-04-15 07:06:33,842 mailreport.ReportPanel                ERROR - Error when attempting to send report: java.net.ConnectException: Connection refused (Connection refused) [AWT-EventQueue-0]
```

Nothing appears in the [target mailing list on SourceForge](https://sourceforge.net/p/jmri/mailman/jmri-reports/) (yes, that should be moved, but that's another Issue). The last report there seems to have been at 2019-10-21 21:46:03

It could be that some upstream infrastructure item has broken (more likely, lost in the Great Rehosting) or it could be a problem in current code. 

This capability hasn't been used much recently, but I've found it a useful way to get complete copies of log files and profiles.

rhwood, 2020-04-15:
I can work on creating a replacement that creates a GitHub issue, but probably not until the 4.21 timeframe (it would require that users who want assistance have GitHub accounts, but I don’t see that as any worse than needing a groups.io account).

Randall

bobjacobsen, 2020-04-15:
@rhwood I think that would be good direction!

The problem seems to be that SourceForge and/or their anti-spam vendor has decided that pretty much every email server should be on their black list for submitting to the mailing list.  Our @emjay.info servers are there, but so are @mac.com servers, @gmail.com servers, and two out of three of my other accounts.  So there seems to (currently) be no broadly-reliable way to send to that email list....

```
<jmri-reports@lists.sourceforge.net>: host mx.sourceforge.net[216.105.38.6]
   said: 550-Message rejected. Your IP (17.58.23.182) has been flagged for
   sending SPAM 550-by the UCE Protect Level 1 blocklist. For more
   information, see 550 http://www.uceprotect.net/en/rblcheck.php . (in reply
   to RCPT TO command)
Reporting-MTA: dns; mr85p00im-zteg06011501.me.com
X-Postfix-Queue-ID: 8CF7C2A082A
X-Postfix-Sender: rfc822; jacobsen@mac.com
Arrival-Date: Wed, 15 Apr 2020 15:17:57 +0000 (UTC)

Final-Recipient: rfc822; jmri-reports@lists.sourceforge.net
Original-Recipient: rfc822;jmri-reports@lists.sourceforge.net
Action: failed
Status: 5.0.0
Remote-MTA: dns; mx.sourceforge.net
Diagnostic-Code: smtp; 550-Message rejected. Your IP (17.58.23.182) has been
   flagged for sending SPAM 550-by the UCE Protect Level 1 blocklist. For more
   information, see 550 http://www.uceprotect.net/en/rblcheck.php .

From: Bob Jacobsen <...@mac.com>
Subject: Test
Date: April 15, 2020 at 8:17:56 AM PDT
To: jmri-reports@lists.sourceforge.net
```

In that example, the 17.58.23.182 mail server is part of Apple's @mac.com service.

rhwood, 2020-04-30:
When addressing this, do we want to allow for anonymous reports, or do we want to require that users be willing to file an issue on GitHub (the app can do it on their behalf, I think without requiring special permissions)?

bobjacobsen, 2020-04-30:
I don’t think we need to _require_ a GitHub account per se. I think we want reliable contact info, and a GitHub account might be the best way.

KenC57, 2020-04-30:
I think the ability to log a set of files without understanding the need for
an GitHub account. This matches what we have been doing for years up to this
point. While it is better that they do have an id, so they can follow the
replies or how it leads to tracking a code change is a plus for them. But
many, I think, won't care or understand that.
-ken c

silverailscolo, 2021-02-13:
Current release send a report into GitHub as an issue. Login account is required. Several of these have arrived since. Closing this issue as solved.

closed 2021-02-13 by silverailscolo