JMRI/JMRI #8364: PR #8335 breaks GitHub Desktop

danielb987, 2020-04-16:
@rhwood @bobjacobsen 
PR #8335 breaks GitHub Desktop. When I merge upstream/master into my master with GitHub Desktop and then "Push origin", I get the error message that the push was rejected by the server for modification of the workflow file `static-analysis.yml`.

I seem to have permission to fix this by giving GitHub Desktop permission to modify this file by authorize the GitHub Desktop application. But since I don't know exactly what all this means, I don't want to do it without permission from either @rhwood or @bobjacobsen .

[image]

[image]

danielb987, 2020-04-16:
For now, I can continue to use GitHub Desktop as long as I **don't** merge upstream/master into the branch I'm working on. But once I have merged upstream/master into one of my branches, I cannot push origin until this is fixed.

bobjacobsen, 2020-04-16:
Check to see if that file has been locally modified to be different from the one in the main repository:  https://github.com/JMRI/JMRI/blob/master/.github/workflows/static-analysis.yml

Also, do you have a local commit to that file?

```
git log -1 -- .github/workflows/static-analysis.yml
```

Is GitDesktop somehow automatically changing that file?

danielb987, 2020-04-16:
```
daniel@daniels-laptop:~/Dokument/GitHub/JMRI$ git log -1 -- .github/workflows/static-analysis.yml
commit 081214c768e1f490c3ff9f637e58037143c45014
Author: Randall Wood <297232+rhwood@users.noreply.github.com>
Date:   Sun Apr 12 16:51:49 2020 -0400

    chore: Add ECJ compilation for warnings
daniel@daniels-laptop:~/Dokument/GitHub/JMRI$ 
```

I don't see any differences between my file and the file on the GitHub server. And I have the problem both on Windows 8.1 and Debian Linux.

danielb987, 2020-04-16:
> Is GitDesktop somehow automatically changing that file?

That file does not exist before #8335. So when I merge upstream/master into my branch, that file is created, together with the workflows folder.

danielb987, 2020-04-16:
It seems that the Authorize Application is about my own repository, not the JMRI repository, which if true, means that everyone using GitHub Desktop needs to authorize GitHub Desktop for their repository because of PR #8335. It's not a problem since GitHub Desktop helps doing that.

rhwood, 2020-04-16:
What happens if you enable workflows in your fork on GitHub.com?

This may be an unanticipated complication of using GitHub actions that may mean we have revert my PR.

danielb987, 2020-04-16:
I see the problem. I have already granted `GitHub Desktop` access to all my repositories with the permissions:
* Full control of private repositories
* Update all user data

Now GitHub Desktop also needs the permission:
* Workflow - Update GitHub Action Workflow files.

For me, that's no problem since JMRI is the main project I'm working on. If this is an issue for others  I don't now. But this is a setting for **all** of my repositories, so I don't seem to be able to grant this permission for JMRI and not grant this permission for another repository.

danielb987, 2020-04-16:
[image]

[image]

rhwood, 2020-04-16:
That is only an issue on repos where that particular directory exists and has recognizable (to GitHub) workflow files in it. If that project owner does not want to allow that, they can block workflows in their project, in which case there is no problem. If they want to allow workflows, but allow you to edit them, they can just reject an PRs you submit that include them.

closed 2020-04-16 by danielb987