JMRI/JMRI #8377: Timeout starting JMRI introduced with 4.17.x

mrchuk, 2020-04-18:
I have been working with 4.16 until now. After some discussion regarding vsdecoder issues, I was advised in the forum to try 4.19.4. I found that even with no startup actions whatever, it took 25 seconds or more to open PanelPro instead of 3 seconds with 4.16. Subsequent testing showed that 4.18 has this problem as well. And I first noticed that JMRI was opening much more slowly with a release of 4.17, which I subsequently reverted.

I have found two issues causing the opening delay relative to 4.16. One is an open issue regarding multi-screen handling. This accounts for 10 seconds of the timing difference.
The other is the subject of this bug report. I changed the log level to DEBUG with 4.18 and found this in the log:
2020-04-18 08:59:19,440 pp.AbstractDCCppInitializationManager DEBUG - Timeout waiting for Command Station Response [AWT-EventQueue-0]

This does NOT occur in 4.16, and seems to have been introduced in 4.17x and all subsequent, along with the multi-screen problem. The timeout here takes roughly 8 seconds.
Please advise any further information that might be needed.

Platform details:

Lenovo E560, 16GB RAM, 4 core intel i7 @ 2.5 Ghz, 500 GB Evo SSD
OS: ubuntu 18.04
JVM: Oracle 1.8.0_201
Hardware: Arduino Uno genuine command station and genuine motor shield
Arduino IDE 1.6.6
DCC++  Base station 1.2.1 from the DCC++ web site, unmodified

mrchuk, 2020-04-18:
I have since found a very surprising thing. I tried setting logging to TRACE to see if I could get more details about actions around the timeout--and it didn't happen!
This appears to be consistent and reproducible. I do see the timeout with logging set to DEBUG. I do not when it's set to TRACE--or it's not being logged.
[messages.log](https://github.com/JMRI/JMRI/files/4498005/messages.log)
[messages.trace.log](https://github.com/JMRI/JMRI/files/4498006/messages.trace.log)

I believe it is also timing out with the default setting of INFO.

bobjacobsen, 2020-04-19:
Could you get a thread dump during the 25 second hang-up please?  [See here for [](how](https://www.jmri.org/help/en/html/doc/Technical/Threads.shtml).)

bobjacobsen, 2020-04-19:
Let try that again:  Could you get a thread dump please?  [This page](https://www.jmri.org/help/en/html/doc/Technical/Threads.shtml) explains how.

mrchuk, 2020-04-19:
I wasn't able to use jvisualvm, it just hung when I tried to open the thread dump. Used kill signal instead. I hope this file contains the info you're after.
[pp.log](https://github.com/JMRI/JMRI/files/4498250/pp.log)

mrchuk, 2020-04-20:
This thread dump is from 4.19.4
[PanelPro-4.19.4-thread-dump.log](https://github.com/JMRI/JMRI/files/4501922/PanelPro-4.19.4-thread-dump.log)

bobjacobsen, 2020-06-25:
See https://groups.io/g/jmriusers/message/176084

closed 2021-04-14 by stale[bot]