JMRI/JMRI #8382: Jenkins macOS packages fail on Catalina

dsand47, 2020-04-18:
The quarantine flags for `JMRI.4.19.5ish+jenkins+20200416T2213Z+Rcf07e830d4.dmg` and `JMRI.4.19.5ish+jenkins+20200418T1618Z+Rec37e130b5.dmg` are incorrect.

I suspect these were not notarized, although it might be a signing issue.

bobjacobsen, 2020-04-19:
Which macOS version do you have?

I think the most recent update has started enforcing some addtional sand-boxing rules.

```
error	18:14:41.404058-0700	kernel	Sandbox: bluetoothd(146) deny(1) mach-lookup com.apple.server.bluetooth
```

dsand47, 2020-04-19:
> Which macOS version do you have?
> 
> I think the most recent update has started enforcing some addtional sand-boxing rules.
> 
> ```
> error	18:14:41.404058-0700	kernel	Sandbox: bluetoothd(146) deny(1) mach-lookup com.apple.server.bluetooth
> ```

I am still on 10.15.3.  I did not want to risk 10.15.4 with the Apple stores being closed.

I have noticed that the dmg packages have not been built for a couple of weeks.  The last good 5ish I have is `JMRI.4.19.5ish+jenkins+20200323T1717Z+Rda16a7e21f.dmg` from 3/23.

bobjacobsen, 2020-04-19:
Where are you looking at the packages?

Here 

https://builds.jmri.org/jenkins/job/development/job/packages/

I see job 140 was done today

dsand47, 2020-04-19:
Yes, that is the URL.  The second dmg that I listed was from 140.

bobjacobsen, 2020-04-19:
Thanks for pointing this out.  I think job 142 has a working one (works for me), would appreciate you checking.

dsand47, 2020-04-19:
> Thanks for pointing this out. I think job 142 has a working one (works for me), would appreciate you checking.

Job 142 works.

closed 2020-04-29 by bobjacobsen