JMRI/JMRI #8389: Race condition on JMRI initialization makes Roster empty.

svatoun, 2020-04-19:
The Roster is being initialized as a part of `JmriConfigurationManager.load`. I want to first point out to that the sequence of `JmriConfigurationManager`'s initialization of `PreferenceManager`s is not fixed - that's  not a bug (!) since the initialization happen according to the dependencies. But the randomness contributes to the bug occuring  **occasionally**.

The effect of the reported bug for the end user is **empty roster**, which has been occasionally reported on JMRI mailing lists already. In short, because of unfortunate timing of `PropertyChangeEvent` from `FileUtilSupport` coming right during initialization of a `RosterConfigManager` (in parallel), the first `Roster` instance that actually registers for events and loads Roster entries gets unregistered in favour of a fresh, but empty `Roster` instance.

I admit am unable (with some tenths of attempts) to reproduce directly in JMRI, but I am hitting the issue rather often with an application built on top of JMRI. The application uses the same approach in initialization phase: run `JmriConfigurationManager.load` in its main thread while GUI thread displays something. 

As a reference I provide patched sources in my fork to correlate with the below stacktraces or logs. Please do not discard the report just because a just a derived application was the "fortunate" to hit the bug, because it equally applies to JMRI itself as well IMHO; you can verify that the parallel run of processes discussed here happens in official JMRI, too. Because of lucky coincidence, the AWT thread runs usually fast enough to complete firing the event *before* Roster constructor is invoked for the first time.

In my case, one of bean initialization sequence was: 
```
>>> apps.StartupActionsManager@41725bce, 
    apps.gui.GuiLafPreferencesManager@12faa494, 
    apps.systemconsole.SystemConsolePreferencesManager@4185212e, 
    jmri.implementation.FileLocationsPreferences@21b023d1, 
    jmri.jmrit.consisttool.ConsistPreferencesManager@729c8bb7, 
    jmri.jmrit.logix.WarrantPreferences@65c2fb63, 
    jmri.jmrit.roster.RosterConfigManager@22db2d4b, 
    jmri.jmrit.symbolicprog.ProgrammerConfigManager@32f4b7b0, 
    jmri.jmrix.ConnectionConfigManager@4abfe46f, 
    jmri.managers.ManagerDefaultSelector@c286983, 
    jmri.server.web.app.WebAppManager@5ce81fbc, 
    jmri.swing.JmriJTablePersistenceManager@2e83fea1
```    
The individual `PreferencesManager`s are initialized using depth-first search along their dependencies (`provider.getRequires()`), that is:
```
StartupActionsManager -> 
        class jmri.server.web.app.WebAppManager, 
        class jmri.jmrix.ConnectionConfigManager, 
        class jmri.jmrit.logix.WarrantPreferences, 
        class jmri.managers.ManagerDefaultSelector, 
1  >>>> class jmri.jmrit.symbolicprog.ProgrammerConfigManager, 
        class apps.gui.GuiLafPreferencesManager, 
        class jmri.swing.JmriJTablePersistenceManager, 
2  >>>> class jmri.jmrit.roster.RosterConfigManager, 
        class jmri.implementation.FileLocationsPreferences
```
`ProgrammerConfigManager` initializes first (before `RosterConfigManager`)
```
ProgrammerConfigManager ->
        class jmri.jmrix.ConnectionConfigManager, 
>>>>    class jmri.implementation.FileLocationsPreferences
```
It's dependency, `FileLocationsPreferences` calls `FileUtilSupport.setUserFilesPath`. That one fires `PropertyChangeEvent`, but that event is fired in **event dispatch queue**, that is in parallel with the applicaiton initialization, which happens in **main** thread (I printed an extra stacktrace there) in JMRI:
```
	at java.lang.Thread.dumpStack(Thread.java:1336)
	at jmri.jmrit.roster.Roster.readFile(Roster.java:862)
	at jmri.jmrit.roster.Roster.reloadRosterFile(Roster.java:1012)
	at jmri.jmrit.roster.Roster.setRosterLocation(Roster.java:1069)
	at jmri.jmrit.roster.RosterConfigManager.initialize(RosterConfigManager.java:72)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:392)
	at jmri.implementation.JmriConfigurationManager.lambda$initializeProvider$6(JmriConfigurationManager.java:388)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1382)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:387)
	at jmri.implementation.JmriConfigurationManager.lambda$load$1(JmriConfigurationManager.java:205)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1382)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:204)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:192)
	at apps.Apps.<init>(Apps.java:276)
	at apps.PanelPro.PanelPro.<init>(PanelPro.java:40)
	at apps.PanelPro.PanelPro.main(PanelPro.java:120)
```

Since `RosterConfigManager` requires `FileLocationsPreferences`, the FileLocationsPreferences initialization will always precede `RosterConfigManager` and will enqueue the `PropertyChangeEvent` firing process in the GUI thread. In an **unfortunate timing**, the logged sequence is like the following.

I have annotated the log with comments, but the parts were not reordered, just interrupted by commentary paragraphs. You can download the log (without comments) for reference from https://www.dropbox.com/s/k30kat8u34p8baj/roster2.log?dl=0

```
2020-04-19 23:44:58 DEBUG JmriConfigurationManager:385 - Initializing provider class apps.StartupActionsManager
[... 116 lines cut ...]
2020-04-19 23:44:59 DEBUG XmlFile:112 - readFile: /home/sdedic/.jmri/Odbocna.jmri/roster.xml from /home/sdedic/.jmri/Odbocna.jmri/roster.xml
2020-04-19 23:44:59 DEBUG XmlFile:142 - reading xml from file: /home/sdedic/.jmri/Odbocna.jmri/roster.xml
2020-04-19 23:44:59 DEBUG RosterConfigManager:208 - Registered roster 488778885 for profile Odbocna - offline
```

While the `PropertyChangeEvent` handling terminates, the original non-GUI thread continues to load roster entries:

```
2020-04-19 23:44:59 DEBUG Roster:875 - readFile sees 13 children
2020-04-19 23:44:59 DEBUG RosterEntry:650 - ctor from element [Element: <locomotive/>]
2020-04-19 23:44:59 DEBUG Roster:225 - Add entry [RosterEntry: Bardotka Krouzek Bardotka_Krouzek.xml      12  MX630 version 37+ Zimo Unified software (version 37+) ] to roster id: 1430640484
2020-04-19 23:44:59 DEBUG RosterEntry:650 - ctor from element [Element: <locomotive/>]
2020-04-19 23:44:59 DEBUG Roster:225 - Add entry [RosterEntry: Bardotka prototip Bardotka_prototip.xml    Lípa  14  MX645 version 37+ Zimo Unified software (version 37+) ] to roster id: 1430640484
2020-04-19 23:44:59 DEBUG RosterEntry:650 - ctor from element [Element: <locomotive/>]
```
Note the logged `System.identityHashcode` at the end, it belongs to the first `Roster` instance, now replaced in the `RosterConfigManager` 
internal map for id `1430640484`

The other issue is that although apparently the `RosterConfigManager.getRoster` happens concurrently 2x, the profile to roster Map is  not synchronized in anyway, so it might get corrupted eventually; but even without that the *populated roster gets de-registered* from the profile, replaced by a Roster that never loade its contents.

Sources that were used to produce the logs: https://github.com/svatoun/JMRI/tree/bug-testing/roster-initialization.

Commit that specifically adds logging: https://github.com/svatoun/JMRI/commit/a685906251d4591a03ca7b9d5790fead79d4790c (ignore that replaceEntry in Roster, not cleaned up)

The supposed fix for `RosterConfigManager` should be simple to trivial; but the principle that `FileUtilSupport` fires events in GUI thread while JMRI is initializing on the background may break more classes or singletons through their Listeners.

rhwood, 2020-04-19:
FileUtilSupport is firing its events on the "Event Dispatch Thread" (EDT) (Oracle's name for that thread, not ours). Are you proposing that the EDT should not be used for firing events?

Pugwash1, 2020-04-20:
This may be another manifestation of #8378

rhwood, 2020-04-20:
@svatoun I'm not opposed to not firing property change events on which ever thread triggers them, it's just that JMRI has a number of listeners (not necessarily for the FileUtilSupport events, but in general) that assume they are being notified on the EDT and we've had lots of strange crashes as a result.

I've been noodling with making it the listener's responsibility to act on the correct thread instead of the firer's responsibility to fire on the correct thread for a listener. Is that a cleaner design principal than how we are doing things currently? 

@Pugwash1 It's not; there is an older problem here that switching threads to fire property change events "solved", but probably incorrectly.

svatoun, 2020-04-20:
> ... on the "Event Dispatch Thread" (EDT) (Oracle's name for that thread, not ours). 

AFAIK, the term "Event Dispatch Thread" has been coined even back in SUN Microsystems era, before Oracle acquired it. In even more ancient times, we informally called that AWT thread at SUN.

> Are you proposing that the EDT should not be used for firing events?

That would be pretty strange proposal, given that all UI events are fired in that thread, wouldn't it ? 

But indeed, rescheduling all events from such base class as `jmri.beans.bean` into EDT may be a challenging decision indeed. Definitely can be dangerous for Beans that *normally* operate in non-EDT context. But maybe all `jmri.beans.Beans` operations are confined to EDT by design (didn't find directly in Bean class' javadoc - I really don't know)

IMHO, all `jmri.bean. Bean`s that fire an event during the startup sequence may (theoretically) create such race conditions in their Listeners. The point is that, even if the specific Bean is confined to EDT in *normal operation*, it processes the listeners asynchronously *during startup*. 

The mentioned trivial fix is to protect just `RosterConfigManager.getRoster` from both reentrancy and concurrency; that *could* be sufficient for the Roster instance duplicate creation, or at least ensure that the initialized Roster instance will prevail. I can provide a patch for that - but that patch would by no means attempt solve other potential issues.

---
> Is that a cleaner design principal than how we are doing things currently?

General answer (probably not useful; and probably long known here - sorry): **each approach** has its exclusive advantages:
1, events dispatched synchronously retain high concurrency. IMHO does not apply in JMRI.
2. events serialized in a thread give a nice property of coming to listeners in *the same order* as the (safe, locked) operation performed the changes. Makes listener's life much simpler.
3. events offloaded to a queue, and processed "later" (such as with `invokeLater`) give another important property: a listener can't just intervene in the middle of the operation, after the first of several changes is made. Multi-change (just) appears as atomic to the listener and can read "whole" change after. Extremely nice for listener implementors AND for event sources: prevents accidental reentrancy, partial state observation etc.
4. if the listener has a different threading policy than the event source, it needs to adapt the event stream to its own requirements :-/ anyway: the event source can't handle all the different strategies.

I don't know enough about `jmri.beans.Bean` & co. design constraints; but if Bean is expected to be used **just** from EDT, then it's the case (3) and serves that function great

End general answer

---

I don't want to mess into that broader scope (given the generally unfriendly attitude for contributors 'outside the pack') - unless directly hit by another blocking bug. 

But I was curious, so I just made another dump and prepared a list - I suspsect not exhaustive. I also don't know which classes MAY have listeners attached during startup: because of varying startup initialization sequence (controlled by InitializationManager), more elaborate logging would be needed to cover _possible_ listener attachments. Here's a list of even sources before the `AppBase` declares the initialization complete (just one startup attempt):
```
at jmri.util.FileUtilSupport.setProgramPath(FileUtilSupport.java:986)
at jmri.profile.ProfileManager.setActiveProfile(ProfileManager.java:213)
at jmri.util.FileUtilSupport.setUserFilesPath(FileUtilSupport.java:762)
at jmri.util.FileUtilSupport.setScriptsPath(FileUtilSupport.java:1065)
at jmri.jmrix.SystemConnectionMemoManager.register(SystemConnectionMemoManager.java:55)
at apps.startup.StartupActionModelUtil.addAction(StartupActionModelUtil.java:126)
at jmri.jmrix.ConnectionConfigManager.add(ConnectionConfigManager.java:193)
at jmri.jmrit.roster.RosterConfigManager.setDefaultOwner(RosterConfigManager.java:136)
at jmri.jmrit.roster.RosterConfigManager.setDirectory(RosterConfigManager.java:193)
```
The startup sequence can be seen as an **exception** from the normal course of events: 
I can imagine, given the relative small number of events involved, that those would be just *queued* during the startup sequence, and released once the asynchronous startup part is over, to distribute changes in "normal" operating mode (= EDT). 

Not sure that this postponing wouldn't break something else. Also can't tell if such logic should be built into `Bean` itself, or some `StartupSafeBean` should be invented for that. There are surely many other possible approaches.

.... sigh; every time I try to write something, it looks more like a novel, than a post. Sorry.

rhwood, 2020-04-21:
We have a design intention to have a "layout" thread that is separate from the EDT, although currently we keep the two conflated (the desire is that EDT activity *not* interrupt the layout and will be allowed to eventually catch up).

The original intent in having `jmri.beans.Bean` fire events on the EDT was that Swing objects that are updating the GUI have been historically (in this project) poorly written to handle responding to an event fired on a different thread, and eventually it just became easier to have everything fire on the EDT than deal with apparently minor changes that cause crashes in specific contexts, scenarios, or environments because a Swing object got redrawn off the EDT.

So, I think for 4.19.x (leading to 4.20 this summer), a quick fix will be good to have, but a more systemic fix will have to wait until 4.21.1 is released (when we have a few months to find the worst offenders for not handling an event fired on a thread other than the EDT).

svatoun, 2020-04-21:
> We have a design intention to have a "layout" thread that is separate
Thanks - I know about that one, it's well described. What I don't know in this case is whether all `Bean`s are "layout objects", "UI objects", or designed to serve as a basis for both. Documentation says "use common sense" for interaction - but there are Beans:
- GuiLafPreferencesManager (almost 100% UI object)
- SignalSpeedMap (I guess layout object)
- WebAppManager (completely other lifecycle)
and others, like FileUtilSupport may fall into several categories. On the other hand, scheduling outgoing interactions to EDT would suggest just UI objects.

Without extra synchronization, accesses from other threads (i.e.a Layout object observed from EDT propertyChage) will be just "weakly" consistent: can well see atomic writes. but:
- a repeated getFoo() in a sequential code may yield different values, which is unexpected; c,f. storing getFoo() into a local
- readout from two or more fields may give inconsistent data, even such thingsk and `obj.getCount` and `obj.getItem(x)`
Good, bad - it depends :) but it's not safe to "just" write UI as if single-threaded.

svatoun, 2020-04-21:
@rhwood Sorry, I've didn't notice your change. That branch in my fork also contains a fix for `GuiLafPreferencesManager` which can eventually throw `ConcurrentModificationException` or fire events off EDT when applying L&F. It's ridiculous, but the Java's `UIDefaults` code uses `SwingPropertyChangeSupport` support to schedule events to EDT, but simply initializes it to be synchronous -- see `UIDefaults.addPropertyChangeListener` ...

PUlvestad, 2020-04-22:
Is this causing issue #8144

svatoun, 2020-04-22:
It might, but can be something different. There's no log (IMHO) that you could enable in the relevant code.
If you can patch the code, put something like
```
        log.info("Roster constructed: {}", System.identityHashCode(this));
        log.info("Created from", new Throwable());
```
into `Roster()` constructor. If you see two "Roster constructed" (should be different numbers after) after startup, the chances are better. If there stacktraces will be similar to the above, then it's almost sure.

closed 2020-04-26 by bobjacobsen

mstevetodd, 2020-04-29:
I can report that I have not seen a recurrence of this "empty Roster" issue since this was merged. Thanks!

PR #8437 (2020-04-26): fix: Beans may fire events on or off EDT
- `jmri.beans.Bean` uses SwingPropertyChangeSupport to allow a Bean to either always fire events on the EDT or to fire events on the thread that generated the event and have constructors to allow subclasses to set that
- `Bean` inherits from `PropertyChangeFirer` to avoid duplicating Javadocs
- add method to Bean to test if Bean is (not) firing events on EDT
- `jmri.util.FileUtilSupport` fires events on generating thread
- `PreferencesBean` fires events on generating thread
- `PreferencesBean` only sets the dirty property if a change really occurs

Fixe [... cut]
commit eba74c46 (2020-04-26): fix: Beans may fire events on or off EDT

- `jmri.beans.Bean` use SwingPropertyChangeSupport to allow a Bean to either always fire events on the EDT or to fire events on the thread that generated the event and have constructors to allow subclasses to set that
- `Bean` inherits from `PropertyChangeFirer` to avoid duplicating Javadocs
- add method to Bean to test if Bean is (not) firing events on EDT
- `jmri.util.FileUtilSupport` fires events on generating thread
- `PreferencesBean` fires events on generating thread
- `PreferencesBean` only sets the dirty property if a change really occurs

Fixe [... cut]