JMRI/JMRI #8390: Jenkins Builds job failing consistently

bobjacobsen, 2020-04-19:
The last few Builds jobs on Jenkins have failed with
```
     [java] 
     [java] Failures (1):
     [java]   JUnit Vintage:EngineSoundTest:testCTor
     [java]     MethodSource [className = 'jmri.jmrit.vsdecoder.EngineSoundTest', methodName = 'testCTor', methodParameterTypes = '']
     [java]     => java.lang.AssertionError: Unexpected ERROR or higher messages emitted: Test jmri.jmrit.vsdecoder.EngineSoundTest left ShutDownTask registered: Writing Blocks (of type class jmri.BlockManager$1)}
     [java]        org.junit.Assert.fail(Assert.java:88)
     [java]        org.junit.Assert.assertTrue(Assert.java:41)
     [java]        org.junit.Assert.assertFalse(Assert.java:64)
     [java]        jmri.util.JUnitUtil.tearDown(JUnitUtil.java:332)
     [java]        jmri.jmrit.vsdecoder.EngineSoundTest.tearDown(EngineSoundTest.java:31)
     [java]        sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
     [java]        sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
     [java]        sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
     [java]        java.lang.reflect.Method.invoke(Method.java:498)
     [java]        org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
     [java]        [...]
     [java] 
     [java] Test run finished after 1397720 ms
     [java] [      6451 containers found      ]
     [java] [         6 containers skipped    ]
     [java] [      6445 containers started    ]
     [java] [         5 containers aborted    ]
     [java] [      6440 containers successful ]
     [java] [         0 containers failed     ]
     [java] [     25759 tests found           ]
     [java] [       306 tests skipped         ]
     [java] [     25434 tests started         ]
     [java] [      3124 tests aborted         ]
     [java] [     22309 tests successful      ]
     [java] [         1 tests failed          ]
```

rhwood, 2020-04-20:
How is that being invoked (i.e. how is Jenkins running its tests that this slipped past the AppVeyor and Travis-CI tests)?

pabender, 2020-04-20:
The builds job is calling ant with the following targets:

realclean warnings tests-warnings logheadlesstest checkPropertiesFiles javadoc

This should be calling fewer tests than are called on appveyor.

closed 2020-05-28 by bobjacobsen