JMRI/JMRI #8391: Unexpected NPE when starting Loconet simulator

crosland, 2020-04-20:
I don't know if this is a genuine issue as I am not generally a Digitrax user.

cd D:\JMRI; "JAVA_HOME=C:\\Program Files\\Java\\jdk1.8.0_231" cmd /c "\"\"C:\\Program Files\\NetBeans-11.2\\netbeans\\java\\maven\\bin\\mvn.cmd\" -Dexec.args=\"-classpath %classpath apps.PanelPro.PanelPro\" -Dexec.executable=\"C:\\Program Files\\Java\\jdk1.8.0_231\\bin\\java.exe\" -Dmaven.ext.class.path=\"C:\\Program Files\\NetBeans-11.2\\netbeans\\java\\maven-nblib\\netbeans-eventspy.jar\" -Dfile.encoding=UTF-8 process-classes org.codehaus.mojo:exec-maven-plugin:1.5.0:exec\""
Scanning for projects...
                                                                        
------------------------------------------------------------------------
Building JMRI 4.19.5-SNAPSHOT
------------------------------------------------------------------------

--- build-helper-maven-plugin:3.0.0:timestamp-property (timestamp-property) @ jmri ---

--- maven-enforcer-plugin:3.0.0-M2:enforce (enforce-maven) @ jmri ---

--- jacoco-maven-plugin:0.8.5:prepare-agent (prepare-agent) @ jmri ---
surefire.argLine set to -javaagent:C:\\Users\\Daddy\\.m2\\repository\\org\\jacoco\\org.jacoco.agent\\0.8.5\\org.jacoco.agent-0.8.5-runtime.jar=destfile=D:\\JMRI\\target\\jacoco-unit.exec,append=true

--- jacoco-maven-plugin:0.8.5:prepare-agent (prepare-agent-integration) @ jmri ---
failsafe.argLine set to -javaagent:C:\\Users\\Daddy\\.m2\\repository\\org\\jacoco\\org.jacoco.agent\\0.8.5\\org.jacoco.agent-0.8.5-runtime.jar=destfile=D:\\JMRI\\target\\jacoco-it.exec,append=true

--- javacc-maven-plugin:2.6:jjtree-javacc (default) @ jmri ---
Skipping - all parsers are up to date

--- buildnumber-maven-plugin:1.4:create-metadata (generate) @ jmri ---
ShortRevision tag detected. The value is '7'.
Executing: cmd.exe /X /C "git rev-parse --verify --short=7 HEAD"
Working directory: D:\JMRI

--- maven-resources-plugin:2.6:resources (default-resources) @ jmri ---
Using 'UTF-8' encoding to copy filtered resources.
Copying 2378 resources
Copying 1 resource

--- maven-compiler-plugin:3.8.1:compile (default-compile) @ jmri ---
Nothing to compile - all classes are up to date

--- exec-maven-plugin:1.5.0:exec (default-cli) @ jmri ---
2020-04-20 20:24:59,962 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-04-20 20:25:00,570 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Daddy\JMRI\log\messages.log [main]
2020-04-20 20:25:00,571 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Daddy\JMRI\log\session.log [main]
2020-04-20 20:25:00,617 apps.Apps                             INFO  - PanelPro version 4.19.5ish+Daddy+2020-04-20T19:24:28Z+R50ac6a7 starts under Java 1.8.0_231 on Windows 10 amd64 v10.0 at Mon Apr 20 20:25:00 BST 2020 [main]
2020-04-20 20:25:09,294 apps.Apps                             INFO  - Starting with profile Digitrax_simulator.3cd06812 [main]
2020-04-20 20:25:15,813 node.NodeIdentity                     INFO  - Using 56402b6d-a3c8-4947-997e-a70150b1f006 as the JMRI storage identity for profile id 3cd06812 [AWT-EventQueue-0]
2020-04-20 20:25:25,995 loconet.LnPacketizer                  WARN  - passing to xmit: unexpected exception:  [LnPowerManager LnTrackStatusUpdateThread]
java.lang.NullPointerException
	at jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage(LnPacketizer.java:104)
	at jmri.jmrix.loconet.LnPowerManager$LnTrackStatusUpdateThread.run(LnPowerManager.java:208)
2020-04-20 20:25:27,505 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2020-04-20 20:25:27,507 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
2020-04-20 20:25:29,452 swing.JmriJTablePersistenceManager    INFO  - Table preferences not found.
This is expected on the first time the "Digitrax simulator" profile is used on this computer. [main]
2020-04-20 20:25:29,462 util.FileUtilSupport                  INFO  - File path program: is D:\JMRI\ [main]
2020-04-20 20:25:29,463 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Daddy\JMRI\Digitrax_simulator.jmri\ [main]
2020-04-20 20:25:29,463 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Daddy\JMRI\Digitrax_simulator.jmri\ [main]
2020-04-20 20:25:29,464 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Daddy\JMRI\ [main]
2020-04-20 20:25:29,464 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Daddy\ [main]
2020-04-20 20:25:29,464 util.FileUtilSupport                  INFO  - File path scripts: is D:\JMRI\jython\ [main]
2020-04-20 20:25:36,822 PanelPro.PanelPro                     INFO  - Main initialization done [main]

devel-bobm, 2020-04-20:
This NPE can happen if running on a slow machine.  It does not happen for me when running code pulled 
from HEAD of master yesterday in my NetBeans 11 environment.

LocoNet Power Manager posts a LocoNet message for transmit after a short delay after it is initialized.  
If the LocoNet Packetizer has not been fully initialized by that time, an NPE results in the packetizer.  
The delay has been tweaked a number of times because of slow machines.  The delay is currently 500mSec 
_after_ the LnPowerManager is initialized.  I do not know exactly how the packetizer and/or 
LnPowerManager are initialized without digging into code, so cannot comment off-the-cuff.

I looked back towards the beginning of March and did not see any changes to the LnPowerManager or the 
LnPacketizer.  So I don't think that any change to LocoNet code is the source of your problem.

Perhaps something has changed in the infrastructure, like the sequence of initialization of the Managers, 
that makes this a problem for you.  Or perhaps something in a panel .XML file that is auto-loaded at start?

rhwood, 2020-04-20:
Do you see this NPE only from with a maven execution, or do you also see it when run with ant?

crosland, 2020-04-21:
It happens with both Maven and Ant.

It's not a real issue for me as I am only using the simulator as a means to an end for testing a tool that is not (yet) supported by a SPROG connection.

crosland, 2020-04-29:
I no longer see this on my new machine, so the comment about slow machines seesm to be true

closed 2020-04-29 by crosland