JMRI/JMRI #8397: Reloading of Jython Modules

colinmhoward, 2020-04-21:
I have been trying to develop Jython scripts to run my layout.
I am a software engineer by trade, and tried to create different modules containing seperated classes for controlling different elements of the layout.
However, after a few frustrating hours, I think I spotted that if you split your Jython classes into seperate modules (files) - those files are not re-loaded (or re-compiled into Java), until you start and stop PanelPro again.
So it seems that the only way to get around this is if you are in a "development phase" of your scripts / classes, you need to include the new classes into the top level script, otherwise changes won't be picked up in related files until you stop and start PanelPro again.
Please can someone correct me if I am wrong, or if I have missed away of getting the seperate modules (or files) to be picked up by the Jython environment.
Thanks in advance.

bobjacobsen, 2020-04-21:
The expected behavior:

- The act of reading/loading a .py file causes it to be executed.
- When it's executed, whatever it defines is defined; if it existed earlier, it's redefined so that later executions of a procedure, later creation of objects (including listeners), threads launched later will get those redefinitions.
 - But anything done earlier with the original definitions is still there:  The previous objects will have the same methods, the existing threads are still running the code in their object, etc.

Is this what you're seeing?  I'm not sure what you mean about the top-level script.

colinmhoward, 2020-04-21:
Hi - thanks for your reply.
Sorry for any confusion regarding the top-level script.
What I have done is create a .py script for a locomotive which contains a class with methods to perform operations on it. Let's call that loco.py containing a class Loco.
I then have a main .py script which runs the layout and imports loco to create a Loco object - let's call that layout.py. This is the one I load into the Script Entry window to exececute.
While debugging and developing I obviously want to be able to change the loco.py script.
What happens is that if I modify the loco.py file, the changes are not taken until I re-start Panel Pro - I can actually see it regegenerate a file called loco.py$.class. But it never gets regenerated after that point.
This is probably because it is not the main layout.py script the is executed - it feels as if only changes to that are detected once it is all up an running. Not anything it depends on.
The only way I have found to get around this during development is to write the Loco class within the layout.py file, then only when I am happy with it, to copy this into its own loco.py script file.
I would rather structure my scripts this way, to stop me having a monololithic script file. It also allows me to re-use scripts / classes later on.
Hope this makes more sense.
Thanks, Colin

bobjacobsen, 2020-04-21:
If instead of importing (which I agree is idempotent), try doing `execfile(..)` to rerun your loco.py file.

colinmhoward, 2020-04-21:
Oh ok - will try and look at that tomorrow. Thanks for that - and will let you know.
Thanks again :)

rhwood, 2020-04-21:
JMRI provides a single Jython interpreter and executes every python script inside that interpreter. Once a module is imported into that interpreter it is not reloaded unless you explicitly call `reload(name-of-module)`. I believe that is consistent with standard python behavior per [this SE answer](https://stackoverflow.com/a/4111670).

If `reload(...)` does not work (I have not tested myself), `execfile(...)` will work.

colinmhoward, 2020-04-22:
Thanks both for your quick reply and comments. FYI reload did not seem to work, but execfile worked like a charm. Thanks again :)

closed 2020-05-30 by bobjacobsen