JMRI/JMRI #8398: Scaling of Track Current Meter is Broken

crosland, 2020-04-21:
Does anyone else see this?

The most fundamental issue is that the icons are not scaled to fit the last saved window size. They are initially drawn scaled 1:1, which means the display may not fit the window:

[image]

The window size is then used as the base size when resizing so trying to resize the window can make things worse.

I am working in a fix, but it's not as straightforward as I expected, based on what fixed the NixieClock, as the scaling was completely re-written.

I have got the initial display correct but resizing the window still causes the scaling to go wrong.

KenC57, 2020-04-21:
The clocks were some rather early code. I suspect there were things done
then that much better ways have come with the later Java versions and just
better ideas. LCD clock was based on Nixie. I had thought of making a flip
clock by adding animated graphics for the digit transitions, but that was
many years ago and I stayed busy with many other things.

crosland, 2020-04-22:
The startHeight and startWidth variables do not get updated after a resize so subsequent resizing is based on the original window size and does not work as expected.

I attempted to update startHeight and startWidth each time the window is resized. The problem is that componentResized() is called many times whilst the window is being resized. As the resize operation finishes the user is moving the mouse more slowly, the old and new window sizes are very close and the scale becomes close to 1.0, so the icons are still drawn incorrectly.

It seems to me that the resizing is fundamentally flawed. @pabender could you comment, as you modified the resizing?

My only idea is to revert to the original scheme where the scale is calculated from the icon size and the window size each time componentResize() is called.

KenC57, 2020-04-22:
I think the problem is the pane should sense it is being resized and wait
for the value to remain steady for some timeout, say 300mS or greater, to
save trying to do all that work until the drag has finished. I suspect the
UI may even have an option about 'still moving' to consider. I think this is
true of most of the panes, they could use a wait until it stops moving
feature.

Watching the LCD clock is great if you move the corner fast and make a large
change. It sits and flashes for quite a while before it catches up.

crosland, 2020-04-23:
Seems I was trying to hard to fix this :) It was easy in the end. Fixed in PR #8379 which is running CI at the moment.

closed 2020-04-23 by crosland