JMRI/JMRI #8469: DecoderPro: Additional column for cvpane feature request

timtashpulatov, 2020-05-01:
It could be beneficial to add another column to CV pane, making the list less cryptic.

For example, a column next to CV number could show variable item or label from decoder definition file:

CV | Description | Value (Decimal)
----|-------|---
66 | Forward Trim|0

If tooltips could also be fetched that'd also be great.

(Oh and being able to switch Value display from Decimal to Hex or Binary would also be handy)

bobjacobsen, 2020-05-01:
Thanks!

Can you suggest a way to handle CVs with multiple variables, like CV29?  Maybe a list of variables, or is there a better idea?

Also, a variable can be in multiple CVs, like the long address. Maybe mark that with an asterisk or some other way of indicating "This isn't all there is"?

Not quite sure what you mean by tooltips.  Do you mean to show them in another column?

Thanks again.

timtashpulatov, 2020-05-01:
Bob, thanks for prompt reply.

I don't think something special must be done for multi-field and multi-byte CVs at all. The idea is to help user recognize what this particular CV is responsible for. So perhaps having something like 'CV29, Configuration' will be just enough.

(TBH, I cannot recall now why I mentioned the tooltips. Maybe that's because I was editing my own decoder definition file way too long)

danielb987, 2020-05-02:
> Can you suggest a way to handle CVs with multiple variables, like CV29? Maybe a list of variables, or is there a better idea?

As a developer, I would prefer a list like this, but I don't know if that's a good idea for a user:
Bit 0: Analog On/Off
Bit 1,2: Speed setting
Bit 3-5: Something else

> Also, a variable can be in multiple CVs, like the long address. Maybe mark that with an asterisk or some other way of indicating "This isn't all there is"?

The string could begin with "Part of", for example "Part of loco address".

dheap, 2020-05-02:
This is getting further away from the concept of DecoderPro being about settings, not CVs.

We already have the opposite; Display CV numbers and Bits in tooltips for Variables (I wrote that code).

Trying to do the opposite is fraught with problems; Variable names are often not user-friendly, they may of necessity be displayed with different labels in different panes or need to be overriden in panes. To complicate things further, runtime qualifiers sometimes change variables attached to a particular CV, based on the value of other CV(s).

Giving the average user tools to tempt them to manipulate CVs and bits directly in the CVs pane rather than using panes seems to me to be encouraging bad practice, going against the DecoderPro philosophy and possibly creating a support nightmare.

timtashpulatov, 2020-05-02:
Dave,

I totally agree the settings panes are great for user dealing with settings.

My point was to see human friendly CV name in the pane, nothing more. Consider my case, I am tuning a particular decoder & configuration, and often have to read full list of CVs from the decoder. 
I don't know in advance which CVs in the decoder are different from the defaults in the definition file, so I stick to CV pane. Now I see the CV#65 has changed, but what exactly it is for? I have to flip through all panes then to find this out.

dheap, 2020-05-02:
There's a quite-possible one-to-many mapping of CVs (or parts thereof) to variables and a lot of factors in play (such as runtime qualification). Doing this (and displaying it concisely) would be a nightmare in my estimation.

As I mentioned earlier, I coded the Variable-to-CV mapping for the tooltips and (complex as it may be) it was a relatively simple operation interrogating only the single current variable.

Doing it the other way involves searching the Variables Table, interrogating each variable for its CV list and searching that, determining the current label being displayed and qualification state, etc. I wouldn't like to tackle it.

timtashpulatov, 2020-05-02:
> Doing it the other way involves searching the Variables Table, interrogating each variable for its CV list and searching that, determining the current label being displayed and qualification state, etc.

From what I understand, there is already a CV list in `cvtable` (looking at xml\programmers\parts\CVsPane.xml). Not knowing how the `cvtable` is compiled, I thought it would be sufficient to take the `item` value at least for simple (unambigous) variables:

`<variable CV="2" item="Vstart" comment="Range 0-255" default="0">`

dheap, 2020-05-02:
The CV list doesn't contain any reference to Variables.

Some modern decoders have over 2,000 CVs and around 10,000 Variables. You have to search the Variable Table and interrogate each variable for its component Cvs.

timtashpulatov, 2020-05-02:
> The CV list doesn't contain any reference to Variables.

I did not say it does. However the `cvtable` manages to also have a list of values, which sometimes are get calculated from multiple entries for the same CVs sharing different bits. Which made me think of tweaking the code that builds `cvtable` to fetch item name would be feasible.

dheap, 2020-05-02:
The CVTable is simply a HashMap of CV names and values.

dheap, 2020-05-02:
CV names map to possibly many items. That involves searching the Variable Table for each CV.

dheap, 2020-05-02:
Item names are often not user-friendly. For example "Motor Option 1". You have to map to current "label".

timtashpulatov, 2020-05-02:
> CV names map to possibly many items. That involves searching the Variable Table for each CV.

I see, isn't this search performed anyway when building the `VariableTableModel` (?) and later creating `cvtable` from it?

dheap, 2020-05-02:
No. The `VariableTableModel` is processed once, one item at a time. The CV table is built up in this pass as each Variable is processed, new entries being created as needed. The CV table doesn't currently include provision for building up lists of Item names, one for each CV table entry. I think offhand that this would involve changes to a lot of implementations of VariableValue.

I've worked on a fair bit of code in this area, particularly with multi-CV variables. I'm not about to volunteer to implement this. I think I've contributed enough to this discussion.

silverailscolo, 2021-01-05:
Closing this issue +8 mo. because of 1-to-many relationship as explained above.
My trick to find uses of a CV is manually change the value in the CV table and look for orange fields on the settings panes (WARN restore the original value before any write!)

closed 2021-01-05 by silverailscolo