JMRI/JMRI #8471: Coverage Report for Single Tests.

dheap, 2020-05-02:
Since the merge of #8457 this works for a selected file or package.

But there's a slightly-annoying issue.

The merge of `jacoco-unit.exec` and `jacoco.exec` occurs _before_ the selected test/package is run. As a result, you have to run the test twice to see any result, initially or for change of code.
[Merge before tests.txt](https://github.com/JMRI/JMRI/files/4566879/Merge.before.tests.txt)

Likewise, there's no merge the first time after a `realclean` .
[Merge skip after realclean.txt](https://github.com/JMRI/JMRI/files/4566882/Merge.skip.after.realclean.txt)

pabender, 2020-05-02:
The merge is never invoked in the output you provided.

It needs to be invoked before the report.

dheap, 2020-05-02:
@pabender 
> The merge is never invoked in the output you provided.

The merge is invoked at lines 26-29 of the [Merge before tests.txt](https://github.com/JMRI/JMRI/files/4566879/Merge.before.tests.txt) file I provided, but before the test(s) (lines 51-60) and report (lines 63-65) have run.

The merge is skipped at lines 178-179 of the [Merge skip after realclean.txt](https://github.com/JMRI/JMRI/files/4566882/Merge.skip.after.realclean.txt) file I provided, again before the test(s) (lines 205-214) and report (lines 217-218) have run. (Thats understandable as there is no `jacoco.exec` at that point in time.)

 > It needs to be invoked before the report.

- The merge is already always being invoked/skipped before _both_ the tests _and_ the report. Hence you get the results of all previous tests, but not the latest test.
- It needs to be run _after_ the tests and _before_ the report. That may be more difficult.

However, you can workaround the problem by employing a "sacrificial"  run:
1. For a single file, simply run it twice after _any_ change.
2. For a time-consuming package test, run the package test then a quick single-file test. You'll see the results of the package test but not of the single-file test.

pabender, 2020-05-02:
@dheap 

I was refering to

`cd /Users/heap/jmri; JAVA_HOME=/Library/Java/JavaVirtualMachines/jdk1.8.0_241.jdk/Contents/Home "/Applications/NetBeans/Apache NetBeans 11.3.app/Contents/Resources/NetBeans/netbeans/java/maven/bin/mvn" -Dtest=jmri.implementation.AccessoryOpsModeProgrammerFacadeTest test jacoco:report`

For the changes in #8457 to be involved, that needs to read

`cd /Users/heap/jmri; JAVA_HOME=/Library/Java/JavaVirtualMachines/jdk1.8.0_241.jdk/Contents/Home "/Applications/NetBeans/Apache NetBeans 11.3.app/Contents/Resources/NetBeans/netbeans/java/maven/bin/mvn" -Dtest=jmri.implementation.AccessoryOpsModeProgrammerFacadeTest test jacoco:merge jacoco:report`

i.e. the merge needs to be explicitly invoked on the command line.

svatoun, 2020-05-03:
@dheap FYI - #8464 should (IMHO) introduce an automated merge after unit tests are run. It worked for me for test package as well.

closed 2021-04-14 by stale[bot]