JMRI/JMRI #8482: Document running in a headless environment

jhdulaney, 2020-05-04:
I plan to run decoderpro in a headless environment on a minimal installation.  The device is very resource constrained, and I do not have the storage space for a bunch of X libraries that don't need to be there; there will not be any form of monitor attached to the device, and running a full gui stack is not doable.

mstevetodd, 2020-05-04:
Hi jhdulaney,
I have some (past) experience running JMRI in a true headless environment (no GUI). You've probably noticed that _many_ people use the term "headless" simply to mean "monitor not attached", 
There is a startup script for JMRI called "JmriFaceless" that sets the headless flag and is what I used to run JMRI headless on the early RaspberryPi's.
What JMRI features (exactly) do you plan to use on your device?
Only certain parts of JMRI have been modified/tested to run headless.  WebServer, WiThrottle Server and the LoconetOverTCP server run properly in JmriFaceless. You can load a "panel" file, but not if it includes a LayoutEditor, PanelEditor, ControlPanelEditor or SwitchboardEditor, as those are based on graphical objects.
Those are the ones I have experience with.
--Stevet

jhdulaney, 2020-05-04:
[jmri@jmri JMRI]$ ls
COPYING  DecoderPro  InstallTest  JmriFaceless  PanelPro  SoundPro  default.lcf  help  jmri.conf  jmri.jar  jython  lib  python.properties  resources  web  xml
[jmri@jmri JMRI]$ ./JmriFaceless 
OpenJDK 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
2020-05-04 19:22:26,397 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-05-04 19:22:26,398 util.Log4JUtil                        INFO  - This log is appended to file: /home/jmri/.jmri/log/messages.log [main]
2020-05-04 19:22:26,398 util.Log4JUtil                        INFO  - This log is stored in file: /home/jmri/.jmri/log/session.log [main]
2020-05-04 19:22:26,411 apps.AppsBase                         INFO  - JmriFaceless version 4.19.5+R6ede87265 starts under Java 14 on Linux amd64 v5.6.8-300.fc32.x86_64 at Mon May 04 19:22:26 UTC 2020 [main]
2020-05-04 19:22:26,515 profile.ProfileManager                INFO  - Created default profile "My JMRI Railroad" [main]
2020-05-04 19:22:26,517 apps.AppsBase                         INFO  - Starting with profile My_JMRI_Railroad.3d592b4e [main]
2020-05-04 19:22:26,601 apps.AppsBase                         INFO  - No pre-existing config file found, searched for '/home/jmri/.jmri/My_JMRI_Railroad.jmri/ProfileConfig.xml' [main]
2020-05-04 19:22:26,601 util.FileUtilSupport                  INFO  - File path program: is /home/jmri/JMRI/ [main]
2020-05-04 19:22:26,601 util.FileUtilSupport                  INFO  - File path preference: is /home/jmri/.jmri/My_JMRI_Railroad.jmri/ [main]
2020-05-04 19:22:26,602 util.FileUtilSupport                  INFO  - File path profile: is /home/jmri/.jmri/My_JMRI_Railroad.jmri/ [main]
2020-05-04 19:22:26,602 util.FileUtilSupport                  INFO  - File path settings: is /home/jmri/.jmri/ [main]
2020-05-04 19:22:26,602 util.FileUtilSupport                  INFO  - File path home: is /home/jmri/ [main]
2020-05-04 19:22:26,602 util.FileUtilSupport                  INFO  - File path scripts: is /home/jmri/JMRI/jython/ [main]
2020-05-04 19:22:27,611 managers.DefaultShutDownManager       INFO  - Normal termination complete [Thread-0]

jhdulaney, 2020-05-04:
Yeah, there's a whole lot of 'it's not working' going on here.

mstevetodd, 2020-05-04:
Did you create an appropriate config file? You can pass it as a parm, or use the default name, which should be JmriFacelessConfig3.xml.
This is documented here: https://www.jmri.org/JavaDoc/doc/apps/JmriFaceless.html
--SteveT

jhdulaney, 2020-05-04:
> or it will use the default of JmriFacelessConfig3.xml

It doesn't seem to be dong that bit?

mstevetodd, 2020-05-04:
Been a while since I used it, maybe the default has changed. Did you try the default specified in the log you posted? Or try passing the config as a parm?

Pugwash1, 2020-05-04:
> ---------- Original Message ----------

mine is started by : note all full pathnames... and I use a profile named pivnc
, and new profile would have .jmri appended. Its a directory name. make sure it
start as a regular profile. dont open panels.
/home/jmri/JMRI-Files/JMRI/JmriFaceless
--profile=/home/jmri/JMRI-Files/JMRI-Profiles/pivnc

Steve G

rhwood, 2020-05-04:
Note that unless the profile being used for a headless server is configured to start a server (WiThrottle, web, srcp, it doesn’t matter which), it will exit immediately since there is no main window waiting for a user to close it (which is the only thing stopping the graphical apps from exiting immediately).

jhdulaney, 2020-05-04:
So, back to the title here, where is literally any of this documented?

For that matter, where is the documentation for the config file?

I'm not exactly expecting Django level good docs, but at least better than, say, podman.

rhwood, 2020-05-04:
It's not documented because last time I tried to systemically make running JMRI headless, and fix problems observed or encountered I and my work on that was actively attacked by other developers.

> I plan to run decoderpro in a headless environment on a minimal installation. The device is very resource constrained, and I do not have the storage space for a bunch of X libraries that don't need [...]

This simply is not possible. JMRI will crash on startup if X libraries are not present; X does not need to be running for parts of JMRI to function without a GUI, but JMRI will still create GUI objects, even if it does not attempt to draw them.

pabender, 2020-05-04:
> It's not documented because last time I tried to systemically make running JMRI headless, and fix problems observed or encountered I and my work on that was actively attacked by other developers.
> 

We need to fix that.

If someone broke something by making it a requirement that we start in headed mode, that violates the second part of our “code wins” philosophy

In other words, code wins as long as the code doesn’t cause other parts of the system to stop working.

We used to be able to run on a system that was headless.  We need to fix that.

pabender, 2020-05-04:
> I plan to run decoderpro in a headless environment on a minimal installation. The device is very resource constrained, and I do not have the storage space for a bunch of X libraries that don't need [...]

What are you running on? 

How much space does it have?

bobjacobsen, 2020-05-05:
There’s some related documentation at 

https://www.jmri.org/help/en/manual/DecoderPro/VSD_Headless.shtml

Bob

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

klk32003, 2020-05-05:
> There’s some related documentation at
https://www.jmri.org/help/en/manual/DecoderPro/VSD_Headless.shtml

jhdulaney, 2020-05-10:
> What are you running on?
> How much space does it have?

I'm on a beaglebone black with 4gb flash total and 512mb ram.

> We need to fix that.

Are y'all going to?

bobjacobsen, 2020-05-10:
> Are y'all going to?

You could [start here](https://www.jmri.org/help/en/html/doc/Technical/getgitcode.shtml).

But why is it necessary to run on such a constrained device?  There is a lower limit to hardware capable of running JMRI, and this seems well below it.

closed 2021-04-14 by stale[bot]