JMRI/JMRI #8499: Unable to create turnout from XT0 after PR #8246

dsand47, 2020-05-07:
Changes to `jmri.jmrix.lenz.XNetTurnoutManager` result in several errors during panel loading when using XpressNet Simulator.  I don't know if this occurs with real hardware.
```
     [java] 2020-05-06 21:49:32,267 jmrix.AbstractMRTrafficController     WARN  - notify: During reply dispatch to jmri.jmrix.lenz.XNetTurnoutManager@48a242ce [AWT-EventQueue-0]
     [java] java.lang.IllegalArgumentException: Unable to create turnout from XT0
     [java] 	at jmri.managers.AbstractTurnoutManager.newTurnout(AbstractTurnoutManager.java:105)
     [java] 	at jmri.managers.AbstractTurnoutManager.provideTurnout(AbstractTurnoutManager.java:46)
     [java] 	at jmri.jmrix.lenz.XNetTurnoutManager.forwardMessageToTurnout(XNetTurnoutManager.java:98)
     [java] 	at jmri.jmrix.lenz.XNetTurnoutManager.message(XNetTurnoutManager.java:86)
     [java] 	at jmri.jmrix.lenz.XNetTrafficController.forwardReply(XNetTrafficController.java:123)
     [java] 	at jmri.jmrix.AbstractMRTrafficController.notifyReply(AbstractMRTrafficController.java:284)
     [java] 	at jmri.jmrix.AbstractMRTrafficController$RcvNotifier.run(AbstractMRTrafficController.java:1233)
```
The problem is that XNetTurnoutManager.message() is trying to create the 2nd turnout when the address failed with a -1.  The second turnout is addr+1 which results in XT0.

bobjacobsen, 2020-05-07:
Not an XpressNet user, but it looks like the simulator might think it's enforcing a physical limitation.  Is XT0 valid with a hardware connection?

dsand47, 2020-05-07:
This appears to have something to do Lenz message processing.

Prior to #8246, a turnout number in a message could be 0 or higher.  -1 indicated a bad turnout number.  If the number was valid, one or two turnouts are requested.

As of #8246, the 2nd process was moved outside of the check for a valid ( >= 0 ) turnout number.

There was also other code removed so I don't know the reason for the changes or their impacts.

There are other strange results.  The xml file has XT1 and XT2.  When loaded using 4.19.4, there are no errors and the two turnouts.

If I put in a check for address -1 to prevent the errors, the loading the same panel also ends with XT41 to XT44.

pabender, 2020-05-07:
I’ll look into this after work today.

I didn’t catch this working with real lenz hardware.

pabender, 2020-05-08:
@dsand47 See if #8505 improves things.  It should at least prevent trying to load XT0, which really isn't valid (and never has been).

I don't no what is in your panel that would cause extra turnouts to appear.  If you send me the panel, I can look.

dsand47, 2020-05-08:
Here is the panel file that the user provided.

[Test Track Iain.xml.txt](https://github.com/JMRI/JMRI/files/4599342/Test.Track.Iain.xml.txt)

closed 2020-05-08 by bobjacobsen

pabender, 2020-05-09:
@dsand47 

The extra turnouts are because of the sensors XS81-XS88 that are defined in the panel.

On real hardware, the command station makes a determination of whether or not the address in the feedback request is for a sensor or a turnout module by talking to the hardware on the RS bus.  itself, though the address range is limited for turnouts with feedback  ( this is a hardware restriction imposed by the manufacturer).

The simulator doesn't know anything about the real hardware, so it makes the determination about turnout vs sensor based purely on the address involved.  Basically, any number below the cutoff point for turnouts with feedback is treated as a turnout, and anything above that point is treated as a sensor.

PR #8505 (2020-05-08): XNetTurnout range check;don't suppress invalid;report motion complete
Fix #8499 by applying range check to both turnouts in the pair.

Allow "invalid" feedback (as defined by the protocol) to be reported to the turnout objects (and not be filtered out).

Some cleanup of XNetReply handling for the monitor as related to turnout feedback messages, including making sure the motion complete bit is reported for turnouts with feedback.