JMRI/JMRI #8503: NamedBeanComboBox drops entries when system name matches user name.

dsand47, 2020-05-07:
A recent user group posting reported an issue where the LocoNet sensors were missing from sensor combo boxes.  In this case the LocoNet sensors had matching system and user names.

The problem is in the sort method in NamedBeanComboBox.  
```
        TreeSet<B> set = new TreeSet<>(comparator);
        set.addAll(manager.getNamedBeanSet());
```
I created a test version where the getNamedBeanSet and TreeSet addAll are separate.  The full list was returned by getNameBeanSet but the resulting TreeSet were missing the sensor/turnouts when the system and user names match.

The posting indicated LocoNet, but my testing indicates that it is not connection specific.

rhwood, 2020-05-07:
Then the comparator needs to be fixed. I think multiple components use the same comparator as the combo box.

bobjacobsen, 2020-05-07:
I don't quite understand the problem.  Is it that a particular sensor had both it's user name and system name the same?  That's certainly valid, but a quick inspection of the code makes me think it's ok.  

Or is it that one sensor's user name is the same as some other's system name?  Aside from being incredibly confusing, that fails because the TreeSet can't have two entries with the same sort value.

dsand47, 2020-05-07:
When a sensor or turnout (only those 2 tested so far) has a user name that matches the system name, the object is not included in the NamedBeanComboBox list.

I did another test.  The set created by getNamedBeanSet was imported into an ArrayList.  The list was then sorted using the same comparator.  The list ended up with all entries.  This implies that the comparator is ok.

bobjacobsen, 2020-05-07:
Still trying to understand, sorry.
````
t1 = turnouts.provideTurnout("LT1")
t2 = turnouts.provide("LT2")
```
Those display OK.
```
t2.setUserName("LT2")
```

LT2 is no longer visible?

dsand47, 2020-05-07:
> Still trying to understand, sorry.
> 
> ```
> t1 = turnouts.provideTurnout("LT1")
> t2 = turnouts.provideTurnout("LT2")
> ```
> Those display OK.
> ```
> t2.setUserName("LT2")
> ```
> 
> LT2 is no longer visible?
> ```

Correct, although it is more complicated.  I have put some testing code into NamedBeanComboBox.  It shows the combo box list content, a similar tree content created by adding one row at a time and the content as an ArrayList sees it.

The output after adding LT1 using the Turnout table.
```
INFO  - cmbo = LT1
INFO  -
INFO  - tree = LT1
INFO  -
INFO  - list = LT1
```

The output after at LT2
```
INFO  - cmbo = LT1
INFO  - cmbo = LT2
INFO  -
INFO  - tree = LT1
INFO  - tree = LT2
INFO  -
INFO  - list = LT1
INFO  - list = LT2
```

Now LT2 has been added as a user name for LT2.
```
INFO  - cmbo = LT1
INFO  -
INFO  - tree = LT1
INFO  -
INFO  - list = LT1
INFO  - list = LT2 (LT2)     << user name (system name) display option
```

Change the user name.
```
INFO  - cmbo = LT1
INFO  - cmbo = LT2X (LT2)
INFO  -
INFO  - tree = LT1
INFO  - tree = LT2X (LT2)
INFO  -
INFO  - list = LT1
INFO  - list = LT2X (LT2)
```

An additional point that I discovered is that matching names are not a problem as long as there is no table entry without a user name.  Once an entry without a user name is processed, TreeSet no longer accepts any subsequent entries with matching names.  

The table entry without a user name is a problem for sensors since ISCLOCKRUNNING does not have a user name which means any sensors with matching names that sort after ISCLOCKRUNNING will be dropped.

bobjacobsen, 2020-05-07:
Instead of taking the list through a TreeSet to order it, maybe it just needs to be sorted:

https://docs.oracle.com/javase/8/docs/api/java/util/Arrays.html#sort-T:A-java.util.Comparator-

That preserves equal elements instead of dropping them.

If I read the code properly, the `NamedBeanUserNameComparator` is being used for the that display option.  I think it might have a logical error in it, as it sorts both items on their system names if _either_ (or both) items don't have a user name.  It should probably be "take the user name of the first, or the system if no user name;  take the user name fo the second, or the system name if no user name, and order those"

E.g. imagine

LT1 XYZ
LT2
LT3  FOO 

(LT3 FOO) < (LT1 XYZ)  sorted on user names

(LT2) < (LT3 FOO)   both sorted on system names

(LT1 XYZ) < (LT2(  both sorted on system names

But that's not a unique ordering, and likely to drive the TreeSet (and maybe the sort) crazy

rhwood, 2020-05-08:
The NamedBeanComboBox does not do a list->set->sort->list process. It takes the set of beans provided by the manager and sorts that into a list.

bobjacobsen, 2020-05-08:
> The NamedBeanComboBox does not do a list->set->sort->list process. It takes the set of beans provided by the manager and sorts that into a list.

Not sure what you're saying here.  Below is the `sort` routine from NamedBeanComboBox. It gets a ordered set of beans from their manager and puts it into the (perhaps differently ordered) TreeSet, then remove the excluded beans, then converts the remaining contents into a Vector. 

I think the problem is that the comparator isn't obeying the transitive rules it's required to, so the TreeSet is deciding that it has duplicate entries and discarding them.

```
    private void sort() {
        // use getItemAt instead of getSelectedItem to avoid
        // possibility of creating a NamedBean in this method
        B selectedItem = getItemAt(getSelectedIndex());
        Comparator<B> comparator = new NamedBeanComparator<>();
        if (displayOptions != DisplayOptions.SYSTEMNAME && displayOptions != DisplayOptions.QUOTED_SYSTEMNAME) {
            comparator = new NamedBeanUserNameComparator<>();
        }
        TreeSet<B> set = new TreeSet<>(comparator);
        set.addAll(manager.getNamedBeanSet());
        set.removeAll(excludedItems);
        Vector<B> vector = new Vector<>(set);
        if (allowNull) {
            vector.add(0, null);
        }
        setModel(new DefaultComboBoxModel<>(vector));
        // retain selection
        if (selectedItem == null && userInput != null) {
            setSelectedItemByName(userInput);
        } else {
            setSelectedItem(selectedItem);
        }
    }
```

dsand47, 2020-05-30:
Here is an example of this issue.
[image]
Beans where the system name and user name are the same are dropped from the NamedBeanComboBox if they sort after a bean without a user name.

The Route Add screen shows both the actual list and the combo box list which makes the error more visible.

This is a low priority issue.

dsand47, 2021-03-30:
Still a problem at 4.23.3ish.

[image]

dsand47, 2021-05-14:
Still needs to be fixed.

dsand47, 2022-02-16:
This is still bug.  User group issue https://groups.io/g/jmriusers/message/201185.  Verified at JMRI 4.99.3.

bobjacobsen, 2022-04-20:
It's definitely a problem that the comparator is non-unique in the case that's in the May 17 comment.
```
    @Test
    public void testForUniqueOrdering() {

        // check the ordering of mix of beans with and without user names
        //    IT3 FOO
        //    IT1 XYZ
        //    IT2

        NamedBeanUserNameComparator<Turnout> t = new NamedBeanUserNameComparator<>();

        Turnout it1xyz = InstanceManager.getDefault(TurnoutManager.class).provideTurnout("IT1");
        it1xyz.setUserName("XYZ");
        Turnout it2 = InstanceManager.getDefault(TurnoutManager.class).provideTurnout("IT2");
        Turnout it3foo = InstanceManager.getDefault(TurnoutManager.class).provideTurnout("IT3");
        it3foo.setUserName("FOO");

        // sorting on user names
        Assert.assertEquals("IT1 > IT3", +1, t.compare(it1xyz, it3foo));

        // user names before system names with no user names
        // Assert.assertEquals("IT1 < IT2", -1, t.compare(it1xyz, it2));  // fails
        Assert.assertEquals("IT2 > IT3", +1, t.compare(it2, it3foo));     // passes
    }
```

So how do we want it to behave in the case of a mix beans with and without user names? 
 - Those with user names should be in alpha order and all before those without? That's what's in the test case above.
 - When a bean doesn't have a user name, it should be sorted by system name (i.e. the system names and user names should be interleaved alphabetically)

Right now, it's not consistently doing either of those, and that's causing the GUI problem.

bobjacobsen, 2022-04-20:
To put the question another way, what order do we want for @dsand47's example?

[image]

It isn't clear (at least to me) what order the combo box is in .

closed 2022-04-22 by danielb987