JMRI/JMRI #8514: Improvement to EnumIoNamesNumbers

svatoun, 2020-05-10:
two subtle issues with the `EnumIoNamesNumbers` implementation
1. it uses enum.toString() to get canonical enum name. toString() is overridable. enum.name() (which is final) is usually preferred for this task.
2. although EnumIONames.inputFromString is @Nonnull, if the map does not contain a value (compatibility issue, corrupted file, ...), the result *will* be null. So at least @CheckForNull should be there.

(1) is easy to fix (will proivide patch). 
re.2 - I'd prefer to discuss what's the desired outcome if the value actually in the file does not correspond to any valid value - with respect to most common/desirable error handling in the XML deserializers.

Next, how frequent is the case where 
```
    public static <T extends Enum<T>> T valueOf(Class<T> enumType,
                                                String name) {
```
would fail with `IllegalArgumentException`.

bobjacobsen, 2020-05-10:
Thanks for looking through this in detail!

There's a not-frequently-used error reporting mechanism in the configureXML packages using the ErrorHandler and ErrorMemo classes.  The default produces logging messages with a somewhat message format (not all situations really fit the ErrorMemo model of coming from a known NamedBean), and there's a variant that we use when loading panel files to show a dialog box.  EnumIO could send errors to that.

svatoun, 2020-05-11:
Thanks - the question is, if it is performance-affortable to use `Enum.valueOf` which throws Exception on an unknown string: that is on really malformed XMLs AND on numbers.
Is there an example on how to use the `ErrorMemo` / `ErrorHandler` in similar situations ?

bobjacobsen, 2020-05-11:
I'm not sure performance matters here.  On another project (which used more and larger enums), throwing out of valueOf to process every enum element in old files was an excessive cost (mostly due to the stack walking blowing out caches).  But JMRI doesn't read or write very many enums, and I expect the processing time here won't matter.

It's fine with me to change that implementation. EnumIO is hiding it away from the ConfigureXML classes in any case.

There's an example of how to handle an errors in parsing with ErrorMemo/ErrorHandler in the [attribute-reading convenience methods](https://github.com/JMRI/JMRI/blob/cac19a92b735b149c9de3be76c95eefb28279b1f/java/src/jmri/configurexml/AbstractXmlAdapter.java#L110).

closed 2021-04-14 by stale[bot]