JMRI/JMRI #8521: Accumulating Layout Editor test files

bobjacobsen, 2020-05-11:
I’d like to put together a larger suite of test Layout Editor panel files. I ideally, this would include all those excellent features that have been added:  Multiple windows, rotated and scaled track work with decorations, etc.  

They don’t have to be huge (though that’s OK), but it’s helpful if they push the boundaries of what the Layout Editor can do.

And they have to _work_, in the sense that they don’t have problems in them, so that I can check that their features _still_ work in updated code.  I initially tried some files from the JMRIusers group, but it became way too confusing to sort out the problems in the files from the problems in the code.

If you have something like that, I’d appreciate it if you’d attach it here (might have to zip it or add a .txt extension because GitHub won't allow .xml files to be uploaded).  Thanks in advance!

JLang1951, 2020-05-12:
Here is one that I've been using on my layout for quite a while. This is the simulated version.
Preferences: NCE simulated
                    CMRI: simulated
No errors in system consol.
To initiate panel click on the Turnout Reset Button then click on the Block Reset button. This will set all turnouts to closed and all blocks to inactive and all mainline signal masts should go to clear. Then you can click on the turnouts to see the mast aspects change according to the SML.

Hope this helps
John Lang 

[Jlang_Panel.zip](https://github.com/JMRI/JMRI/files/4616271/Jlang_Panel.zip)

PUlvestad, 2020-05-12:
[MonP Signal Panel (20200112).txt](https://github.com/JMRI/JMRI/files/4617732/MonP.Signal.Panel.20200112.txt)
Panel for the Edmonton Model Railroad Assoc.
Has a couple of turnout error messages but those are something I haven't cleaned out yet.

dsand47, 2020-05-14:
Here is a sample panel with a number of features:

- Two panels with edge connectors
- Double slip
- Turntable
- Crossover
- Level crossing
- A turnout with continuing route set to thrown.
- Signal masts with signal mast logic

The Edit function to set a continuing route is not working.  The logic is in the specific turnout classes but they are not active yet.

The logic for the additional 3 block entries for crossovers is an empty method.  I had to manually add the additional blocks to the xml.

Note:  The second panel might not display properly due to an existing issue in the LE drawing process.  A zoom in and out will force a redraw.

[Sample Layout.xml.txt](https://github.com/JMRI/JMRI/files/4625920/Sample.Layout.xml.txt)

bobjacobsen, 2020-07-25:
Thanks for the help!

closed 2020-07-25 by bobjacobsen

PUlvestad, 2020-07-25:
Bob,
There is a file in the users group files area that was put there for this. Can I delete that file now?
Peter

bobjacobsen, 2020-07-25:
Yes, please do.