JMRI/JMRI #8526: Corrupted Roster.xml file with 4.19.6? It looked to be less frequent with previous 4.19.5.

alain355, 2020-05-12:
Dear Team

Following a "New USER" topic on groups.io, where I posted a bypass solution to regenerate the lost roster (using at that time Beta 4.19.5), when 4.19.6 was released a few days ago, I posted messages #173206, #173208, #173214 and #173217 in the JMRI groups.io newsgroup, with parts of the message.log file, to report a new kind of issue.

I think that sometime when closing JMRI, it creates a nearly empty roster.xml file (not populated at all with the existing entries).

I first saw that issue happened sometime when switching from simulator (no DCC command station) to my ECoS with older beta versions. However with 4.19.6, it happens nearly all the time, and when recreating roster (my bypass is 1st to recreate roster Index, then to update decoder definition), column size is completely lost (it was not the case with 4.19.5!).

Hope that somebody can look at the issue.

Extract from the message.log file where I didn't see any error messages :(
>>>>>>>>>>>>>>>>>>>>

2020-05-10 18:06:17,150 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-05-10 18:06:17,256 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\acara\JMRI\log\messages.log [main]
2020-05-10 18:06:17,259 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\acara\JMRI\log\session.log [main]
2020-05-10 18:06:17,293 apps.AppsBase                         INFO  - DecoderPro version 4.19.6+Rdb3d7c2db starts under Java 1.8.0_251 on Windows 10 amd64 v10.0 at Sun May 10 18:06:17 CEST 2020 [main]
2020-05-10 18:06:21,723 gui3.Apps3                            INFO  - Starting with profile simulation.3f3abea9 [main]
2020-05-10 18:06:23,291 node.NodeIdentity                     INFO  - Using b9962ec7-75fe-4571-9857-29c88903aad8 as the JMRI storage identity for profile id 3f3abea9 [AWT-EventQueue-0]
2020-05-10 18:06:33,709 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-05-10 18:06:33,710 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\acara\JMRI\Mon_Réseau_JMRI.jmri\ECOS\ [main]
2020-05-10 18:06:33,712 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\acara\JMRI\simulation.jmri\ [main]
2020-05-10 18:06:33,714 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\acara\JMRI\ [main]
2020-05-10 18:06:33,716 util.FileUtilSupport                  INFO  - File path home: is C:\Users\acara\ [main]
2020-05-10 18:06:33,719 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2020-05-10 18:06:59,116 decoderdefn.DecoderIndexFile          INFO  - update decoder index [initialize decoder index]
2020-05-10 18:10:21,330 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
2020-05-10 18:10:36,603 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-05-10 18:10:36,667 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\acara\JMRI\log\messages.log [main]
2020-05-10 18:10:36,670 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\acara\JMRI\log\session.log [main]
2020-05-10 18:10:36,707 apps.AppsBase                         INFO  - DecoderPro version 4.19.6+Rdb3d7c2db starts under Java 1.8.0_251 on Windows 10 amd64 v10.0 at Sun May 10 18:10:36 CEST 2020 [main]
2020-05-10 18:10:56,159 gui3.Apps3                            INFO  - Starting with profile Mon_Réseau_JMRI.3f011120 [main]
2020-05-10 18:10:57,231 node.NodeIdentity                     INFO  - Using b9962ec7-75fe-4571-9857-29c88903aad8 as the JMRI storage identity for profile id 3f011120 [AWT-EventQueue-0]
2020-05-10 18:11:00,686 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-05-10 18:11:00,687 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\acara\JMRI\Mon_Réseau_JMRI.jmri\ECOS\ [main]
2020-05-10 18:11:00,689 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\acara\JMRI\Mon_Réseau_JMRI.jmri\ [main]
2020-05-10 18:11:00,691 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\acara\JMRI\ [main]
2020-05-10 18:11:00,692 util.FileUtilSupport                  INFO  - File path home: is C:\Users\acara\ [main]
2020-05-10 18:11:00,693 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2020-05-10 18:11:28,743 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]
2020-05-10 18:11:55,289 audio.DefaultAudioManager             INFO  - Shutting down active AudioFactory [AWT-EventQueue-0]
2020-05-10 18:11:56,341 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
2020-05-10 18:12:17,920 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-05-10 18:12:17,990 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\acara\JMRI\log\messages.log [main]
2020-05-10 18:12:17,993 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\acara\JMRI\log\session.log [main]
2020-05-10 18:12:18,018 apps.AppsBase                         INFO  - DecoderPro version 4.19.6+Rdb3d7c2db starts under Java 1.8.0_251 on Windows 10 amd64 v10.0 at Sun May 10 18:12:18 CEST 2020 [main]
2020-05-10 18:12:21,225 gui3.Apps3                            INFO  - Starting with profile Mon_Réseau_JMRI.3f011120 [main]
2020-05-10 18:12:22,389 node.NodeIdentity                     INFO  - Using b9962ec7-75fe-4571-9857-29c88903aad8 as the JMRI storage identity for profile id 3f011120 [AWT-EventQueue-0]
2020-05-10 18:12:25,990 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-05-10 18:12:25,992 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\acara\JMRI\Mon_Réseau_JMRI.jmri\ECOS\ [main]
2020-05-10 18:12:25,994 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\acara\JMRI\Mon_Réseau_JMRI.jmri\ [main]
2020-05-10 18:12:25,996 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\acara\JMRI\ [main]
2020-05-10 18:12:25,997 util.FileUtilSupport                  INFO  - File path home: is C:\Users\acara\ [main]
2020-05-10 18:12:25,998 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2020-05-10 18:18:45,944 roster.Roster                         INFO  - Roster rebuilt, stored in C:\Users\acara\JMRI\Mon_Réseau_JMRI.jmri\ECOS\roster.xml [AWT-EventQueue-0]
2020-05-10 18:23:07,740 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]
2020-05-10 18:27:03,650 audio.DefaultAudioManager             INFO  - Shutting down active AudioFactory [AWT-EventQueue-0]
2020-05-10 18:27:04,729 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
2020-05-10 18:28:25,552 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-05-10 18:28:25,625 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\acara\JMRI\log\messages.log [main]
2020-05-10 18:28:25,629 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\acara\JMRI\log\session.log [main]
2020-05-10 18:28:25,658 apps.AppsBase                         INFO  - DecoderPro version 4.19.6+Rdb3d7c2db starts under Java 1.8.0_251 on Windows 10 amd64 v10.0 at Sun May 10 18:28:25 CEST 2020 [main]
2020-05-10 18:28:39,206 gui3.Apps3                            INFO  - Starting with profile Mon_Réseau_JMRI.3f011120 [main]
2020-05-10 18:28:40,253 node.NodeIdentity                     INFO  - Using b9962ec7-75fe-4571-9857-29c88903aad8 as the JMRI storage identity for profile id 3f011120 [AWT-EventQueue-0]
2020-05-10 18:28:44,030 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-05-10 18:28:44,032 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\acara\JMRI\Mon_Réseau_JMRI.jmri\ECOS\ [main]
[... cut ...]
2020-05-12 11:53:30,308 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\acara\JMRI\simulation.jmri\ [main]
2020-05-12 11:53:30,308 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\acara\JMRI\ [main]
2020-05-12 11:53:30,308 util.FileUtilSupport                  INFO  - File path home: is C:\Users\acara\ [main]
2020-05-12 11:53:30,308 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2020-05-12 12:11:33,638 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]

rhwood, 2020-05-12:
This appears to be a specific problem with the Ecos system, since it attempts to write the roster as JMRI is shutting down.

PUlvestad, 2020-05-12:
While there may be a Ecos specific problem, if the roster has been opened, JMRI writes the roster.xml file when JMRI shuts down even if no changes are made. I'm using a LocoNet connection.

As noted on the users group, I haven't seen the roster disappearing roster since #8537 when it had been happening almost every time prior.

rhwood, 2020-05-12:
@PUlvestad The Roster is saved on shutdown except by the EcosLocoAddressManager of if the WarrantManager determines the speed profile of a RosterEntry needs to be saved.

> JMRI writes the roster.xml file when JMRI shuts down even if no changes are made

Can you point me to where that occurs, since this should not be happening?

bobjacobsen, 2020-05-12:
> if the roster has been opened, JMRI writes the roster.xml file when JMRI shuts down even if no changes are made

I don't see that, but perhaps I'm not recreating it properly.  I've been starting DecoderPro (is that what you mean by roster open?) with various simulated connections.  Does it perhaps need a real one?

PUlvestad, 2020-05-12:
I use PanelPro so if I want to see the roster I need to open it from the menu bar.

Pugwash1, 2020-05-13:
I have seen this only a couple of times. I use digitrax. When i discovered the roster missing, I always did a kill -9, sure kill, when I started afain the roster was there. This points to the startup not working rather than a failure ate the close.
Steve G.

On May 12, 2020 7:57:36 PM EDT, Peter Ulvestad <notifications@github.com> wrote:
>I use PanelPro so if I want to see the roster I need to open it from
>the menu bar.

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

rhwood, 2020-05-13:
@Pugwash1 Are you stating you have seen this since 4.19.6 was released, or only before that release?

Pugwash1, 2020-05-13:
Always run from latest master. Seen the problem may be 2 weeks ago and then 2 or 3 days ago. Not a high hit issue and basically I ignored. Just thought it was a hiccup.

On May 12, 2020 8:30:57 PM EDT, Randall Wood <notifications@github.com> wrote:
>@Pugwash1 Are you stating you have seen this since 4.19.6 was released,
>or only before that release?

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

alain355, 2020-05-13:
Dear team
Some additional informations: I only run JMRI using Decoderpro (I don't have a real layout). My roster has got a bit more than 104 entries.
**With 4.19.5**

- When running 4.19.5 in either Simulation or ECoS Mode, roster is rewritten each time JMRI is run. Columns width is kept as needed.
- When using 4.19.5 in ECoS Mode, and trying to acces the ECoS Lok database, sometime JMRI is missing some entries stored in the ECoS, however after closing then relauching DecoderPro in ECoS mode many times, the ECoS Loco DB is fully populated.

**With 4.19.6** reinstalled today

- When running 4.19.6 in Simulation Mode, 1st launch after installing it, roster is OK, with columns width as before. Roster is rewritten when closing DecoderPro.
- When running 4.19.6 in Simulation Mode, 2nd run, 3rd, 4th, roster in rewritten each time at closure, still OK with columns width being OK. 

ECoS Mode following those 4 simulation modes:

- 1st ECos Mode run: Roster (the one saved on last Simulation mode) is OK, with columns being OK. ECoS Loco DB fully correct. 
- If killing JMRI from the Windows task Mgr, then relaunching in ECoS Mode: Roster is OK, columns still OK, but the ECoS Loco DB view within JMRI is partly wrong about 17 entries are missing. Roster.xml file saved but still fully populated.
- 3rd ECoS run after a nice Exit from 2nd run: Roster looks empty (but file is there and fully populated), so I recreate roster,  Roster groups created, roster file still populated, but roster view only shows the roster groups, So I "update" decoder definitions >> Roster view populated, columns still OK. But ECoS Loco DB now fully empty. Roster file still present with right size after closure
- 4th ECoS run after a nice exit from 3rd run:  Roster OK, columns size OK, ECoS Loco DB partly repopulated. I think the missing entries are the same when looking at the screen, but I'm not fully sure. Roster file still OK after closure.
- 5th ECoS run: But no roster entries again, while file roster.xml still there and fully populated, I'm using again the same trick to rebuilt roster then update decoder definitions to get a correct roster view, with still the ECoS DB being fully empty.
- 6th Run: same as 5th run, no roster at launch. Exiting directly: a new **empty** roster file is created
- 7th run: using that empty roster as entry, I rebuilt Roster (groups created, no roster view), then update decoder definition to get the roster view.

Today, it seems the column's width problem didn't appears!

**Going back to 4.19.5 **

- Running ECoS Mode: Roster view OK, Columns width OK, ECoS Loco DB OK
- 2nd and 3rd run in ECoS Mode: still everything being OK.
- Switching to Simulation Mode: Everything looks OK
- Switching back to ECoS Mode: everything looks OK
- Staying in ECoS Mode: everything still OK

I've attached both today's log file, screen pictures of the various states described. I hope it will help to debug.
[Extract_from_JMRI_messages.log](https://github.com/JMRI/JMRI/files/4620668/Extract_from_JMRI_messages.log)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Alain

PUlvestad, 2020-05-16:
https://groups.io/g/jmriusers/message/173576

hcripe, 2020-05-17:
Thanks Peter - I will try the suggestions in the threads to rebuild the roster. My problem isn't that it gets corrupted as far as I know, but that the roster is blank when starting up and wants to re-add all of the roster entries from the ECOS controller whenever I start JMRI. I will monitor the roster.xml file to see if I can narrow it down.

Howard

sidlo64, 2020-05-20:
I have same problem #7607

rhwood, 2020-05-20:
@sidlo64 You reported that #7607 was fixed in 4.18, yet this problem is new in 4.19.6 and is very specific to EcOS connections (#7607 appears to be reported against LocoNet connections). Are you attempting to suggest that your report that #7607 was fixed for you was never correct, or are you attempting to suggest something else?

sidlo64, 2020-05-21:
@rhwood Now I see that the description of problem #7607 was not complete. The connection Digitrax - Loconet simulator - Linux was affected.

In my opinion, the mistake was never fixed. The error just disappeared.

Today I tested version 4.19.7ish+jenkins+20200521T0417Z+R0c44bb57f9 
Connection: Digitrax - Loconet simulator - Linux
It works properly.

alain355, 2020-05-21:
Dear contributors
On my side the issue happens with an ECoS Command station and 4.19.6 (thus I'm currently back to 4.19.5).
I'll try tomorrrow to see if it happens using Jenkins built JMRI.4.19.3ish+jenkins+20200129T0314Z+R6ee334f.exe

Alain in Paris

alain355, 2020-05-21:
I've got an issue, I can't access on Jenkins anything more recent that this 4.19.3
JMRI.4.19.3ish+jenkins+20200129T0314Z+R6ee334f.exe 

Can someone give me an updated link?

dheap, 2020-05-21:
https://builds.jmri.org/jenkins/job/development/job/packages/

alain355, 2020-05-22:
Today's update using jenkins JMRI.4.19.7ish+jenkins+20200521T1028Z+R0c44bb57f9.exe
in short : From bad to worse

With a fresh PC (I had a BSOD last week, thus reinstalled everything from a Win10 ISO), upgrading from previous 4.19.5 to this jenkins built, it a was a full LOST ROSTER session when switching to ECoS: from Simulation (no DCC command station to ECoS) the roster was ther, but once JMRI closed then relaunched in ECoS Mode, the roster wasn't there, SO:

1) I uninstalled JMRI, 
2) I fully deleted my JMRI folder (I have a backup in another PC...)
3) I rebooted the PC
4) I installed jenkins 4.19.5
5) I defined a simulation mode wo any command station
6) I created a new folder to host the roster file location (step 8), under the simulation folder
7) I deleted the default JMRI directory created at 1st JMRI run (step4),
8) I changed the default directory for the simulation mode, restarted JMRI
9) I checked that the new default directory was OK in prefrences
10) I copied all my roster entries (100 and some)
11) I launched JMRi in Simulation mode a lot of time, everything OK, except that all roster entry medias (copied in the roster folder with xml files) were not shown in the roster, so after a few launches I added them to each entry. And they were shown OK !
CC at step 11 running only in Simulation mode, JMRI looks perfectly OK roster picked up, columns size OK, icons for engines OK. **I save the roster file with a different extension**

Then start the mess:
I define a new config, in that new configuration the ECC system is ESU, ECoS, I change the roster directory to be the same than in simulation mode. I save the preferences then restart JMRI: BINGO, no roster, everything lost again. I try to recreate then update (my previous trick), no result, each restart in ECoS mode shows an empty roster.

I relaunch in Simulation mode: no roster, so I recreate it, update decoder definitions, roster created and showing, wo the engines media.
I check (using Winmerge) the roster.xml file created with the .bak one: same files except the JMRI time stamp. But when comparing with the file saved before running in EcoS mode: they are not the same content and size being different). 
Last action, I restore the roster file I saved in step 11, relaunch in simulation mode: a perfect roster is still there, with roster media pictures, columns being OK.

Thus the issue is only happening all the time when JMRI is run with the ECoS.

Giving up for toady, but something has bein changed between 4.19.x and 4.19.6, then 4.19.7
log file:
[messages.log](https://github.com/JMRI/JMRI/files/4668086/messages.log)

Roster in sim mode
[image]

rhwood, 2020-05-24:
I think I understand the issue, but do not have a fix for it yet (there is no simulation mode for ECoS connections, so I can't test anything):

Since JMRI 4.1.4 (Nov 2015), the Roster in JMRI has been loaded and configured by a RosterConfigManager (RCM) *after* connections are configured and started. We have merely been lucky that the ECoS has taken long long enough to start up (possibly because too much work occurs on the GUI thread) that fixing #8144 and #8389 have the effect of bringing the ECoS connection up faster than the RCM. This is causing the ECoS to create a Roster (and immediately force it to be written to disk) before the RCM reads any existing Roster index file.

alain355, 2020-05-24:
Thanks Randall.If I can help to debug, just ask. Maybe adding a sleeping delay before the ECoS startup could fix it (I’ve done such tricks years ago when programming « green waves » traffics lights in my first job delivering city traffic management software).
Alain

rhwood, 2020-05-24:
@alain355 Can you build JMRI from the "ecos-roster" branch in my fork (https://github.com/rhwood/JMRI/tree/ecos-roster) and test that?

alain355, 2020-05-24:
Hi RandallNo sure I know how to do it, as I had quite a lot of issues with Github when pushing updated decoder definitions a few months ago.
Alain

rhwood, 2020-05-24:
What Git client are you using?

rhwood, 2020-05-24:
@alain355 What git client are you using?

alain355, 2020-05-24:
Hi Randall, AFAIK, none. I tried wo success when JMRI repository moved from SourceForge to Github, I even submitted a bug report to Github at that time. 

My contributions to JMRI have only been:
1) on xml files for decoders definition for either French RR models (Mistral Trains) or updates on LAISDCC, Lenz and Zimo. For thoses definitions or updates, I first checked the new decoder definition locally on my Windows PC, then updated it on Github, checked integration and asked my updates to be merged into JMRI's master, with the help of either Bob Jacobsen, or David Heap when the merge crashed.
2) translating from English to French for the French version of Engine Driver Android App with Peter Ackers.

Thus if you can create the .exe file for Win10, I'll happily test it next week. I can also use a Linux Mint 32 bits version, as I have and old notebook running Mint at home.

Alain in Paris

rhwood, 2020-05-24:
I can’t build EXEs for Windows. What IDE do you use?

alain355, 2020-05-24:
Randall,

Sorry, but none at all. I'm going to GitHub using either Chrome or Firefox. When updating or creating xml decoder files, I use XML Marker to insure xml are valid, then check them locally with JMRI beforte pushing them to GitHub..

But there's another option: Can you build a Linux Mint 32 bits version, as I have a small notebook running Linux Mint at home which can connect to my ECoS?

danielb987, 2020-05-24:
Do you need an exe file? If you go to the installation folder, which for me is `C:\Program Files (x86)\JMRI`, there is a file `jmri.jar`. That file can @rhwood easily build for you.

If you try this, make sure you take a backup on the folder first in case something goes wrong.

alain355, 2020-05-25:
Thanks Daniel and Randall

Daniel: (I never developped SW using Java)
Do you mean only changing the jmri.jar produced for any platform will be sufficient, ie. a jmri.jar file is the same whatever the platform using it?

Randall:
In that case, can you send me an updated jmri.jar file which I will put into the right Program directory to test.

Alain in Paris

danielb987, 2020-05-25:
Yes, Java is compiled to bytecode that runs on a virtual machine. C++ and other languages are often compiled to machine code that only runs on a particular type of CPUs.

So compiled Java code, as jmri.jar, can run on any computer that has a Java virtual machine.

The exe file distributed with JMRI starts the Java virtual machine.

alain355, 2020-05-25:
Thanks Daniel, I'll be a little less ignorant nos
Randall, can you provide an updated JMRI.jar file?
Alain in Paris

rhwood, 2020-05-25:
Please try https://www.icloud.com/iclouddrive/06Bh9niP3V16iq2TUrGAcIRJA#JMRI.4.19

alain355, 2020-05-25:
Hi Randall

A huge ** THANKS.** It works under Linux Mint 19.3 Cinnamon (see below) 

 I tried your 4.19.7 on my old and slow notebook running Linux Mint (SAMSUNG N130 10 years old), launched it 1st using the simulation (roster OK), then using the ECoS config once (OK), then a 2nd time OK, added EcoS object ID to the roster view, all OK, launched a 3rd time, OK with the new columns I added OK, 4th and 5th launch OK. **Version good to go on Linux Mint** EXCELLENT.

Now copying only the jmri.jar file from the linux notebook to windows, replacing the previous jmri;jar file (in C:\Program Files (x86)\JMRI ). NOK, JMRI is crashing after stratup whatever config I use However I think something additional is needed by Java which I didn't copied. Attached in the session log from Windows

**However your fix looks perfectly OK, as it was fully OK under Linux Mint.**

Can you please merge your fix to a future beta release so I get a perfect Win10 JMRI?

[session.log](https://github.com/JMRI/JMRI/files/4678569/session.log)

danielb987, 2020-05-25:
@alain355 
JMRI releases [test releases](https://www.jmri.org/releasenotes/jmri4.19.6.shtml).

On that page, you have "Download links" on the middle of the page. Try to download the Windows release and see if that works. If so, try the jmri.jar file you got from @rhwood.

alain355, 2020-05-26:
Daniel,
I installed the fully Jenkins 4.19.7 build provided by Randall, downloaded from his iCloud repository, to my Linux Mint HW. **It worked fine**.
Then I copied the extracted jmri.jar (coming from the .tgz compressed archive) to my Win10 HW where was installed a previous 4.19.7 Jenkins Build ( JMRI.4.19.7ish+jenkins+20200521T1028Z+R0c44bb57f9.exe from may 21st). result is the log published yesterday.

So I thought I've done what you suggest. But as I may be completely wrong:
Today's test session on Win10 only (_**as Randall's build is fully OK on Linux Mint**_):
1) Downloading latest Jenkins's build JMRI.4.19.7ish+jenkins+20200525T2217Z+Rfa7a9b6f6b.exe , installation OK
2) Launch JMRI using this build **as is**: configuration being simulation (no DCC command station): **OK as expected**, closing JMRI

**Tests:**
3) **Simulation config**: replacing the jmri.jar file by the one provided with Randall's Linux build. Launch JMRI with the replacement jmri.jar file using simulation config: **OK**, relaunching same simulation config: **OK**

4) **ECoS config**: 
4.1) **1st ECoS launch** Roster OK, but engine's icons lost, columns's size changed, both pictures not shown on media tab. Closing JMRI. BTW, none of those appeared yesterday using the full Randall's build. :(
4.2) **2nd ECoS launch**: same result: Roster OK, but engine's icons lost, columns's size changed, both pictures not shown on media tab. I then add media pictures to one roster entry (I've a bit more than 100 of entries). Picture added using the Media tab,
4.2.1) BTW using that jmri.jar file, AFAIK, my preferences are all lost: both the additional startup buttons I've in my preferences don't show, display back to default, Programmer choice lost, so the ability to add pictures from the roster entry is not there; Preferences Startup tab is fully empty, delay startup (it was 16 sec.) back to 4 sec. :( . I reset some of my preferences, then Save and Restart JMRI.
4.3) 3rd ECoS Config: Preferences added in step 4.2.1 now showing, startup delay back to the value I've set, the engine icon I added is still there, the Media tab is now back on the roster entry I want to manage.
So Step 4 conclusion is that it works more or less, as it seems that my preferences are lost at 1st ECoS Startup, but I can add them later wo loosing them again..

Step 5 for "fun", restoring latest Jenkins jmri.jar file (from build JMRI.4.19.7ish+jenkins+20200525T2217Z+Rfa7a9b6f6b) instead of the one from RH build.
5.1) **Simulation 1st launch:** Same result than with 4.3, my preferences are here OK, the icon I added (1 lok only) still there
5.2) **Simulation 2nd launch:** OK
5.3) **ECoS 1st launch:** As expected, roster is lost, and as it's lost, engine medias are also lost. But Preferences are not lost.
5.4) **ECoS 2nd and 3rd launch:** Roster from 1st Launch still there with the same missing media pictures.

Step 6:  restoring the current (RH Build on iCloud) jmri.jar on Win10, restoring my previous roster file 
6.1) **Simulation 1st Launch:** Roster OK, columns size OK, my engines 's medias pictures are back, Many preferences are back OK 👍 
6.2) **Simulation 2nd, 3rd and 4th Launch:** OK  👍 
6.3) **ECoS Config 1st Launch:** As in step 4.1: media Icon lost, column size NOK, but roster is here, preferences look to be OK.
6.4) **ECoS Config 2nd, 3rd and 4th Launch:** Roster as in step 1. In launch 3, I add one Media picture to on roster entry. It's back as expected in Launch4. **After Launch4, I restore my backuped  roster.xml file.**
6.5) **ECoS Config 5th Launch:** Medias Icons lost again, If rebuilding roster, the added in step 6.4 Launch 3 media icon is there. I update a 2nd roster entry with media picture OK.
6.6)  **ECoS Config 6th Launch:** 2 roster entries OK with media 100 to do. WIP

My own conclusions of today's test:  **the new jmri.jar file from Randall is definitely needed, if possible wo losing the media pictures**.

This fix should be integrated in a future Beta then Base release.

 ** Thanks Randall and Daniel**
Attached Today's sessions's log file
[messages.log](https://github.com/JMRI/JMRI/files/4680888/messages.log)

alain355, 2020-05-26:
Thanks Randall :)

BTW, the trick I've used to avoid re-adding medias to each roster entry with the ECoS Config (as they are stored in roster directory under the simulation config directory) is to copy all medias both in the simulation.jmri roster directory (where all roster entries are stored), and in the ECoS.jmri roster directory.

closed 2020-05-26 by bobjacobsen

alain355, 2020-05-26:
I used the wrong build (JMRI.4.19.7ish+jenkins+20200526T1022Z+R802338f149 form May 26th, 12:21), issue still presen). Now using JMRI.4.19.7ish+jenkins+20200526T1620Z+R86df4f27ba:

1) Simulation config OK (still Media icons not showing, but when rebuilding roster, they are here).
2) ECoS config: OK at 1st launch, with Roster and Media icons. OK at 2nd launch, 3rd launch.

Issue definitely fixed. thanks again Randall, Daniel and Bob.

PR #8593 (2020-05-26): fix: Delay loading ECoS roster until JMRI Roster is initialized
Fixes #8526