JMRI/JMRI #8530: LayoutEditorTools and the LayoutEditor Project

dsand47, 2020-05-14:
The LayoutEditorTools class provides support for attaching signal heads, signal masts and entry/exit sensors to layout track components.  For signal heads, it can also create the SSL (simple signal logic).  Attaching theses items completes the implementation of layout connectivity which is the bases for signal mast logic, entry/exit and Dispatcher automation.  The attach process can also add the appropriate icons to the LE panel.

For signal masts and entry/exit sensors, it also possible to detach those items.  The signal head logic does not support the detach action, but it should.

Here is a summary of the class:
- Signal Heads (Context menu and LE Tools menu)
   - Block Boundary (anchor point)
   - Turnout (LH, RH, Wye)
   - Slip 
   - Level crossing
   - XOver
   - Throat to Throat -- LE Tools menu only
   - 3 Way -- LE Tools menu only
- Signal Masts (Context menu)
   - Block Boundary (positional points: A, EB, EC)
   - Turnout (LH, RH, Wye, XOver)
   - Slip
   - Level crossing
- NX Sensors (Context menu)
   - Block Boundary (positional points: A, EB, EC)
   - Turnout (LH, RH, Wye, XOver)
   - Slip
   - Level crossing

I plan to add the ability to detach signal heads and optionally delete the related SSL.

To avoid conflict with the LE project, I have a couple of questions.
- Is this class within the scope of the project?
- Due to the size of this class, should it be refactored?
- Should the scope include support for issue #8339?  This changes how turnouts are referenced.

rhwood, 2020-05-14:
> The LayoutEditorTools class provides support for attaching signal heads, signal masts and entry/exit sensors to layout track components. For signal heads, it can also create the SSL (simple signal [...]
This class definitely should be refactored. If not for the size, it should also be refactored because it has too many responsibilities.

dsand47, 2020-05-16:
> This class definitely should be refactored. If not for the size, it should also be refactored because it has too many responsibilities.

It seems to me that the obvious split is to create classes for signal heads, signal masts and sensors that extend LayoutEditorTools with common methods remaining in LayoutEditorTools.  I don't see any value in going lower, such as classes for SensorAtPositionalPoint, SensorAtTurnout, SensorAtLevelCrossing, etc.

LayoutEditorTools is one of the `per LayoutEditor panel singletons`.  Instead of adding the new classes to LayoutEditor, I am thinking of having LayoutEditorTools still be main entry point and it would pass along requests to the appropriate class and method.  This way the menus that invoke the tools don't need to be changed.

pabender, 2020-05-17:
> It seems to me that the obvious split is to create classes for signal heads, signal masts and sensors that extend LayoutEditorTools with common methods remaining in LayoutEditorTools. I don't see [...]

This really should follow the the SOLID design principles.

https://en.wikipedia.org/wiki/SOLID

The first of those says that each class should have a single responsibility.

> LayoutEditorTools is one of the `per LayoutEditor panel singletons`. Instead of adding the new classes to LayoutEditor, I am thinking of having LayoutEditorTools still be main entry point and it [...]

Why should that be an issue?  We SHOULD be able to redesign the implementation of the actions of any GUI element without significant change to the implementation of that GUI element.  Changing the class invoked by a menu item is definitely not a significant change, so long as the result of the selection doesn’t change.

dsand47, 2020-05-17:
> > > This class definitely should be refactored. If not for the size, it should also be refactored because it has too many responsibilities.
> > 
> > 
> > It seems to me that the obvious split is to create classes for signal heads, signal masts and sensors that extend LayoutEditorTools with common methods remaining in LayoutEditorTools. I don't see any value in going lower, such as classes for SensorAtPositionalPoint, SensorAtTurnout, SensorAtLevelCrossing, etc.
> 
> This really should follow the the SOLID design principles.
> 
> https://en.wikipedia.org/wiki/SOLID
> 
> The first of those says that each class should have a single responsibility, which goes to @rhwood‘s point.
> 
> > LayoutEditorTools is one of the `per LayoutEditor panel singletons`. Instead of adding the new classes to LayoutEditor, I am thinking of having LayoutEditorTools still be main entry point and it would pass along requests to the appropriate class and method. This way the menus that invoke the tools don't need to be changed.
> 
> Why should that be an issue? We SHOULD be able to redesign the implementation of the actions of any GUI element without significant change to the implementation of that GUI element. Changing the class invoked by a menu item is definitely not a significant change, so long as the result of the selection doesn’t change.

Where is the point of diminishing returns for refactoring?

Here is the set sensor dialog when a turnout context menu was used.
[image]
The number of boxes depends on the number of block boundaries.  The dialog has two functions:  attach/detach sensors at block boundaries and add/remove NX sensor icons.  There are separate dialogs and supporting code for positional points, turnouts, slips and level crossings.

The signal mast dialogs follow the same pattern.

The signal head dialogs have a different format with 7 variations.  They also have the option of creating the signal head SSL.
[image]
The signal head dialogs currently are not able to detach signal heads or remove icons.

Signal heads have an additional set of entry points from the LE Tools menu.  This requires dialog changes since the track component is not preselected, so the dialogs have to include selection combo boxes.

closed 2021-04-14 by stale[bot]