JMRI/JMRI #8535: Unable to run jmrix.loconet.bluetooth.ConnectionConfigTest via ant

icklesteve, 2020-05-15:
Hi Folks,

Since the ant test-single target was updated to enable the test whole class .* format, I've been unable to run the jmri.jmrix test locally without failure.

Any thoughts?

```
 Failures (1):
   JUnit Vintage:ConnectionConfigTest:testLoadDetails
     MethodSource [className = 'jmri.jmrix.loconet.bluetooth.ConnectionConfigTest', methodName = 'testLoadDetails', methodParameterTypes = '']
     => java.lang.NullPointerException
        jmri.jmrix.loconet.bluetooth.ConnectionConfig.getPortNames(ConnectionConfig.java:79)
        jmri.jmrix.AbstractSerialConnectionConfig.loadDetails(AbstractSerialConnectionConfig.java:323)
        jmri.jmrix.AbstractSerialConnectionConfigTestBase.lambda$testLoadDetails$0(AbstractSerialConnectionConfigTestBase.java:19)
        org.assertj.swing.edt.GuiActionRunner$2.executeInEDT(GuiActionRunner.java:124)
        org.assertj.swing.edt.GuiTask.run(GuiTask.java:36)
        java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
        java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
        java.awt.EventQueue.access$500(EventQueue.java:97)
        java.awt.EventQueue$3.run(EventQueue.java:709)
        java.awt.EventQueue$3.run(EventQueue.java:703)
        [...]
```

rhwood, 2020-05-15:
@icklesteve Can you run the same test using maven on your computer (`mvn test -Dtest= jmri.jmrix.loconet.bluetooth.ConnectionConfigTest`)?

danielb987, 2020-05-15:
@icklesteve 
Which command line are you using? I tried it and it worked for me:
```
ant realclean
ant test-single -Dtest.includes=jmri.jmrix.loconet.bluetooth.ConnectionConfigTest
```

bobjacobsen, 2020-05-15:
Except for a ShutDownManager warning (which is really testing the test, as opposed to testing the useful code), it works fine for me on macOS Mojave:

```
JMRI/projects/JMRI/ % ./runtest.csh jmri.jmrix.loconet.bluetooth
May 15, 2020 12:56:00 PM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/Users/jake/Documents/Trains/JMRI/projects/JMRI/target/test-classes/junit-platform.properties].
Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
BlueCove log redirected to log4j
BlueCove version 2.1.1-SNAPSHOT on mac
ERROR - Test jmri.jmrix.loconet.bluetooth.ConnectionConfigTest left ShutDownTask registered: Writing Blocks (of type class jmri.BlockManager$1)} [main] jmri.util.JUnitUtil.checkShutDownManager()
java.lang.Exception: traceback
	at jmri.util.JUnitUtil.checkShutDownManager(JUnitUtil.java:994)
	at jmri.util.JUnitUtil.tearDown(JUnitUtil.java:291)
	at jmri.jmrix.loconet.bluetooth.ConnectionConfigTest.tearDown(ConnectionConfigTest.java:27)

Test run finished after 3357 ms
[         6 containers found      ]
[         0 containers skipped    ]
[         6 containers started    ]
[         0 containers aborted    ]
[         6 containers successful ]
[         0 containers failed     ]
[        19 tests found           ]
[         0 tests skipped         ]
[        19 tests started         ]
[         7 tests aborted         ]
[        11 tests successful      ]
[         1 tests failed          ]

Failures (1):
  JUnit Vintage:ConnectionConfigTest:testLoadDetails
    MethodSource [className = 'jmri.jmrix.loconet.bluetooth.ConnectionConfigTest', methodName = 'testLoadDetails', methodParameterTypes = '']
    => java.lang.AssertionError: Unexpected ERROR or higher messages emitted: Test jmri.jmrix.loconet.bluetooth.ConnectionConfigTest left ShutDownTask registered: Writing Blocks (of type class jmri.BlockManager$1)}
       org.junit.Assert.fail(Assert.java:88)
       org.junit.Assert.assertTrue(Assert.java:41)
       org.junit.Assert.assertFalse(Assert.java:64)
       jmri.util.JUnitUtil.tearDown(JUnitUtil.java:344)
       jmri.jmrix.loconet.bluetooth.ConnectionConfigTest.tearDown(ConnectionConfigTest.java:27)
       sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
       sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
       sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
       java.lang.reflect.Method.invoke(Method.java:498)
       org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
       [...]
BlueCove stack shutdown completed
```

bobjacobsen, 2020-05-15:
Similar with the `ant` target:
```
% ant test-single -Dtest.includes=jmri.jmrix.loconet.bluetooth.ConnectionConfigTest
Buildfile: /Users/jake/Documents/Trains/JMRI/projects/JMRI/build.xml

-pre-single-test:

-test-single-src:

-test-single-test:

init:

copyfiles:

jjtree:

javacc:

update-template-code:
     [copy] Copying 1 file to /Users/jake/Documents/Trains/JMRI/projects/JMRI/target/classes

compile-generated-source:

compile:

debug:

tests:
    [javac] Compiling 3 source files to /Users/jake/Documents/Trains/JMRI/projects/JMRI/target/test-classes
    [javac] Note: /Users/jake/Documents/Trains/JMRI/projects/JMRI/java/acceptancetest/step_definitions/apps/ApplicationTestAcceptanceSteps.java uses or overrides a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.

runtime-library-selection:
     [echo] arch.lib.path /Users/jake/Documents/Trains/JMRI/projects/JMRI/lib/macosx/x86_64:/Users/jake/Documents/Trains/JMRI/projects/JMRI/lib/macosx

test-single:
     [java] May 15, 2020 12:57:54 PM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
     [java] INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/Users/jake/Documents/Trains/JMRI/projects/JMRI/target/test-classes/junit-platform.properties].
     [java] Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
     [java] BlueCove log redirected to log4j
     [java] BlueCove version 2.1.1-SNAPSHOT on mac
     [java] ERROR - Test jmri.jmrix.loconet.bluetooth.ConnectionConfigTest left ShutDownTask registered: Writing Blocks (of type class jmri.BlockManager$1)} [main] jmri.util.JUnitUtil.checkShutDownManager()
     [java] java.lang.Exception: traceback
     [java] 	at jmri.util.JUnitUtil.checkShutDownManager(JUnitUtil.java:994)
     [java] 	at jmri.util.JUnitUtil.tearDown(JUnitUtil.java:291)
     [java] 	at jmri.jmrix.loconet.bluetooth.ConnectionConfigTest.tearDown(ConnectionConfigTest.java:27)
     [java] 
     [java] Thanks for using JUnit! Support its development at https://junit.org/sponsoring
     [java] 
     [java] 
     [java] Failures (1):
     [java]   JUnit Vintage:ConnectionConfigTest:testLoadDetails
     [java]     MethodSource [className = 'jmri.jmrix.loconet.bluetooth.ConnectionConfigTest', methodName = 'testLoadDetails', methodParameterTypes = '']
     [java]     => java.lang.AssertionError: Unexpected ERROR or higher messages emitted: Test jmri.jmrix.loconet.bluetooth.ConnectionConfigTest left ShutDownTask registered: Writing Blocks (of type class jmri.BlockManager$1)}
     [java]        org.junit.Assert.fail(Assert.java:88)
     [java]        org.junit.Assert.assertTrue(Assert.java:41)
     [java]        org.junit.Assert.assertFalse(Assert.java:64)
     [java]        jmri.util.JUnitUtil.tearDown(JUnitUtil.java:344)
     [java]        jmri.jmrix.loconet.bluetooth.ConnectionConfigTest.tearDown(ConnectionConfigTest.java:27)
     [java]        sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
     [java]        sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
     [java]        sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
     [java]        java.lang.reflect.Method.invoke(Method.java:498)
     [java]        org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
     [java]        [...]
     [java] 
     [java] Test run finished after 4430 ms
     [java] [         3 containers found      ]
     [java] [         0 containers skipped    ]
     [java] [         3 containers started    ]
     [java] [         0 containers aborted    ]
     [java] [         3 containers successful ]
     [java] [         0 containers failed     ]
     [java] [         8 tests found           ]
     [java] [         0 tests skipped         ]
     [java] [         8 tests started         ]
     [java] [         5 tests aborted         ]
     [java] [         2 tests successful      ]
     [java] [         1 tests failed          ]
     [java] 
     [java] BlueCove stack shutdown completed

BUILD FAILED
/Users/jake/Documents/Trains/JMRI/projects/JMRI/build.xml:752: Java returned: 1

Total time: 22 seconds
```

icklesteve, 2020-05-20:
Thanks for testing this folks.

Full ant task being used
`ant test-single -Dtest.includes=jmri.jmrix.loconet.bluetooth.ConnectionConfigTest`

I attempted running via maven, though I'm not a Maven user and kept on getting the error
`Unknown lifecycle phase "jmri.jmrix.loconet.bluetooth.ConnectionConfigTest".`

I then realised that this laptop may actually have BT.
There's no BT button on case or icon on screen, although Windows device manager did find a BT adapter.
Win PC > Control Panel > Hardware and Sounds > Device Manager > BlueTooth Radios > Generic BlueTooth Adapter.
Disabling the adapter, restarted the machine and I was able to run the test 10 times without issue.

Enabled the adapter and the test failed instantly.

Im leaving it disabled so I can run jmrix package wide tests, though do find it odd that a BT connection test fails with a null pointer when a potential BT connection point is found.
Changing issue title to reflect this.

rhwood, 2020-05-20:
The maven command to run a single test is `mvn test -Dtest= jmri.jmrix.loconet.bluetooth.ConnectionConfigTest`

icklesteve, 2021-03-30:
Still unable to run single unit tests without this error

closed 2021-05-29 by stale[bot]