JMRI/JMRI #8536: BlockManager ShutDownTask in single tests

icklesteve, 2020-05-15:
In latest seperate tests CI scan, ie
https://builds.jmri.org/jenkins/job/development/job/separate-tests/19/consoleFull
there are 853 tests failing with the Block Manager still being present.

Would linting the offending tests by adding JUnitUtil.deregisterBlockManagerShutdownTask(); be of benefit to the project?

```
Sat May  9 06:29:04 CEST 2020
jmri/jmrix/loconet/bluetooth/ConnectionConfigTest
Picked up JAVA_TOOL_OPTIONS: -Dfile.encoding=UTF-8
May 09, 2020 6:29:05 AM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/var/lib/jenkins/workspace/development/separate-tests/target/test-classes/junit-platform.properties].
Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer
BlueCove log redirected to log4j
Native Library bluecove_x64 not available
ERROR - Test jmri.jmrix.loconet.bluetooth.ConnectionConfigTest left ShutDownTask registered: Writing Blocks (of type class jmri.BlockManager$1)} [main] jmri.util.JUnitUtil.checkShutDownManager()
java.lang.Exception: traceback
	at jmri.util.JUnitUtil.checkShutDownManager(JUnitUtil.java:994)
	at jmri.util.JUnitUtil.tearDown(JUnitUtil.java:291)
	at jmri.jmrix.loconet.bluetooth.ConnectionConfigTest.tearDown(ConnectionConfigTest.java:27)

Test run finished after 1215 ms
[         3 containers found      ]
[         0 containers skipped    ]
[         3 containers started    ]
[         0 containers aborted    ]
[         3 containers successful ]
[         0 containers failed     ]
[         8 tests found           ]
[         0 tests skipped         ]
[         8 tests started         ]
[         5 tests aborted         ]
[         2 tests successful      ]
[         1 tests failed          ]

```

rhwood, 2020-05-16:
> Would linting the offending tests by adding JUnitUtil.deregisterBlockManagerShutdownTask(); be of benefit to the project?

Absolutely not. These tests should *not* be showing an exception that does not occur when running them as part of the standard build. Understanding why they are failing would be of benefit, but just plastering over that problem without understanding it would not be of benefit.

bobjacobsen, 2020-05-16:
To 1st order, we do understand this:  In regular operation , the code _doesn't_ need to clean up the shut down operations.  They're there for when the program _shuts_ _down_, which always eventually happens.  So there isn't built in logic for cancelling shut down operations that, in real operation, are never cancelled.

Insisting that 800+ pieces of code provide extra, clean support for doing something that _never_ _happens_ during real operation seems like a very low priority to me.

rhwood, 2020-05-16:
What I mean is that I don't understand why 853 tests create a BlockManager when run separately, but don't create one when run as part `ant alltest` or `mvn test` (or alternately why when run inside those constructs, clean up behind themselves, but don't when run individually).

pabender, 2020-05-16:
> On May 16, 2020, at 8:26 AM, Randall Wood <notifications@github.com> wrote:
> 
> ﻿What I mean is that I don't understand why 853 tests create a BlockManager when run separately, but don't create one when run as part ant alltest or mvn test (or alternately why when run inside those constructs, clean up behind themselves, but don't when run individually).

Right, this is confusing to me as well.  

We have other existing tests that have nothing to do with blocks that have had the BlockManager shutdown code attached to them, and I don’t completely understand why that happened.

(  There is at least one connection config test suite and at least two throttle tests suites where this happened. )

I know the BlockManager is loaded as an InstanceManagerAutoDefault.  For some reason the InstanceManager is loading it when there is no reason to load it.*

Paul

*This is an example of where using a Mock InstanceManager would prevent an issue in these tests.  A mock wouldn’t include the logic to load auto defaults.

rhwood, 2020-05-16:
The reason the BlockManager is now visible everywhere is #8367.

The InstanceManager is *only* loading it on request, and the nature of those requests have not changed for, as far as I can tell, years.

bobjacobsen, 2020-05-16:
The `jmri.shutdownmanager` mechanism could be used to load a test-specific `BlockManager`, but I'm not quite sure what that buys in the long run.  If the question is "why is the BlockManager being loaded?", one can put a `.printStackTrace` in the constructor, run the test(s), and see why.

I'm not sure why a specialized `InstanceManager` would help here, as opposed to a specialized `BlockManager`, but that can be done via
 - Change the various control files to put target/test-classes before target/classes
 - Put the instrumented-facade-for-instanceManager code in java/test

pabender, 2020-05-16:
> The `jmri.shutdownmanager` mechanism could be used to load a test-specific `BlockManager`, but I'm not quite sure what that buys in the long run. If the question is "why is the BlockManager being [...]

I am not saying to have a specialized ‘InstanceManager’.  We don’t want to maintain the code for that, and I would never argue that that is a good idea.

What I am suggesting is that we should be using a mocking library to mock the InstanceManager in unit tests.  Mocking allows you to generate the specific expected responses from an object that is NOT the class under test.  If you don’t specifically state in the test setup that you want a specific action to occur in the mock, it doesn’t happen.  In this case, if you don’t tell the mock it knows about a block manager, and the test itself doesn’t create one, the block manager doesn’t exist.

This is a pain however, because the InstanceManager has way to many static methods, and the mocking libraries we use don’t do well with static methods ( to the best of my knowledge none of them do).  I implemented a solution to this problem, but you didn’t like it.

pabender, 2020-05-16:
Ok, so lets take a look at running just one of those tests suites that has had the `JUnitUtil.deregisterBlockManagerShutdownTask();` line added to it's teardown.

For this exercise, I picked https://github.com/JMRI/JMRI/blob/master/java/test/jmri/jmrix/loconet/CsOpSwAccessTest.java

After adding a `new Exception().printStackTrace()` as the second line of the BlockManager constructor (so it is only after a super() call).  We see the attached output.

[testoutput.txt](https://github.com/JMRI/JMRI/files/4638675/testoutput.txt)

Nowhere in the test itself is there ANYTHING that should be triggering a BlockManager initialization, but yet there are several BlockManager creations recorded.

I see two types of creations:

The first creation type is caused by `jmri.jmrix.SystemConnectionMemo` changing the available action list, which causes a scan for actions, which eventually loads the `BlockManager` (by loading the `RouteTableAction`, which loads the `DefaultLogixManager`, which loads a `BlockManager`).

The second creation type appears to be caused by the call to `deregisterBlockManagerShutdownTask` itself.

pabender, 2020-05-16:
#8541 shows how to prevent loading the BlockManager in CsOpSwAccessTest.

Note that this Mocks the LocoNetSystemConnectionMemo, which prevents the first creation type from occurring, and it removes the call to `deregisterBlockManagerShutdownTask` which prevents the second type from occurring.

bobjacobsen, 2020-05-17:
> I am not saying to have a specialized ‘InstanceManager’. We don’t want to maintain the code for that, and I would never argue that that is a good idea.
>
> What I am suggesting is that we should be [...]

We can go over this as many times as you like.

1) You proposed _changing_ _the_ _use_ of InstanceManager.  I have no problem with inserting (mocked or otherwise) objects for testing, but it’s not mocking if we’ll have to change 5000+ references.

2) There are many ways to change the behavior of InstanceManager et al in testing that doesn’t break the basic contract for running code.  C.f. the jmri.shutdownmanager  property which is providing a specific manager during testing.  C.f. InstanceInitializer via the service provider mechanism.  C.f. complete replacement with a testing facade at load time.  Etc.

Bob

pabender, 2020-05-17:
Sent from my iPhone
>> On May 17, 2020, at 10:58 AM, Bob Jacobsen <notifications@github.com> wrote:
>> This is a pain however, because the InstanceManager has way to many static methods, and the mocking libraries we use don’t do well with static methods ( to the best of my knowledge none of them do). I implemented a solution to this problem, but you didn’t like it.
> 
> We can go over this as many times as you like.
> 
> 1) You proposed _changing_ _the_ _use_ of InstanceManager. I have no problem with inserting (mocked or otherwise) objects for testing, but it’s not mocking if we’ll have to change 5000+ references.

My goal is still to get us away from making so many static calls and to simplify the test initialization and teardown process.  That process is much more complicated than it should be, especially for simple objects.

We can do this with different approaches than my previous proposal, which I won’t reintroduce, but it is the static calls to the instance manager that are a problem, and our test setup and teardown is causing more of those.

A case in point is caused by deregisterBlockManagerShutdownTask. When it is called and there isn’t a block manager, it is asking the InstanceManager for one and the InstanceManager creates one.  We shouldn’t be creating new objects during test cleanup just to immediately clean those objects up.  I’ve already cleaned up one instance of this particular issue, and I am sure there are more.

Paul

bobjacobsen, 2020-05-17:
See #8550 for another approach to `deregisterBlockManagerShutdownTask` itself.

Although they are themselves static, the `InstanceManager` accessor end up going through a replaceable object obtained with `getDefault()`.  That in turn can change.

pabender, 2020-05-18:
> Although they are themselves static, the `InstanceManager` accessor end up going through a replaceable object obtained with `getDefault()`. That in turn can change.

This is true, but the problem we know about is that late calls (calls after the result of getDefault() has changed) are problematic.  We keep putting out fires because of that.

The problem traced through CsOpSwAccessTest however is different and frustrating for it's own reasons.

Creation of a SystemConnectionMemo that registers actions triggers creation of a StartupActionModelUtil object.  That StartupActionModelUtil object triggers creating a RouteTableAction, which  @rhwood and I both independently discovered, is asking the LogixManager to load a static variable:

https://github.com/JMRI/JMRI/blob/660468520178bdc0c3150a32ab9df27d4b5f432a/java/src/jmri/jmrit/beantable/RouteTableAction.java#L2096-L2100

It's loading the LogixManager that causes the BlockManager to load, and this only happens once because the variable being loaded is static.  But this is also the reason that so many tests are failing when run separately.

@rhwood has this fixed in #8551.  I have a more extensive refactoring of the RouteTableAction that I will be pushing up soon that also fixes this.

rhwood, 2020-06-11:
I think that as of #8659 all of the static initializers that require a BlockManager have been found and fixed.

icklesteve, 2020-06-12:
Closing issue as resolved.
There are a small number of outstanding Block Manager Shutdown Tasks being found by the Single Tests, however these appear valid and are all shown correctly in the failure logs.

closed 2020-06-12 by icklesteve