JMRI/JMRI #8567: Layout Editor Grid never resized Disappearing Grid

Pugwash1, 2020-05-22:
The grid area on a Layout Panel appears to be fixed to the first sized used and never updated as the window is resized.
If you create the panel initially on say an Acer One the layouteditor xml has this 
windowheight="528" windowwidth="1024" panelheight="480" panelwidth="1004"
If you open this on a machine with a larger screen size, expand and save, the layouteditor xml now has this
windowheight="814" windowwidth="1393" panelheight="480" panelwidth="1004".
The panelsize has not been updated and that is the size of the grid.

dsand47, 2020-05-22:
The panel size will be updated as objects are placed outside of the grid which results in the grid and panel size expanding to include the new object.

Pugwash1, 2020-05-22:
@dsand47 So its a bug. If the grid is on it should always fill the area you that you can add stuff, to not fill the area unfortunately implies that you cannot add stuf there.

dsand47, 2020-05-22:
It is not a bug if it is working as designed.  The design is questionable and probably needs to be changed.

The obvious solution is to make the grid based on the window size rather than the panel size.

rhwood, 2020-05-22:
Given the excellent unit testing and design documentation of anything in the LayoutEditor package, it is simply not possible to declare this is "working as designed", since we have zero design documentation and precious little tests to verify that anything is working as intended.

dsand47, 2020-05-22:
Since there is no formal design document, then I probably should have said it is working as implemented.  I have looked at enough of the code to know that a panel is defined as the bounds of its objects with an extra margin to the right and bottom.  The grid is tied to the panel bounds.  This means that when the panel is scrolled the grid scrolls with it.  The grid is also dependent on the panel zoom factor.

It is still not a bug.  This is a request that the grid consider the window size.

rhwood, 2020-05-22:
If we are to accept that “working as implemented” means “not a bug” then, by definition, there are no bugs in JMRI. Should we just close every single bug report then?

Randall

dsand47, 2020-05-22:
A bug implies broken.  A disagreement about behavior is not a bug.

pabender, 2020-05-23:
Sent from my iPad

> On May 22, 2020, at 4:36 PM, Dave Sand <notifications@github.com> wrote:
> 
> Since there is no formal design document, then I probably should have said it is working as implemented. I have looked at enough of the code to know that a panel is defined as the bounds of its objects with an extra margin to the right and bottom. The grid is tied to the panel bounds. This means that when the panel is scrolled the grid scrolls with it. The grid is also dependent on the panel zoom factor.

If the code is working as intended, please add a test that verified this.

Tests are a form of documentation for the design.

If the code is not working as intended, you should also write a test that demonstrates this....  this test will fail... but then you change the code so that the test passes.

Remember that documentation is not for those who know how something works, but for the rest of us who are not so confident we know how things should behave.

Paul

jerryg2003, 2020-05-26:
My item #3 in #8556 would appear to be reporting the same "issue" (not wanting to enter into the debate on design, implementation, bug, change, intended, unintended, etc.).  Item #2 may or may not be a bug, may or may not be in the same area of code. [Item #1 is a veiled "request for change/improvement."]

Jerry

closed 2021-04-14 by stale[bot]