JMRI/JMRI #8568: Tests create zombie AbstractMRTrafficControllers

rhwood, 2020-05-23:
Can you see what's wrong with this code?
https://github.com/JMRI/JMRI/blob/30b9ec5fb5db08bfe6090fb51e5ca31fcfa6aa6c/java/test/jmri/jmrix/tmcc/TmccSystemConnectionMemoTest.java#L22-L39

A hint is in the teardown method:
https://github.com/JMRI/JMRI/blob/30b9ec5fb5db08bfe6090fb51e5ca31fcfa6aa6c/java/test/jmri/jmrix/tmcc/TmccSystemConnectionMemoTest.java#L41-L48
and that this teardown method causes a failure:
```
    @Override
    @After
    public void tearDown() {
        scm.getTrafficController().terminateThreads();
        scm.dispose();
        JUnitUtil.tearDown(); // fails with registered ShutDownTask
    }
```

rhwood, 2020-05-23:
The problem in this case is that an unused `SerialTrafficController` object is being created. This unused object is registering shutdown tasks and I don't know what else, and then being disposed of in the `setUp()` method.

Unfortunately in our unit tests, this seems to be a not entirely uncommon error that may leave dangling threads, dangling references to otherwise released objects, and other junk in the tests and may account for later tests being unreliable.

rhwood, 2020-05-23:
It also means that every time this line:
```
JUnitUtil.clearShutDownManager(); // put in place because AbstractMRTrafficController implementing subclass was not terminated properly
```
has been put in place a test is creating a TrafficController (and maybe launching its threads) such as:
https://github.com/JMRI/JMRI/blob/cec187ad536ddc9cda061b137819b53844247d31/java/test/jmri/jmrix/tmcc/serialmon/SerialMonFrameTest.java#L90-L97
(I don't have a clue what this is testing, since the test cases are supposed to be testing a `SerialMonFrame`.)

pabender, 2020-05-23:
Sent from my iPad

> On May 23, 2020, at 5:54 AM, Randall Wood <notifications@github.com> wrote:
> It also means that every time this line:
> JUnitUtil.clearShutDownManager(); // put in place because AbstractMRTrafficController implementing subclass was not terminated properly
> has been put in place a test is creating a TrafficController (and maybe launching its threads) such as:
> https://github.com/JMRI/JMRI/blob/cec187ad536ddc9cda061b137819b53844247d31/java/test/jmri/jmrix/tmcc/serialmon/SerialMonFrameTest.java#L90-L97
> (I don't have a clue what this is testing, since the test cases are supposed to be testing a SerialMonFrame.)
> 
That is probably a leftover from early testing efforts and can be removed.

Paul

pabender, 2020-05-23:
On May 23, 2020, at 5:50 AM, Randall Wood <notifications@github.com> wrote:
> ﻿The problem in this case is that an unused SerialTrafficController object is being created. This unused object is registering shutdown tasks and I don't know what else, and then being disposed of in the setUp() method.

Some of those, like the one you pointed to in the TMCC code, are being registered with the memo ( this happens in the constructor for the traffic controller... ) 

In all likelihood, one or more of the managers requires access to the traffic controller when the memo creates them, or did when the test was created.
> Unfortunately in our unit tests, this seems to be a not entirely uncommon error that may leave dangling threads, dangling references to otherwise released objects, and other junk in the tests and may account for later tests being unreliable.
> 
If I were writing that again today, I would use a mock instead of a real traffic controller with overridden methods.  That would prevent the treads from starting in the first place.

Paul

closed 2021-04-14 by stale[bot]