JMRI/JMRI #8582: Failure to load NCE-Specific Startup actions

dheap, 2020-05-24:
Current HEAD doesn't load my NCE status actions:
[image]

V4.9.16:
[image]

The build log flags a depreciation warning:
```
/Users/heap/jmri/java/src/apps/startup/StartupActionModelUtil.java:188: warning: [deprecation] StartupActionFactory in apps.startup has been deprecated
            asLoader.forEach(factory -> addActions(factory));
```

[NetBeans Log.txt](https://github.com/JMRI/JMRI/files/4674579/NetBeans.Log.txt)
Another example:
[image]

[image]
[NCE_Power_Pro_Bare_copy.jmri.zip](https://github.com/JMRI/JMRI/files/4674581/NCE_Power_Pro_Bare_copy.jmri.zip)
[NCE_Power_Pro.jmri.zip](https://github.com/JMRI/JMRI/files/4674582/NCE_Power_Pro.jmri.zip)

rhwood, 2020-05-24:
Have you tried a complete clean and build?

dheap, 2020-05-25:
Yes.

dheap, 2020-05-25:
@rhwood Yes.

dheap, 2020-05-25:
Different machine, different checkout, ant command line:
[image]
[image]
[Current.txt](https://github.com/JMRI/JMRI/files/4674803/Current.txt)
[V4-19-6-MP.txt](https://github.com/JMRI/JMRI/files/4674804/V4-19-6-MP.txt)
[NCE_Simulator.jmri.zip](https://github.com/JMRI/JMRI/files/4674805/NCE_Simulator.jmri.zip)

rhwood, 2020-05-25:
@dheap For some reason none of the NCE-specific actions in the profiles you attached are associated with any connection by systemPrefix. They need to be (I thought they always needed to be); I'll document that requirement in the release notes.

rhwood, 2020-05-25:
@dheap I am getting crashes attempting to run the profiles you included with "invalid preferences:" locations.

dheap, 2020-05-25:
> @dheap I am getting crashes attempting to run the profiles you included with "invalid preferences:" locations.

That's a long-standing issue I've drawn attention to in the past, but nothing has been done to fix it.

If the `preference:`  references an non-existent path for your machine, JMRI crashes and won't let you change it. The only ways to fix it are to:

1. Create the non-existent path so you can change it in the GUI.
or:
2. Manually edit `jmri-implementation.user-files=...` in profile.properties.

It's an error often encountered if you use roaming profiles and launch that profile from a new machine. It inherits the path and it's time for manually editing `profile:profile.properties` and/or `profile:storage-identity/profile.properties`.

dheap, 2020-05-25:
> @dheap For some reason none of the NCE-specific actions in the profiles you attached are associated with any connection by systemPrefix. They need to be (I thought they always needed to be); I'll document that requirement in the release notes.

That's because Preferences->Startup has never in the past allowed you to associate an NCE action with a connection. The dropdown has always been blank and greyed out.

That's going to be a big job for me to edit all my NCE-containg profiles 👎

rhwood, 2020-05-25:
> > @dheap I am getting crashes attempting to run the profiles you included with "invalid preferences:" locations.
> 
> That's a long-standing issue I've drawn attention to in the past, but nothing has been done to fix it.

I can not find any issue that refers to this.

dheap, 2020-05-27:
> > > @dheap I am getting crashes attempting to run the profiles you included with "invalid preferences:" locations.
> > 
> > 
> > That's a long-standing issue I've drawn attention to in the past, but nothing has been done to fix it.
> 
> I can not find any issue that refers to this.

#5184 refers.

PR #8588 (2020-05-26): fix: associate a memo with startup action stored within memo prefix
If an action that needs a specific connection to act on is stored without a valid identifier for that connection, we should treat is as invalid action; however, there is a special case that needs to be supported for historical reasons--we never made most actions that act on a specific connection do so until #8544; they simply acted on the last connection of a specific type (regardless of the number of connections of a specific type).

Fixes #8582
Fixes #8583
commit 5358f8ae (2020-05-26): fix: associate a memo with startup action stored within memo prefix

If an action that needs a specific connection to act on is stored without a valid identifier for that connection, we should treat is as invalid action; however, there is a special case that needs to be supported for historical reasons--we never made most actions that act on a specific connection do so until #8544; they simply acted on the last connection of a specific type (regardless of the number of connections of a specific type).

Fixes #8582
Fixes #8583
closed 2020-05-27 by dheap