JMRI/JMRI #8583: NPE with Digitrax simulatoe

dheap, 2020-05-25:


dheap, 2020-05-25:
Current HEAD. Doesn't happen with V4.19.6.
[Digitrax Simulator error.txt](https://github.com/JMRI/JMRI/files/4674815/Digitrax.Simulator.error.txt)
[Digitrax_Simulator.jmri.zip](https://github.com/JMRI/JMRI/files/4674817/Digitrax_Simulator.jmri.zip)

bobjacobsen, 2020-05-25:
So far, I haven't been able to recreate this.

The key part is this:
```
2020-05-25 08:50:34,867 node.NodeIdentity                     INFO  - Using 5985ad00-56f1-11e8-8000-001ff34f18ff as the JMRI storage identity for profile id 3f561514 [AWT-EventQueue-0]
2020-05-25 08:50:37,761 loconet.LnPacketizer                  WARN  - passing to xmit: unexpected exception:  [LnPowerManager LnTrackStatusUpdateThread]
java.lang.NullPointerException
	at jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage(LnPacketizer.java:104)
	at jmri.jmrix.loconet.LnPowerManager$LnTrackStatusUpdateThread.run(LnPowerManager.java:181)
2020-05-25 08:50:38,037 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2020-05-25 08:50:38,040 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
```

The `LnPowerManager` kicked off the `LnTrackStatusUpdateThread` to get the status of track power from the command station.  It waits 500msec, and then tries to send a message to the LocoNet.  That wasn't long enough, apparently, as the LocoNet objects weren't ready and error out.  

I can make that wait smarter by having it check for OK from the `LnTrafficController`. That code is now in the bobjacobsen/Issue-8583 branch.  Since I can't recreate the problem, could you try that branch please before I make a PR and merge it?  Thanks.

closed 2020-05-26 by bobjacobsen

dheap, 2020-05-27:
@bobjacobsen Sorry, shamed for a few days and unable to check. But appears fixed now in HEAD, thanks.

PR #8588 (2020-05-26): fix: associate a memo with startup action stored within memo prefix
If an action that needs a specific connection to act on is stored without a valid identifier for that connection, we should treat is as invalid action; however, there is a special case that needs to be supported for historical reasons--we never made most actions that act on a specific connection do so until #8544; they simply acted on the last connection of a specific type (regardless of the number of connections of a specific type).

Fixes #8582
Fixes #8583
commit 5358f8ae (2020-05-26): fix: associate a memo with startup action stored within memo prefix

If an action that needs a specific connection to act on is stored without a valid identifier for that connection, we should treat is as invalid action; however, there is a special case that needs to be supported for historical reasons--we never made most actions that act on a specific connection do so until #8544; they simply acted on the last connection of a specific type (regardless of the number of connections of a specific type).

Fixes #8582
Fixes #8583