JMRI/JMRI #8590: New deprecation warning in master

bobjacobsen, 2020-05-25:
Recently, the HEAD of master has been showing:
```
    [javac] /Users/jake/Documents/Trains/JMRI/projects/JMRI/java/src/apps/startup/StartupActionModelUtil.java:160: warning: [deprecation] StartupActionFactory in apps.startup has been deprecated
    [javac]             asLoader.forEach(factory -> {

```

I don't understand what the proper fix for this is.  I've tracked the change back to a commit in PR #8544:
```
commit cbc8781e581d4559f110bdfbb50130bd6b731a0e
Author: Randall Wood <297232+rhwood@users.noreply.github.com>
Date:   Fri May 1 06:26:30 2020 -0400

    fix: Migrate some startup into `jmri.util.startup`
    
    This migrates most usages of `apps.startup` classes and interfaces by `jmri` classes into `jmri.util.startup`.

:040000 040000 1c3c607df574b7284e5ad70768f6b831fdfa08b9 4190d9ca0aa1725e6e5cdad953255964ccf6854e M	java
```

rhwood, 2020-05-25:
I put a suppressing annotation where NetBeans (or maybe Eclipse, I forget which (I switch between the two  IDEs)) suggested it needed to be.

Since Spotbugs is not failing with an error, we can either move the annotation to the whole method or add/remove the flag for javac to report/not report depreciations (I forget if there is a flag to show or a flag to hide and if we are using it).

Randall

bobjacobsen, 2020-05-26:
I'll move the annotation to the top of the method.

Is there replacement that doesn't use the deprecated method?

rhwood, 2020-05-26:
> Is there replacement that doesn't use the deprecated method?

It is a deprecated class `apps.startup.StartupActionFactory` that is used, not a method, and the in this case, the class is used by the ServiceLoader to find all `META-INF/services/apps.startup.StartupActionFactory` files that exist in the class path.

https://github.com/JMRI/JMRI/blob/802338f1495d65d117a7b1bf2cc97d8ec7195eb1/java/src/apps/startup/StartupActionModelUtil.java#L187-L192

In the only use we look first for the deprecated classes, and then three lines later for its replacement.

bobjacobsen, 2020-05-30:
Resolved by #8592

closed 2020-05-30 by bobjacobsen