JMRI/JMRI #8595: reopening #8526, Lost Roster with ECoS

alain355, 2020-05-26:
Hi Randall (@rhwood ) , Daniel (@danielb987), Bob (@bobjacobsen)

Sorry to reopen #8526 , but when using latest jenkins build (May the 26th, 12:21) from Bob (JMRI.4.19.7ish+jenkins+20200526T1022Z+R802338f149.exe), the issue is back if using the ECoS configuration (no issue if using the simulation mode wo a DCC command station), whatever previous tricks I try, there's no roster with the ECoS.

But, using yesterday jmri.jar file from the Linux iCloud build made by Randall, it's as in my latest comment before the merged version a few hours ago (Thus OK, except the columns size, which I changed OK, and the trick copying Media picture in both directories).

BTW, as I'm a jury at Engineers's certification boards up to next monday (University closure time in France), I won't have a lot of time before next week. 

Alain

rhwood, 2020-05-26:
That build you pulled predates the fix getting merged. You want the builds built at 1620Z.

rhwood, 2020-05-26:
https://builds.jmri.org/jenkins/job/development/job/packages/lastSuccessfulBuild/ should contain the fix to test.

rhwood, 2020-05-26:
Also, please reopen an issue instead of opening a new issue to discuss the closed issue.

alain355, 2020-05-26:
I close this issue and reopen #8526

closed 2020-05-26 by alain355