JMRI/JMRI #8598: PanelPro 4.19.7ish not stopping

dsand47, 2020-05-27:
I am running 4.19.7ish build 211 on macOS 10.15.3.  When doing a Quit, all of the windows except for the main PanelPro window close but PanelPro is still running.  Menu items still work.

the problem does not occur with each PanelPro session.  A force quit does work.

A thread dump is attached.

Dave Sand

[threaddump.txt](https://github.com/JMRI/JMRI/files/4689567/threaddump.txt)

rhwood, 2020-05-27:
Some shutdowntask is likely starting a new thread and waiting for it. Please attach the profile you are using (exporting it through JMRI preferences and copying the user files tends to work best).

dsand47, 2020-05-27:
> Some shutdowntask is likely starting a new thread and waiting for it. Please attach the profile you are using (exporting it through JMRI preferences and copying the user files tends to work best).

I have created a zip file of the profile and its contents.

[DavidPaul.jmri.zip](https://github.com/JMRI/JMRI/files/4689692/DavidPaul.jmri.zip)

rhwood, 2020-05-27:
> [DavidPaul.jmri.zip](https://github.com/JMRI/JMRI/files/4689692/DavidPaul.jmri.zip)

This profile is failing to fail to shutdown on JMRI 4.19.7ish build 212 on OpenJDK 11.0.7+10 / macOS 10.15.5. Your thread dump suggests you provided a different profile than the one that is failing and that this matters in terms of what is (or is not) running when you initialize the stop. 

(Your thread dump suggests you are using an RMI server or client in the profile, while the provided one uses a simulator.)

rhwood, 2020-05-28:
Also did you wait 30 seconds before forcing PanelPro to quit?

dsand47, 2020-05-28:
Since I have no layout hardware, I always use a simulator.  In the case LocoNet.  `LocoNet HexFileFrame` is in the trace dump.

I never start WiThrottle or the Web Server unless absolutely needed.  It takes too long for them to stop.

PR #8601 (2020-05-30): fix: Do not listen to network topology changes on macOS
Fixes #8598
commit 7cb33a30 (2020-05-30): fix: Do not listen to network topology changes on macOS

Fixes #8598
closed 2020-05-30 by bobjacobsen