JMRI/JMRI #8605: Migration for LE XOver Edit needs an intermediate fix.

dsand47, 2020-05-28:
@bobjacobsen 

Currently `LayoutTurnoutEditor` calls an empty stub for the 3 additional blocks for a cross-over.
```
    // for extension in e.g. XOver Editor
    protected void extendBlockBCDSetup(Container contentPane) {}
```
The actual code is in `LayoutXOverEditor` but that code is not being called yet.

The XOver edit also includes the continuing route section that does not belong.  This is a minor issue.

bobjacobsen, 2020-05-29:
Could you tell me how to recreate the problem?

It looks to me like `LayoutXOverEditor#editLayoutTrack` invokes `LayoutTurnoutEditor#editLayoutTrack`, which calls `extendBlockBCDSetup` which is implemented in `LayoutXOverEditor#extendBlockBCDSetup`

So I must be missing something.  Perhaps `LayoutXOverEditor#editLayoutTrack` isn't getting invoked on some path?

Any suggestions for how to recreate and what to look for would be appreciated.  I'd like to get this mistake fixed by the release in a couple days.

dsand47, 2020-05-29:
> Could you tell me how to recreate the problem?
> 
> It looks to me like `LayoutXOverEditor#editLayoutTrack` invokes `LayoutTurnoutEditor#editLayoutTrack`, which calls `extendBlockBCDSetup` which is implemented in `LayoutXOverEditor#extendBlockBCDSetup`

`extendBlockBCDSetup` is also implemented in `LayoutTurnoutEditor`.

> So I must be missing something. Perhaps `LayoutXOverEditor#editLayoutTrack` isn't getting invoked on some path?
> 
> Any suggestions for how to recreate and what to look for would be appreciated. I'd like to get this mistake fixed by the release in a couple days.

Recreate Steps:
1. Start PanelPro.
1. Select **Panels >> New Panel >> Layout Editor**.
1. Select one of the XO options from the Toolbar.
1. Shift click to add it to the panel.
1. Right click the XO and select Edit from the context menu.

The resulting edit dialog:
[image]

The edit dialog should be (4.19.5):
[image]

Here is a traceback from `LayoutTurnoutEditor#extendBlockBCDSetup`.  This method is an empty method.
```
     [java] 2020-05-29 10:21:53,352 youtEditorDialogs.LayoutTurnoutEditor INFO  - extendBlockBCDSetup [AWT-EventQueue-0]
     [java] java.lang.Exception: traceback
     [java] 	at jmri.jmrit.display.layoutEditor.LayoutEditorDialogs.LayoutTurnoutEditor.extendBlockBCDSetup(LayoutTurnoutEditor.java:248)
     [java] 	at jmri.jmrit.display.layoutEditor.LayoutEditorDialogs.LayoutTurnoutEditor.editLayoutTrack(LayoutTurnoutEditor.java:168)
     [java] 	at jmri.jmrit.display.layoutEditor.LayoutTurnout$6.actionPerformed(LayoutTurnout.java:2967)
     [java] 	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
```

bobjacobsen, 2020-05-29:
Thanks again for catching this and the patience in getting me to understand it.

There's a first fix in #8609.  There were multiple private variables with the same name, and I hadn't translated that properly.  

If you get a chance, could you give it a look please?

It's a bit of an inefficient fix, as it creates the `*Editor` object before it's needed. But if this fixes the problem, I can defer those constructors until later.

closed 2020-05-30 by bobjacobsen