JMRI/JMRI #8616: PanelPro Quit not finished 4.19.7

sidlo64, 2020-05-31:
Sometimes PanelPro does not quit after Quit. I can't simulate it. If I use a JMRI web server, the error is more likely.

See log
[session.log](https://github.com/JMRI/JMRI/files/4708114/session.log)

pabender, 2020-05-31:
I am seeing this also.  This is related to the recent changes to shutdown tasks I believe.

This is from a test run with head of master and a Lenz LI101F configured:

```
     [java] 101553 apps.AppsBase                         ERROR - Continuing after error in handleQuit [AWT-EventQueue-0]
     [java] java.util.ConcurrentModificationException
     [java] 	at java.util.HashMap$KeySpliterator.forEachRemaining(HashMap.java:1561)
     [java] 	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:481)
     [java] 	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:471)
     [java] 	at java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:708)
     [java] 	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
     [java] 	at java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:499)
     [java] 	at jmri.managers.DefaultShutDownManager.shutdown(DefaultShutDownManager.java:252)
     [java] 	at jmri.managers.DefaultShutDownManager.shutdown(DefaultShutDownManager.java:183)
     [java] 	at apps.AppsBase.handleQuit(AppsBase.java:400)
     [java] 	at apps.Apps.handleQuit(Apps.java:999)
     [java] 	at apps.PanelPro.PanelPro$1.actionPerformed(PanelPro.java:77)
     [java] 	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
     [java] 	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
     [java] 	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
     [java] 	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
     [java] 	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
     [java] 	at java.awt.Component.processMouseEvent(Component.java:6539)
     [java] 	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
     [java] 	at java.awt.Component.processEvent(Component.java:6304)
     [java] 	at java.awt.Container.processEvent(Container.java:2239)
     [java] 	at java.awt.Component.dispatchEventImpl(Component.java:4889)
     [java] 	at java.awt.Container.dispatchEventImpl(Container.java:2297)
     [java] 	at java.awt.Component.dispatchEvent(Component.java:4711)
     [java] 	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
     [java] 	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
     [java] 	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
     [java] 	at java.awt.Container.dispatchEventImpl(Container.java:2283)
     [java] 	at java.awt.Window.dispatchEventImpl(Window.java:2746)
     [java] 	at java.awt.Component.dispatchEvent(Component.java:4711)
     [java] 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
     [java] 	at java.awt.EventQueue.access$500(EventQueue.java:97)
     [java] 	at java.awt.EventQueue$3.run(EventQueue.java:709)
     [java] 	at java.awt.EventQueue$3.run(EventQueue.java:703)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
     [java] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
     [java] 	at java.awt.EventQueue$4.run(EventQueue.java:733)
     [java] 	at java.awt.EventQueue$4.run(EventQueue.java:731)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
     [java] 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
     [java] 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
     [java] 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
     [java] 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
     [java] 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

pabender, 2020-05-31:
@rhwood, I believe this is the same problem that was reportedly fixed related to issue #8598.

rhwood, 2020-05-31:
Completely unrelated; the problem in #8598 was that, on macOS, creating a JmmDNS NetworkListener enters into a deadlock, and for some reason, which I could trace, we were creating one only on shutdown, when it should have been created when the first ZeroConfService was created.

commit 0ed5c013 (2020-06-03): fix: prevent CME while running tasks when stopping

Fixes #8616
PR #8617 (2020-06-04): fix: prevent CME while running tasks when stopping
Fixes #8616
commit 2525eff4 (2020-06-04): fix: prevent CME while running tasks when stopping

Fixes #8616
closed 2020-06-03 by bobjacobsen