JMRI/JMRI #8644: SML Discovery discovers incorrect mast

devel-bobm, 2020-06-05:
Working with "head of master" pulled at about 2:30pm Eastern US time 05Jun2020.  Developing a new Layout Editor panel.  SML Discovery from one mast _incorrectly_ detects a signal mast that **IS NOT** reachable.  Path would require a "reverse move in the plant" to get to the destination signal. 

This can be seen in the attached screen capture image - note that SML for LA228 shows _two_ active destination masts - LA224 (correct) and RA230 (incorrectly identified!).  It is _not_ possible to reach RA230 from LA228, without restoring 227B to "normal" followed by a "reverse" move.

[image]

I suspect that this problem occurs because I have added "track segment"s to each leg of switch 227B and assigned them to the _same_ block as switch 227B.  I am guessing that SML Discovery can become confused when these track segments are added to the switches.  Alternately, this could relate to starting the panel in a previous JMRI development version, since the last edit before today's experimenting was in October of 2019.

Attached is the XML I'm using, with a LocoNet Simulator connection configuration.

[LE Panel with troublesome SML discovery.zip](https://github.com/JMRI/JMRI/files/4738728/LE.Panel.with.troublesome.SML.discovery.zip)

dsand47, 2020-06-06:
This issue is caused by an unusual block configuration.

Signal masts placed at turnout block boundaries protect the turnout block and the blocks connected to the turnout are known as facing blocks.  This means that the turnout block will never be a facing block since it is not possible to attach a signal mast so that it can be seen while standing in the turnout.  In other words, controlling the exit from the turnout.

By extending OS277 with a track segment to the right with **mast ra230** attached to the anchor point block boundary, the turnout block has become both a protected block and a facing block.  That makes **mast ra230** an eligible destination mast.

There are three solutions.
- Delete the bad destination mast from the Signal Mast Pairs list.
- Delete **mast ra230** from the panel.
- Change the block segment between the turnout and the anchor point to have a different block name.

devel-bobm, 2020-06-06:
Thanks, Dave, for your comments.

The first suggested solution is not a very good one as it would need to be done at every panel load.  I am not fond of requiring a jython script to be run at every panel load.

Removal of RA230 is not acceptable.  That mast is indeed there.

I suspect moving the turnout-adjacent track segments to new Blocks, where the occupancy is the same as for the turnout itself may be a solution.

Alternately, perhaps I can simply "stretch" the turnout to be large enough to get the GUI connectivity I need, without any additional track segments.

devel-bobm, 2020-06-06:
Perhaps something could be added to the docs about this troublesome "use case".

dsand47, 2020-06-06:
Bob,

The first solution does work.  The additional destination mast **only** occurs during discovery.  The only risk is that if you re-discover the destinations for LA228, the bad destination will be found at which time it can be deleted again.

The legs of a turnout can be as long or as short as you want them.  The size of the thoat and normal legs are always the same length from the center point.  Changing one changes the other.  The length of the diverging leg is independent.

For RA230 there is obviously an unknown requirement.  I have never seen an interlocking plant where a train can enter the plant but an outbound signal at the plant boundary can stop a train from leaving the plant.

n3ix, 2020-06-06:
This seems to only happen to me at crossovers.   When the panel is drawn it is often desirable to have the entry and exit points of both tracks align vertically.  However you cannot extend the horizontal legs of the crossover turnouts without screwing up the track spacing.  So the only alternative is to add tracks segments to the panel.  Logically it seems like the thing to do is just use the same block – I know that I’ve done the same thing and had the same result.   So maybe in addition to noting the case as a problem another nice thing would be to have a crossover turnout icon where the righthand anchor points have the same X coordinate?  Same for the lefthand anchor points.  Just a thought.

 

Robin

 

Robin Becker

San Diego, CA

dsand47, 2020-06-06:
> This seems to only happen to me at crossovers. When the panel is drawn it is often desirable to have the entry and exit points of both tracks align vertically. 

This can be done using a track segment and an anchor point to extend the short crossover legs.
[image]
The single head masts were attached to the anchor point protecting the turnout block.  The mast position protecting the next track segment is left empty.

The requirement for any turnout is that its block can only contain inbound signal masts.  Bob's situation is that he also has an outbound signal mast.

n3ix, 2020-06-06:
Dave,

 

Interesting, that’s how I do it with crossovers.  Maybe I was thinking of crossovers I created using 2 turnouts?  Anyway appreciate the enlightenment!

 

Robin

 

 <https://user-images.githubusercontent.com/17280992/83947426-69eced00-a7dc-11ea-8b8f-4a3eac8e4fc3.png>

devel-bobm, 2020-06-07:
Dave,

If I discover SML, delete the unwanted SML entry, save the panel, quit PanelPro, re-start PanelPro, load the panel, and open the mast's SML, I find that the unwanted SML entry has been re-created, and needs to once-again be deleted.  To me, the need to delete every time the panel is loaded is a non-starter.

There is a specific prototype for this arrangement - MX Cabin, just West of Hinton W. Va on the C&O's Alleghany (sp?) Sub.  

This location is part of what I am attempting to describe in the Layout Editor panel.  The prototype used the Main 2 signals RA230/LA226 act as "holding signals".  Specifically, a loaded (Eastward) coal train departing Avis Yard (the middle track at the left edge of the panel) would get an air test of the power and cars and caboose, then proceed by signal RA226 indication, onto the main and just past the interlocking (with clearance hooped-up when passing MX Cabin), stopping short of signal RA230.  The helpers were then added at the rear, and an "initial terminal" air test was completed on those helpers.  Once the air test passed, the operator at MX Cabin would be notified, the dispatcher notified to clear signal RA230 and the clearance hooped-up.

I am under the impression that an "outgoing" holding signal as part of an OS is not a common implementation, but they do exist in some modern signaling implementations.

Regards,
Billybob

dsand47, 2020-06-07:
> Dave, If I discover SML, delete the unwanted SML entry, save the panel, quit PanelPro, re-start PanelPro, load the panel, and open the mast's SML, I find that the unwanted SML entry has been [...]

That was a very good description of the prototype practice.  Very interesting.

I have tried several ideas but I cannot get **mast la228** to automatically create the bad destination after loading.  Do you have any scripts or startup actions?

You do have a large number of SML errors.  I recommend that you delete all of the SML table entries, re-generate and then remove the bad entry for la228.  It probably will not fix the problem but at least the errors will be gone.

 While researching this problem I came across another one.  **mast la228** is a "Double searchlight high absolute signal".  The appearance file for this mast type is appearance-SL-2-high-abs.xml.

The destination masts, **Avis inbound** and **Hinton Eastward Reverse movement**, can only display **Stop and Proceed**.  The other aspects have been disabled.  The appearance file does not have a aspect mapping entry for **Stop and Proceed** so **mast la228** always displays **Stop**.  That also affects **mast la226**.

closed 2021-04-14 by stale[bot]