JMRI/JMRI #8646: RFE: add support for sensor stretching to JMRI

mrchuk, 2020-06-06:
This RFE follows on from a discussion in the forums in this thread:
https://groups.io/g/jmriusers/topic/sensor_debounce_delays_what/74596643?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,20,74596643

Like many other people, I'm using commodity IR sensors for train detection on the layout. These are not very smart, and prone to generating very many transitions in state even when operating correctly, in a short space of time. This in turn is reported by AbstractSensor.java as 'sensor flapping',

While AbstractSensor.java supports debounce for sensors, it does not support stretching (both terms as supplied and defined by Bob J here:
https://github.com/JMRI/JMRI/issues/5529#issuecomment-403263465)

I believe adding 'stretching' support via one of the well-known algorithms for this would make JMRI more versatile dealing with hardware such as  I and many others have. Something along the lines of what is described here:
http://www.sumidacrossing.org/LayoutElectricity/microcontrollers/Arduinos/IRSensors/SensorSmoothing/

though the library referred to doesn't seem to have been published. I have asked the author about this.

Hopefully enough people think this might be worthwhile enough to get some traction.
Charlie

KenC57, 2020-06-06:
As I see the example, the debouncesenesor.py would simply be set with zero
for the on time requirement and the off time would be longer than those
pulses. The output would go on with the first pulse but not turn off until
it had been greater than the expected pulse. If trying to detect trains via
IR the on delay would be zero and the off would be a second or two to allow
for slow trains. But at some point IR fails when the train is super slow or
stops 'just right'. To use the existing built in sensor, the 'delay to
active' would be zero, while the 'delay to inactive' would be much longer.
-ken c

bobjacobsen, 2020-06-07:
Are you interested only in INACTIVE -> ACTIVE -> INACTIVE?  Or the other case, ACTIVE -> INACTIVE -> ACTIVE too?

mrchuk, 2020-06-07:
That's an interesting question Bob. The more I look into this, the trickier it seems to get. There appear to be loads of solutions out there for smoothing analog sensors but not much for dealing with digital ones in the scenario you've called 'stretching'.

Would there need to be a separate implementation for each case?

I'm not personally concerned about false positives or negatives (though others may be), the sensors I have are reliable enough in that respect. It's more a case of too many positives and too many negatives.

Some sort of hold-state timer for each sensor perhaps--ignoring state changes that occur within that window? Averaging a series of observations, or summing them over time and comparing to a floor value e.g. were there 3 ACTIVE out of the last 5 samples? 

I'm sure there are methods out there to deal with this situation--I just don't happen to know what they are.

closed 2021-04-14 by stale[bot]