JMRI/JMRI #8664: Layout Editor Regression: Zoom in and then unable to Zoom back out

pnbgit, 2020-06-09:
I'm not sure what release this problem crept in, but in certain conditions you can Zoom a layout editor panel in to see it better to work on it in edit mode, but then you cannot Zoom back out to the level you want it to be for production use.  All the Zoom levels below the current one get greyed out.  

Dave S tells me this is intentional behaviour as LE is claiming to somehow know that you don't need to Zoom out again. I don't understand how it can know what Zoom level you want, it definitely got it wrong in my cases. This behaviour is very confusing for the user, particularly beginners. 

There are workarounds, such as resizing the Window or deleting certain files in the profile, but none of these are easy or obvious to a beginner, and still a real PITA even for an experienced user like myself.

This is a regression as the problem does not occur in earlier releases. Greying out certain Zoom levels serves no useful purpose for the user, AFAICS.  

Please could we have the previous behaviour back, where the user can always choose the Zoom level according to their own preference.

Here is a simple example with some random track elements placed on a panel. Having Zoomed in to work on it, now I can't Zoom back out to Zoom level 0 which is how I want to use the panel.

[First panel ver 1.2.zip](https://github.com/JMRI/JMRI/files/4750644/First.panel.ver.1.2.zip)

geowar1, 2020-06-09:
I can "unfix" this… did you try "zoom to fit"?

KenC57, 2020-06-09:
Zoom to fit at least let me line things up again. But I can't shrink either.