JMRI/JMRI #8671: Defaults preferences retains connection even when connection is removed

rhwood, 2020-06-11:
It appears that when the Connections are changed, and a connection that is currently set to provide the default connection for anything is removed, the Defaults preferences do not set that preference to an existing connection.

Note that I do not know what the Defaults preferences should do, just that it should not hang on to a will-no-longer-be-valid connection, so the correct behavior is open to discussion.

See https://groups.io/g/jmriusers/topic/74789380

KenC57, 2020-06-11:
How is this for an idea of how to handle the defaults:
1. When an interface is deleted, any defaults that point to that interface
are set to Internal.
2. When an interface is added, any defaults that are pointing to Internal
that are supported by the new interface become the default.

This way deleting or adding a secondary interface does not touch defaults of
a primary interface. Also if you were 'replacing' a default interface, it
would pick up the new one as default while leaving anything that secondary
interfaces had set alone.

Cliff-Anderson, 2020-06-13:
As requested by @rhwood in https://groups.io/g/jmriusers/message/175285 zipped Config Profile folder is attached.  

Folder was "Copied" a few minutes ago using version 4.18 to show incomplete update in the "profile.properties" file.   JMRI System Console shows WARN messages that duplicate those in the OP.  

If version 4.19.7 is used to change the connection, Uncaught Exception forces me to "Quit" to preserve input data.

[test_Loconet.jmri.zip](https://github.com/JMRI/JMRI/files/4775179/test_Loconet.jmri.zip)
[broken session.log](https://github.com/JMRI/JMRI/files/4775180/broken.session.log)

Cliff-Anderson, 2020-06-13:
For the sake of safety, the Config Profile folder that was Copied might provide some clues.  

Certainly the Defaults look consistent on the "previous" folder even if the were not copied correctly.  

[test_EZ.jmri.zip](https://github.com/JMRI/JMRI/files/4775237/test_EZ.jmri.zip)

rhwood, 2020-06-13:
There are two different things going on. The exception shown in the logs copied here and reported on the user's group have nothing to do with the defaults, but are reproducible when removing a connection which provides no defaults.

So the defaults not changing is not the source of the exception reported and are unrelated to the exception reported. See #8679 for that issue.

dheap, 2020-06-14:
As reported on [jmriusers](https://groups.io/g/jmriusers/message/175305):
I started with this profile:
[SPROG_Command_Station_with_NCE_Simulator.jmri.zip](https://github.com/JMRI/JMRI/files/4775698/SPROG_Command_Station_with_NCE_Simulator.jmri.zip)
Duplicated it to create:
[SPROG_Simulator_with_NCE_Simulator.jmri.zip](https://github.com/JMRI/JMRI/files/4775699/SPROG_Simulator_with_NCE_Simulator.jmri.zip)

[messages.txt](https://github.com/JMRI/JMRI/files/4775701/messages.txt)

Looking at it again, it may not be the same problem...

silverailscolo, 2021-01-05:
Verify the deletion of a connection exception was fixed by PR #9271, allowing the process of reassigning defaults to complete.
This leaves open the issue of either silently replacing orphaned defaults to the first available "real" connection or popping up a dialog to choose (for 4.23.1?)

closed 2021-04-14 by stale[bot]