JMRI/JMRI #8673: 4.19.7 PanelPro CPE - Item Palette - multiple Preview pane

sidlo64, 2020-06-11:
Non critical issue
How to reproduce
Use 4.19.7 release
Start PanelPro
Panels – New Panel – Control Panel Editor
Open Add Items – Item Palette
Switch to Turnout Indicator tab
In tab is Turnout Table, Sensor pane, **one Preview pane**, radio buttos pane and buttons pane.
Click New Icon Set. 
Fill name Bug and click OK. In tis moment Turnout Indicator tab has **two Preview panes**.
Open Occupied Edit Icons. Now tab has **three Preview panes**.
Fill turnout icon and click Done. Tab has **four Preview panes**.
Open Clear Edit Icons. Now tab has **five Preview panes**.
…
After restart PanelPro Item Palette has only one Preview pane.

[image]

petecressman, 2020-08-02:
This issue is fixed in the pending PR #8862. Hopefully to be merged into 4.21.?

silverailscolo, 2021-01-27:
Bug is fixed in current release

closed 2021-01-27 by silverailscolo